Koenkk/zigbee2mqtt #27483: [New device support]: Quoya MC11 Blind controller

dannyuk1982, 2025-05-23:
### Link

https://www.quoyaliving.com/products/mc11-battery-powered-smart-blinds-motor-gateway-bundle

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x7cc6b6fffe684386","nwkAddr":33432,"manufId":4098,"manufName":"_TZE200_vdiuwbkq","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_vdiuwbkq","powerSource":3,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1747995426735}

### Zigbee2MQTT version

2.3.0

### Comments

Sorry I'm pretty new to HA, I tried following the guide at https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_1-pairing-the-device-with-zigbee2mqtt but I can't work out what I am meant to do after `[2.1. Extending the generated external definition](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-1-extending-the-generated-external-definition)`.

I'm guessing the device isn't supported? Is it possible to add in support or can anyone guide me in the direction on how to?

The blinds go up and down and the motor direction back/forwards work but there doesn't seem to be a way to set limits, the border up/down/down_delete don't seem to do anything. I can set limits in the Tuya app, but can't work how to in HA/zigbee2mqtt

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_vdiuwbkq',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

I don't even understand what this does or how it would work or not work, sorry.

dannyuk1982, 2025-05-26:
Well I managed to work this out - it didn't load and override the default converter a) because I put it in `zigbee2mqtt/data/external_converters` and it just needed to be in `zigbee2mqtt/external_converters` and then b) because there were errors that didn't show up in the `z2m-external-converter-dev` codespace. It doesn't see that z2m gives any feedback on the error other than renaming the file to .invalid

The docs are super confusing on this, the Tuya example page gives a .js example when it seems we need to use .mjs now, and it imports/requires loads of things that are never used or needed. And the data directory seems to be wrong too. Working out what exposes, cover_position, meta etc did was hard work too as these don't seem to be mentioned anywhere I could find.

In case anyone else wants to use this, this works for me - save it as quoya-mc11.mjs in zigbee2mqtt/external_converters and restart z2m

```
import {access, presets} from "zigbee-herdsman-converters/lib/exposes";
import * as tuya from "zigbee-herdsman-converters/lib/tuya";

export default {
    fingerprint: tuya.fingerprint("TS0601", ["_TZE200_gubdgai2", "_TZE200_vdiuwbkq"]),
    model: "M515EGBZTN V2",
    vendor: "Quoya",
    description: "Quoya Zigbee Roller Blind (M515EGBZTN V2)",
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        presets.cover_position().setAccess("position", access.STATE_SET).setAccess("state", access.STATE_SET),
        presets.enum("motor_direction", access.STATE_SET, ["forward", "back"]).withDescription("Motor direction"),
        presets.enum("cover_state", access.STATE, ["opening", "closing"]).withDescription("Cover state"),
        presets.enum("border", access.STATE_SET, ["up", "down", "up_delete", "down_delete", "remove_top_bottom"]).withDescription("Limit setting"),
        //presets.binary("fault", access.STATE, true, false).withDescription("Device faults"),
        //presets.numeric("position_best", access.STATE_SET).withValueMin(0).withValueMax(100).withUnit("%").withDescription("Best position"),
    ],
    meta: {
        tuyaDatapoints: [
            [1, "state", tuya.valueConverterBasic.lookup({OPEN: tuya.enum(0), STOP: tuya.enum(1), CLOSE: tuya.enum(2)})],
            [2, "position", tuya.valueConverter.coverPositionInverted],
            [5, "motor_direction", tuya.valueConverterBasic.lookup({forward: 0, back: 1})],
            [7, "work_state", tuya.valueConverterBasic.lookup({opening: 0, closing: 1})],
            [16, "border", tuya.valueConverterBasic.lookup({up: tuya.enum(0),down: tuya.enum(1), up_delete: tuya.enum(2), down_delete: tuya.enum(3), remove_top_bottom: tuya.enum(4)}),],
            //[12, "fault", tuya.valueConverter.raw],
            //[19, "position_best", tuya.valueConverter.raw],
        ],
    },
};
```

Koenkk, 2025-05-26:
@dannyuk1982 Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

dannyuk1982, 2025-05-27:
> [@dannyuk1982](https://github.com/dannyuk1982) Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

I think I have? Not familiar with this at all, only installed HA last week! Also not sure what the difference between Typescript and JS is - looks like the same syntax, I had to change the access, presets, options imports to e, ea and exposes.options - not sure if this is correct.

closed 2025-08-04 by github-actions[bot]