Koenkk/zigbee2mqtt #27490: [New device support]: please add support for ENGO controll box 'ECB62-ZB', _TZE200_oahqgdig

3DN1M, 2025-05-23:
### Link

https://engocontrols.com/en/produkt/ecb62-zb-3/

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xdc8e95fffe8ade6d","nwkAddr":51603,"manufId":4098,"manufName":"_TZE200_oahqgdig","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,2,3,4,5,6],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6,3,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"ŵ�/iƵ�/iƵ�/iƵ�/iǵ�/iǵ�/i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_oahqgdig","stackVersion":0,"dateCode":"","appVersion":66}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002d12c331","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,3,61184],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002d12c331","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6,3,61184],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002d12c331","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"4":{"profId":260,"epId":4,"devId":256,"inClusterList":[4,5,6,3,61184],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002d12c331","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"5":{"profId":260,"epId":5,"devId":256,"inClusterList":[4,5,6,3,61184],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002d12c331","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"6":{"profId":260,"epId":6,"devId":256,"inClusterList":[4,5,6,3,61184],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002d12c331","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1748029910636}

### Zigbee2MQTT version

2.3.0-1

### Comments

I've tried adapting examples from other issues for my device but to no success, maybe I am missing someting 

information from mqtt explorer for switch 2 f.e. not sure if this helps:
{
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state",
      "value_template": "{{ value_json.state }}"
    }
  ],
  "command_topic": "zigbee2mqtt/0xdc8e95fffe8ade6d/2/set",
  "device": {
    "hw_version": 1,
    "identifiers": [
      "zigbee2mqtt_0xdc8e95fffe8ade6d"
    ],
    "manufacturer": "_TZE200_oahqgdig",
    "model": "Automatically generated definition",
    "model_id": "TS0601",
    "name": "0xdc8e95fffe8ade6d",
    "via_device": "zigbee2mqtt_bridge_0x00124b002d12c331"
  },
  "name": "2",
  "object_id": "0xdc8e95fffe8ade6d_2",
  "origin": {
    "name": "Zigbee2MQTT",
    "sw": "2.3.0",
    "url": "https://www.zigbee2mqtt.io"
  },
  "payload_off": "OFF",
  "payload_on": "ON",
  "state_topic": "zigbee2mqtt/0xdc8e95fffe8ade6d",
  "unique_id": "0xdc8e95fffe8ade6d_switch_2_zigbee2mqtt",
  "value_template": "{{ value_json.state_2 }}"
}

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_oahqgdig',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3,"4":4,"5":5,"6":6}}), m.onOff({"powerOnBehavior":false,"endpointNames":["1","2","3","4","5","6"]})],
    meta: {"multiEndpoint":true},
};
```

### What does/doesn't work with the external definition?

switches appear in the HA GUI, but when I try to change the state relays does not activate on the device itself.

3DN1M, 2025-05-24:
clicking on on the first switch creates this log message in the z2m logs:

z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xdc8e95fffe8ade6d', payload '{"linkquality":218,"state_1":"ON","state_2":"OFF","state_3":"OFF","state_4":"OFF","state_5":"OFF","state_6":"OFF"}'

3DN1M, 2025-05-31:
switching a switch z2m debug log:
`[2025-05-30 23:10:07] debug: 	z2m: Publishing 'set' 'state' to '0xdc8e95fffe8ade6d'
[2025-05-30 23:10:07] debug: 	zh:controller:endpoint: ZCL command 0xdc8e95fffe8ade6d/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-05-30 23:10:07] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xdc8e95fffe8ade6d:7295/1 (0,0,1)
[2025-05-30 23:10:07] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":7295,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":234,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,24,0]}}
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:writer: --> frame [254,13,36,1,127,28,1,1,6,0,234,0,30,3,1,24,0,163]
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-05-30 23:10:07] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,234,44]
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,234,44]
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,234] - 44
[2025-05-30 23:10:07] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":234}
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: <-- [254,25,68,129,0,0,6,0,127,28,1,1,0,153,0,244,13,197,0,0,5,8,24,11,0,0,127,28,29,124]
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: --- parseNext [254,25,68,129,0,0,6,0,127,28,1,1,0,153,0,244,13,197,0,0,5,8,24,11,0,0,127,28,29,124]
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: --> parsed 25 - 2 - 4 - 129 - [0,0,6,0,127,28,1,1,0,153,0,244,13,197,0,0,5,8,24,11,0,0,127,28,29] - 124
[2025-05-30 23:10:07] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":7295,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":153,"securityuse":0,"timestamp":12914164,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[8,24,11,0,0]}}
[2025-05-30 23:10:07] debug: 	zh:controller: Received payload: clusterID=6, address=7295, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=153, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":24,"commandIdentifier":11},"payload":{"cmdId":0,"statusCode":0},"command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}}
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-05-30 23:10:07] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xdc8e95fffe8ade6d', payload '{"linkquality":153,"state":"OFF","state_1":"OFF","state_2":"OFF","state_3":"OFF","state_4":"OFF","state_5":"OFF","state_6":"OFF"}'
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: <-- [254,35,68,129,0,0,0,0,127,28,1,1,0,153,0,231,76,197,0,0,15,8,60,10,1,0,32,66,226,255,32,31,228,255,32,0,127,28,29,71]
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: --- parseNext [254,35,68,129,0,0,0,0,127,28,1,1,0,153,0,231,76,197,0,0,15,8,60,10,1,0,32,66,226,255,32,31,228,255,32,0,127,28,29,71]
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: --> parsed 35 - 2 - 4 - 129 - [0,0,0,0,127,28,1,1,0,153,0,231,76,197,0,0,15,8,60,10,1,0,32,66,226,255,32,31,228,255,32,0,127,28,29] - 71
[2025-05-30 23:10:07] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":7295,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":153,"securityuse":0,"timestamp":12930279,"transseqnumber":0,"len":15,"data":{"type":"Buffer","data":[8,60,10,1,0,32,66,226,255,32,31,228,255,32,0]}}
[2025-05-30 23:10:07] debug: 	zh:controller: Received payload: clusterID=0, address=7295, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=153, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":60,"commandIdentifier":10},"payload":[{"attrId":1,"dataType":32,"attrData":66},{"attrId":65506,"dataType":32,"attrData":31},{"attrId":65508,"dataType":32,"attrData":0}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-05-30 23:10:07] debug: 	zh:controller:endpoint: ZCL command 0xdc8e95fffe8ade6d/1 genBasic.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":60,"writeUndiv":false})
[2025-05-30 23:10:07] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xdc8e95fffe8ade6d:7295/1 (0,0,1)
[2025-05-30 23:10:07] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":7295,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":235,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,60,11,10,0]}}
[... cut ...]
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,235,45]
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,235,45]
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,235] - 45
[2025-05-30 23:10:07] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":235}
[2025-05-30 23:10:07] debug: 	zh:zstack:unpi:parser: --- parseNext []`

closed 2025-08-24 by github-actions[bot]

alexisrolland30-sketch, 2025-11-18:
It's possible to see your configuration, for use swith ?

thx ;)

3DN1M, 2025-11-21:
no, I was not able to pair 6 'zigbee' zones to z2m or any other way
it seems that they only can be paired to engo thermostats.
and I am not sure if its even technically possible, if it is it can be updated theoretically from HA with new features.