Koenkk/zigbee2mqtt #27491: Ikea RODRET wireless dimmer switch is Unsupported

MrTeamaker, 2025-05-24:
### What happened?

I paired a RODRET wireless dimmer switch with my system. I was able to connect to by Coordinator, however the device came through as Unsupported.

It currently only exposes battery and linkquality, N/A for Action. 

### What did you expect to happen?

I expected it to be supported and integrate without any issues as it's listed in the supported devices list
(https://www.zigbee2mqtt.io/devices/E2201.html).

Although I did notice that the supported device had a vendor "IKEA" whereas my device had a manufName "IKEA of Sweden".

### How to reproduce it (minimal and precise)

First enable Permit to join in the Z2M frontend, then press the connect button on the RODRET x4. The red LED will flash a few times, then it'll show in Z2M as Unsupported.

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Setup

Running Z2M in Proxmox, in a container on a mini HP. Was installed using the Proxmox community scripts, same as the MQTT container an the HASS VM.

### Debug log

[log.log](https://github.com/user-attachments/files/20419430/log.log)

champagnearni, 2025-05-27:
Similar issue. Same device, but actions are responding. Also listed as Not supported

MrTeamaker, 2025-05-28:
Hi @Koenkk, here is my debug log. 

I've tried to pair it a few times and had to force-remove the device in the process, not sure how that impacts the joining.

[log2.log](https://github.com/user-attachments/files/20495871/log2.log)

MrTeamaker, 2025-05-29:
@Koenkk Just a quick update. As it turns out, the actions of the dimmer switch actually come through, not sure why they do now and not before. The device shows as unsupported but after a while the device actions are registered and I get the on/off/dim-up/dim-down.

Koenkk, 2025-05-31:
added it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-08-07 by github-actions[bot]

andrei-lazarov, 2025-12-15:
@bchhabra 
You can get the RODRET converter from zigbee-herdsman-converters, adjust the imports and put it in your external_converters folder.

But I would much rather do a backup, update and keep Z2M updated (and replace the coordinator if necessary).

bchhabra, 2025-12-23:
Hi @andrei-lazarov,

I upgraded to Zigbee2MQTT 1.40.2, and now the RODRET wireless dimmer works as expected in Home Assistant — I can see all actions being triggered correctly.

However, Z2M still shows the device as unsupported. I did not use external_converters, as I was expecting it to be recognised natively in 1.40.2.