Koenkk/zigbee2mqtt #27493: Zigbee modules on different adapters crashing regularly

Dinth, 2025-05-24:
### What happened?

Hi. Ive been using an USB Sonoff dongle stuck into my server until a couple of years ago. That was working stable in general, but had connectivity issues to the devices because of the server location, so around 2 years ago i have migrated to PoE coordinator located in the centre of my house. This greatly improved connectivity between the coordinator and zigbee devices, but i found that those adapters are regularly crashing. 
1. Doesnt matter the make and model, i have tried many SLZB-06 line devices which i have purchased, basically all models available, and with some models i have tried different units, i also tried ZigStar UZG-01 which i perceive as slightly more stable.
2. I have tried different drivers (as different SLZB-06 models have different chipsets): zstack, ember and the one which ember replaced
3. I have tried all kinds of Zigbee firmwares and ESP firmware combinations.
4. I have tried powering the coordinator devices both via PoE and USB-C
5. I have tried both ethernet and wifi set up on the coordinators.
6. I was suspecting some intermittent issues with my network, but i also have a very similar network based coordinator for ZWave (TubeZB Zwave kit) and that one works perfectly fine on PoE.

Generally speaking, its hard to find a thing which i havent tried over those frustrating 2 years. The adapters are always freezing, depending on the device model which im currently using as a coordinator and Zigbee firmware installed anywhere between every 1-14 days. The freezes as i call them usually look like this: the device quietly (doesnt trigger any events on mqtt, doesnt trigger watchdog in Z2M) stops communicating with the endpoints and stops responding to the version queries in Debug section on coordinator control panel. Because of the way it all fails quietly, its very hard to troubleshoot because usually i only notice its frozen when for example my plants start dying because not getting water from a zigbee controlled watering system - usually to late to find anything in the logs. 
To unfreeze it, i need to completely power cycle (just restarting Zigbee coordinator and ESP from the coordinators control panel DOES NOT help) the coordinator device and then restart Zigbee2MQTT. 

### What did you expect to happen?

Zigbee should work in a stable manner for longer.
When it fails, especially when it fails completely, it shouldnt do it quietly. 
Also, Z2M watchdog should kick in. 

### How to reproduce it (minimal and precise)

Not sure, it just happens to me with every configuration i have tried since migrating to network based. more modern Zigbee coordinators a couple of years ago.

### Zigbee2MQTT version

2.3.0-1

### Adapter firmware version

i think i have tried all stable and many unstable ones, but most recently it was 20250321

### Adapter

UZG-01, SLZB-06, SLZB-06M, SLZB-06p7, SLZB-06p10

### Setup

Since this issue started occuring i had two setups:
1. Intel based Mac Mini running HAOS bare metal
2. Dell R230 server running Proxmox and HAOS within Proxmox container. 
Z2M was running as an addon in both cases

### Debug log

Z2M log: [45df7312_zigbee2mqtt_2025-05-24T05-36-48.770Z.log](https://github.com/user-attachments/files/20421536/45df7312_zigbee2mqtt_2025-05-24T05-36-48.770Z.log)

Log from Debug window in coordinators control panel:
```
6876194] -> fe 0f 24 01 92 57 01 01 00 00 d6 00 1e 05 10 24 00 00 00 16
[756904711] -> fe 0f 24 01 62 dd 01 01 00 00 d7 00 1e 05 10 25 00 00 00 6c
[756928746] -> fe 0f 24 01 41 86 01 01 00 00 d8 00 1e 05 10 26 00 00 00 18
[756952834] -> fe 0f 24 01 9d 9c 01 01 00 00 d9 00 1e 05 10 27 00 00 00 de
[760452452] -> fe 0f 24 01 c9 d0 01 01 00 00 da 00 1e 05 10 28 00 00 00 ca
[760469759] -> fe 0f 24 01 4f 76 01 01 00 00 db 00 1e 05 10 29 00 00 00 ea
[760671002] -> fe 0f 24 01 6a 44 01 01 00 00 dc 00 1e 05 10 2a 00 00 00 f9
[760986057] -> fe 0f 24 01 46 ca 01 01 00 00 dd 00 1e 05 10 2b 00 00 00 5b
[761173492] -> fe 0f 24 01 25 9b 01 01 00 00 de 00 1e 05 10 2c 00 00 00 6d
[761207554] -> fe 0f 24 01 f9 87 01 01 00 00 df 00 1e 05 10 2d 00 00 00 ad
[763072377] -> fe 0f 24 01 e4 9a 01 01 00 00 e0 00 1e 05 10 2e 00 00 00 91
[763082721] -> fe 0f 24 01 5b a2 0b 01 00 00 e1 00 1e 05 10 2f 00 00 00 1c
[763118276] -> fe 0f 24 01 00 3f 01 01 00 00 e2 00 1e 05 10 30 00 00 00 cc
[763275761] -> fe 0f 24 01 f1 c6 01 01 00 00 e3 00 1e 05 10 31 00 00 00 c4
[763283768] -> fe 0f 24 01 f2 9e 01 01 00 00 e4 00 1e 05 10 32 00 00 00 9b
[763345551] -> fe 0f 24 01 eb 71 01 01 00 00 e5 00 1e 05 10 33 00 00 00 6d
[763370289] -> fe 0f 24 01 0a 62 01 01 00 00 e6 00 1e 05 10 34 00 00 00 9b
[763381872] -> fe 0f 24 01 bc 55 01 01 00 00 e7 00 1e 05 10 35 00 00 00 1a
[763389874] -> fe 0f 24 01 0a 3d 01 01 00 00 e8 00 1e 05 10 36 00 00 00 c8
[763649587] -> fe 0f 24 01 08 55 01 01 00 00 e9 00 1e 05 10 37 00 00 00 a2
[763679474] -> fe 0f 24 01 36 4b 01 01 00 00 ea 00 1e 05 10 38 00 00 00 8e
[763799786] -> fe 0f 24 01 17 64 01 01 00 00 eb 00 1e 05 10 39 00 00 00 80
[... 87 lines cut ...]
[777680021] -> ef fe 00 21 01 20
[777685638] -> fe 00 21 01 20
[777691643] -> fe 00 21 01 20
[777699721] -> ef
[777700723] -> fe 00 21 01 20
[777706731] -> fe 00 21 01 20
[777712736] -> fe 00 21 01 20
[777720857] -> ef
[777721859] -> fe 00 21 01 20
[777727866] -> fe 00 21 01 20
[777733868] -> fe 00 21 01 20
[777743023] -> ef
[777743154] -> fe 00 21 01 20
[777758840] | [ZBCHK] Wrong answer
[777758840] | [ZBVER] Unknown
[777758845] -> fe 00 21 01 20 fe 00 21 01 20
[777763328] -> ef
[777764331] -> fe 00 21 01 20
[777770334] -> fe 00 21 01 20
[777776340] -> fe 00 21 01 20
[777784465] -> ef
[777785469] -> fe 00 21 01 20
[777791470] -> fe 00 21 01 20
[777797476] -> fe 00 21 01 20
```

Dinth, 2025-05-28:
Another crash. Sorry for spamming with the logs, but maybe it will be possible to correlate them to find whats causing the crashes

[45df7312_zigbee2mqtt_2025-05-28T18-54-06.642Z.log](https://github.com/user-attachments/files/20493741/45df7312_zigbee2mqtt_2025-05-28T18-54-06.642Z.log)

danir-de, 2025-05-30:
Related to #27553?

Dinth, 2025-05-31:
Looks very similar, but my problems started way before v2.8.5 and are not limited to SLZB hardware (happening on ZigStar too which i believe uses a developed independently fork of a very old version of SLZB firmware)

danir-de, 2025-05-31:
Maybe it helps, but for me I noticed similar issues since the Z2M v2 release, it just exploded for me once I upgraded, forcing me to really look at the issue.

closed 2025-08-07 by github-actions[bot]