Koenkk/zigbee2mqtt #27494: Zemismart display switch. Problem sending names not in English

ntk9, 2025-05-24:
### What happened?

I'm having an issue with my new Zemismart switch. When I try to change the display name via Z2M to any language other than English, the name either doesn't show up at all, or it displays numbers and English letters instead of the language I entered. The switch is supposed to support all of these languages.

chinese, english, french, garman, russian, portuguese, korean, japanese, vietnamese, arabic, hebrew, spanish

And from tuya app it work with other languages. So i think the problem is with the encoding of other languages send from z2m

The switch model is

Zigbee Model
TS0601

Zigbee Manufacturer
_TZE204_wwaeqnrf

Model
[ZMS-206US-4]

But i think the problem is with all the zemismart name switch (1-2-3-4 gang)

### What did you expect to happen?

To show the names correctly on yhe switch display 

### How to reproduce it (minimal and precise)

Just enter name in language other than English from the z2m device gui

### Zigbee2MQTT version

2.3.0-dev commit: 1303340a

### Adapter firmware version

7.4.4 [GA]

### Adapter

Sonoff EmberZNet

### Setup

Im running HA on VM from window 11. And zigbee2mqtt edge add on HA

Drag0n64, 2025-06-02:
Found a working workaround in the blog below.
In short - use notepad++ to enter your text in UTF, change encoding character set to western european ISO-8859-1, copy text and paste in HA. 
[https://github.com/hardycheng-github/smarthome-blog/blob/main/blog/zigbee/zemi_diy_name_118_wall_switch.md?fbclid=IwY2xjawKqwPVleHRuA2FlbQIxMABicmlkETFFSW9teWU0cGtUQ2tMbjFZAR5CfZpjkTIuAqSy5MLaIhC4Z3AKgiD3rx3bZU8hmRW63AGvaJLO_UUXkbeg8g_aem_kjOdIsOzgeXyMuEUT_kNqA](url)

ntk9, 2025-06-04:
From my test. This is working. But i get only 5-5 letters. Instead od 12-13 when using toya hub

closed 2025-08-12 by github-actions[bot]