Koenkk/zigbee2mqtt #27495: [Wrong device]: AwoX Rotemote Controller (Model: 33952, ERCU_Zm)

OUARZA, 2025-05-24:
### Link

https://www.zigbee2mqtt.io/devices/33952.html

### Model

33952

### Description

Remote Controller

### Vendor

Awox

### Picture (link)

https://community.jeedom.com/uploads/default/original/3X/2/7/273c3eefaa857a0238b228ba676de3c749df0e25.jpeg

### Database entry

{"id":48,"type":"EndDevice","ieeeAddr":"0xa4c13809e93e2d35","nwkAddr":50687,"manufId":4417,"manufName":"AwoX","powerSource":"Battery","modelId":"ERCU_Zm_CUSTOM","epList":[1,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":2048,"inClusterList":[0,3,4,4096],"outClusterList":[0,3,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"modelId":"ERCU_Zm","manufacturerName":"AwoX","powerSource":3,"zclVersion":3,"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.7.6_1477"}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":2048,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.7.6_1477","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1748031309157}

### Notes

---

### Description of the device

This PR adds full support for the **AwoX Remote Controller, Model 33952 (Zigbee Model ID: ERCU_Zm)**.

This is a multi-function remote that supports on/off, dimming, scene control, and color/color temperature adjustments.

### Device Interview Details

* **Model:** 33952
* **Vendor:** AwoX
* **Zigbee Model ID:** ERCU_Zm
* **Manufacturer Name:** AwoX
* **Endpoint:** 1
* **Power Source:** Battery (confirmed by device type "EndDevice" and typical remote behavior)

---

### Device Functionality (from logs)

I have captured detailed debug logs for each button press on the remote. The remote sends specific cluster commands and `raw` data payloads for various functions, alongside a generic `commandOn` on `genOnOff` for many actions. The proposed converter prioritizes specific actions to avoid redundant `on` commands.

Here's a breakdown of the observed functionality and corresponding Zigbee messages:

* **On:** `commandOn` on `genOnOff`
* **Off:** `commandOff` on `genOnOff`
* **Color Buttons (Blue, Green, Yellow, Red):** `raw` messages on `lightingColorCtrl` cluster. The 5th byte (`data[4]`) in the payload differentiates the colors:
    * Blue: `{"data":[17,214,48,0,169,8,0],"type":"Buffer"}` (0xD6)
    * Green: `{"data":[17,212,48,0,84,8,0],"type":"Buffer"}` (0xD4)
    * Yellow: `{"data":[17,210,48,0,42,8,0],"type":"Buffer"}` (0xD2)
    * Red: `{"data":[17,208,48,0,254,8,0],"type":"Buffer"}` (0xD0)
* **Brightness+:** `commandStepWithOnOff` on `genLevelCtrl` with `{"stepmode":0,"stepsize":18,"transtime":2}`
* **Brightness-:** `commandStep` on `genLevelCtrl` with `{"stepmode":1,"stepsize":18,"transtime":2}`
* **Light Movement (e.g., color cycle):** `commandEnhancedMoveHue` on `lightingColorCtrl` with `{"movemode":1,"rate":3072}`
* **"Refresh" Button (unlabeled):** `raw` message on `genLevelCtrl` with `{"data":[17,223,16,1,1],"type":"Buffer"}` (identified by `data[1]` being `0xDF`)
* **Favorite 1:** `commandRecall` on `genScenes` with `{"groupid":0,"sceneid":1}`
* **Favorite 2:** `commandRecall` on `genScenes` with `{"groupid":0,"sceneid":2}`
* **Warm Color:** `commandStepColorTemp` on `lightingColorCtrl` with `{"maximum":0,"minimum":0,"stepmode":1,"stepsize":32,"transtime":2}`
* **Cold Color:** `commandStepColorTemp` on `lightingColorCtrl` with `{"maximum":0,"minimum":0,"stepmode":3,"stepsize":32,"transtime":2}`

---

### Proposed Converter Code

I have developed a custom `fromZigbee` converter that correctly parses these messages and exposes appropriate actions.

```javascript
// Add this definition to a new file or an appropriate existing file in the Zigbee2MQTT device definitions.
// Example: devices/awox.js or a new file like devices/awox_33952.js

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ERCU_Zm'],
    model: '33952',
    vendor: 'AwoX',
    description: 'AwoX Remote Controller',
    supports: 'on/off, dimming, scene control, color control, color temperature control',
    fromZigbee: [fz.legacy_action_rate, fz.on_off, fz.awox_remote_actions], // Using a unique name for the custom converter
    toZigbee: [],
    exposes: [
        e.action([
            'on', 'off',
            'brightness_step_up', 'brightness_step_down',
            'color_blue', 'color_green', 'color_yellow', 'color_red',
            'color_temp_warm', 'color_temp_cold',
            'light_movement',
            'refresh',
            'scene_1', 'scene_2',
        ]),
    ],
    configure: async (device, coordinator, ledger) => {
        // No specific configuration needed based on current observations.
    },
};

const awox_remote_actions = {
    cluster: 'genOnOff', // Base cluster, actual actions depend on type and payload
    type: ['commandOn', 'commandOff', 'raw', 'commandStepWithOnOff', 'commandStep', 'commandEnhancedMoveHue', 'commandRecall', 'commandStepColorTemp'],
    convert: (model, msg, publish, options, meta) => {
        const payload = msg.data;
        let action = null;

        // Prioritize specific actions over generic on/off
        if (msg.cluster === 'lightingColorCtrl') {
            if (msg.type === 'raw') {
                const colorByte = payload.data[4];
                switch (colorByte) {
                    case 0xD6: action = 'color_blue'; break;
                    case 0xD4: action = 'color_green'; break;
                    case 0xD2: action = 'color_yellow'; break;
                    case 0xD0: action = 'color_red'; break;
                }
            } else if (msg.type === 'commandEnhancedMoveHue') {
                action = 'light_movement';
            } else if (msg.type === 'commandStepColorTemp') {
                if (payload.stepmode === 1) {
                    action = 'color_temp_warm';
                } else if (payload.stepmode === 3) {
                    action = 'color_temp_cold';
                }
            }
        } else if (msg.cluster === 'genLevelCtrl') {
            if (msg.type === 'commandStepWithOnOff' && payload.stepmode === 0) {
                action = 'brightness_step_up';
            } else if (msg.type === 'commandStep' && payload.stepmode === 1) {
                action = 'brightness_step_down';
            } else if (msg.type === 'raw' && payload.data && payload.data[1] === 0xDF) {
                action = 'refresh';
            }
        } else if (msg.cluster === 'genScenes') {
            if (msg.type === 'commandRecall') {
                if (payload.sceneid === 1) {
                    action = 'scene_1';
                } else if (payload.sceneid === 2) {
                    action = 'scene_2';
                }
            }
        }
        // Handle generic On/Off last if no specific action is found
        else if (msg.cluster === 'genOnOff') {
            if (msg.type === 'commandOn') {
                action = 'on';
            } else if (msg.type === 'commandOff') {
                action = 'off';
            }
        }

        if (action) {
            return {action: action};
        }
    },
};

module.exports = [
    definition,
];

Koenkk, 2025-05-25:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/awox.ts)?

OUARZA, 2025-05-25:
Hello @Koenkk , I did it, but I think I messed up.

closed 2025-08-01 by github-actions[bot]