Koenkk/zigbee2mqtt #27496: Unsupported Attributes for Clicksmart + CMA30651

Great-Chart, 2025-05-24:
### What happened?

Since recent update (2.3.0) the Clicksmart consumer unit is now correctly reporting rather than as generic Tuya device - as reflected in https://github.com/Koenkk/zigbee-herdsman-converters/pull/9234
However, backlight mode / Child lock (and since the update) also Power-on behaviour options (in exposes tab) do not function - they are reported as unsupported attributes.
In State tab these entries show up with values of "null"

### What did you expect to happen?

Update was envisaged to have provided full functionality

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

20221226 (newer FW has proven problematical for me)

### Adapter

Sonoff Dongle P

### Setup

Home Assistant OS on HP mini PC

### Debug log

From Database:-
{"id":130,"type":"Router","ieeeAddr":"0x8cf681fffeeacbce","nwkAddr":9380,"manufId":4098,"manufName":"_TZ3210_fhx7lk3d","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012","65506":31,"65508":0,"65534":0,"modelId":"TS0001","manufacturerName":"_TZ3210_fhx7lk3d","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":64,"powerSource":1}},"genOnOff":{"attributes":{"onOff":1,"offWaitTime":0,"onTime":0,"moesStartUpOnOff":0}},"manuSpecificTuya_3":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024bc1862","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1748082648251}

Pel1can111, 2025-07-25:
Just noticed this as well unfortunately.

thewhitehouse115-byte, 2025-08-05:
Having exactly the same problem.

Great-Chart, 2025-08-06:
@thewhitehouse115-byte; @Pel1can111; @stu3886 
Either of you find that the device has the child-lock applied and thus won't switch on/off from the manual switch / button?

Before the update when power cycled it'd reset that lock and the switch button would function but mine appears to have locked itself and won't release on a power cycle etc

stu3886, 2025-08-06:
> [@thewhitehouse115-byte](https://github.com/thewhitehouse115-byte); [@Pel1can111](https://github.com/Pel1can111); [@stu3886](https://github.com/stu3886) Either of you find that the device has the [...]

Mine has the child lock switched off (I never managed to turn it on). I just assumed the only way to do this was by using the clicksmart hub and app instead.

thewhitehouse115-byte, 2025-08-26:
Latest updates have now completely bricked the device. It now responds with errors every time I try to switch it on/off or alter any settings.  These devices worked perfectly when I first installed them about a year ago. I’m now left with 2  FCUs that no longer work and I can’t turn on my electric radiators.

Koenkk, 2025-08-29:
> However, backlight mode / Child lock 

Are these features supported by the device? (CC: @dodger421 )

Pel1can111, 2025-11-21:
Did anyone have any luck with this? Would be great to get the child lock working. Question to anyone else running these fused spurs... do you have issues with the power light locking on randomly? I tried the click support and they just said it suggests an issue with the device but it happens with all 3 I have. Resetting the power or reconnecting the device fixes it.

closed 2026-01-29 by github-actions[bot]

Pel1can111, 2026-09-22:
I have sniffed the traffic when using the clicksmart hub and device and got the child lock function working. The backlight option has never worked and is not available when using the official hub. I have created an external definition that can be added to get this working but this was created using claude. If someone wants to check over this and fix any mistakes AI has made/submit a pull request, feel free.

The default time-sync option for these devices should also be changed to 1970 as this appears to fix the orange error LED randomly coming on which can only be turned off by re-pairing or power cycling.

// External definition: ClickSmart+ CMA30651 13A Smart Switched Fused Connection Unit
// Tuya TS0001 / _TZ3210_fhx7lk3d  (Zigbee2MQTT 2.x)
//
// Child lock, confirmed from a Wireshark capture of the ClickSmart+ hub controlling the
// real device: cluster manuSpecificTuya (0xEF00), ZCL command ID 0x04 (NOT the library's
// standard "dataRequest"/DP-write helper, which uses a different command ID and was
// silently ignored by this unit). Payload observed: 00 0a 65 01 00 01 01
//   seq=0x000a, dp=0x65 (101, child_lock), type=0x01 (bool), len=0x0001, value=0x01 (true)
//
// Switch, countdown, power-on behaviour unchanged from the previous working version.

import {Zcl} from "zigbee-herdsman";
import * as exposes from "zigbee-herdsman-converters/lib/exposes";
import * as m from "zigbee-herdsman-converters/lib/modernExtend";
import * as tuya from "zigbee-herdsman-converters/lib/tuya";
import {access as ea} from "zigbee-herdsman-converters/lib/exposes";

const e = exposes.presets;
const CHILD_LOCK_DP = 0x65; // 101

// tuyaOnOff registers its own power_on_behavior (via the unknown attribute name
// "moesStartUpOnOff"), which shadows ours. Keep switch + countdown, drop its power-on control.
const rawOnOff = tuya.modernExtend.tuyaOnOff({onOffCountdown: true});
const isPob = (x) => x?.name === "power_on_behavior" || x?.property === "power_on_behavior";
const keyHasPob = (c) => (Array.isArray(c?.key) ? c.key.includes("power_on_behavior") : c?.key === "power_on_behavior");
const baseOnOff = {
    ...rawOnOff,
    exposes:
        typeof rawOnOff.exposes === "function"
            ? (...args) => rawOnOff.exposes(...args).filter((x) => !isPob(x))
            : (rawOnOff.exposes ?? []).filter((x) => !isPob(x)),
    toZigbee: (rawOnOff.toZigbee ?? []).filter((c) => !keyHasPob(c)),
};

// Add ONE extra command (ID 4, matching the capture) to the REAL, already-registered
// manuSpecificTuya cluster, in place, instead of replacing the whole cluster definition.
// (deviceAddCustomCluster substitutes the entire cluster and was wiping out Tuya's normal
// attributes/commands for this device, which broke the frontend's Bind tab.)
const CLICKSMART_CMD_NAME = "clickSmartSendData";
try {
    const tuyaCluster = Zcl.Utils.getCluster("manuSpecificTuya", undefined, {});
    if (tuyaCluster && !tuyaCluster.commands[CLICKSMART_CMD_NAME]) {
        tuyaCluster.commands[CLICKSMART_CMD_NAME] = {
            ID: 0x04,
            parameters: [{name: "payload", type: Zcl.BuffaloZclDataType.BUFFER}],
        };
    }
} catch (error) {
    // If the lookup API differs on this version, the command below will fail with a
    // clear "no command" error in the log rather than silently doing nothing.
}

let tuyaSeq = 1;
function buildDpBoolPayload(dp, value) {
    tuyaSeq = (tuyaSeq + 1) & 0xffff;
    return Buffer.from([
        (tuyaSeq >> 8) & 0xff, tuyaSeq & 0xff, // seq
        dp,                                    // datapoint id
        0x01,                                  // datatype: bool
        0x00, 0x01,                            // length: 1
        value ? 0x01 : 0x00,                   // value
    ]);
}

const tzChildLock = {
    key: ["child_lock"],
[... 50 lines cut ...]
    model: "CMA30651",
    vendor: "ClickSmart+",
    description: "13A Smart Switched Fused Connection Unit",
    extend: [
        baseOnOff,
        m.enumLookup({
            name: "power_on_behavior",
            lookup: {off: 0, on: 1, previous: 2},
            cluster: "genOnOff",
            attribute: {ID: 0x8002, type: Zcl.DataType.ENUM8},
            description: "Controls the behavior when the device is powered on after power loss",
            entityCategory: "config",
        }),
    ],
    fromZigbee: [fzChildLock],
    toZigbee: [tzChildLock],
    exposes: [e.child_lock()],
    options: [timeResponseOption],
    onEvent,
    configure: async (device, coordinatorEndpoint) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint);
    },
};

export default definition;