Koenkk/zigbee2mqtt #27499: [New device support]:

nickdisa, 2025-05-24:
### Link

https://www.aliexpress.com/item/1005008453758733.html?spm=a2g0o.order_list.order_list_main.39.298d18020I6sVx

### Database entry

{"description":"Automatically generated definition","model":"TS0601","supports":"linkquality","vendor":"_TZE284_idn2htgu"},"failed":[],"friendlyName":"F3Pro","ieeeAddr":"0xa4c138276f5de1ee","lastSeen":1748089188433,"manufacturerName":"_TZE284_idn2htgu","modelID":"TS0601","networkAddress":63280,"type":"Router"}

### Zigbee2MQTT version

2.3.0-1

### Comments

The device is unsupported and cannot make it work

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_idn2htgu',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

cannot make it work

ilmarh, 2025-06-11:
isn't it duplicate of #24624 ?

nickdisa, 2025-06-11:
Yes it is the same deviceNico Di ***@***.***

namanbharihoke, 2025-09-30:
https://github.com/namanbharihoke/-tuya-f3pro-zigbee2mqtt