Koenkk/zigbee2mqtt #27500: error: 	zh:ember:ezsp: The adapter has run out of buffers, causing general malfunction. Remediate network congestion, if present.

LeoeLeoeL, 2025-05-24:
### What happened?

I tried to pair 2 IKEA LED2201G8 bulbs.
Z2M has problems with them.
Initially log reported many:
error: 	z2m: Failed to read state of 'Luce Cucina W1' after reconnect (ZCL command 0x94ec32fffe012fc3/1 genLevelCtrl.read(["currentLevel"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '4473'.))

and:

error: 	z2m: Publish 'set' 'state' to 'Luce Cucina W2' failed: 'Error: ZCL command 0xc4d8c8fffeef7b46/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '64623'.)'

There were also many changes of Network address.

Then it started to show many:
error: 	zh:ember:ezsp: The adapter has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=86 Len=10].

Sometimes there were also:
error: 	zh:ember: Delivery of BROADCAST failed for '65533'.
and:
error: 	zh:controller: Failed to disable join on stop: Error: ~x~> [ZCL BROADCAST destination=65533] Failed to send with status=BUSY.
error: 	zh:controller: Failed to disable join on stop: Error: ~x~> [ZCL BROADCAST destination=65533] Failed to send with status=ALLOCATION_FAILED.

What i make to solve:
1) Remove the 2 Ikea bulbs (I had to force remove) - DID NOT SOLVE !
2) Restart Zigbee2MQTT - DID NOT SOLVE !
3) Restart Zigbee2MQTT container in Proxmox - DID NOT SOLVE !
4) Run an older release (2.2.1) of Zigbee2MQTT - DID NOT SOLVE !
5) Unplug and plug the coordinator (Sonoff Dongle E) - DID NOT SOLVE !
6) Restart all Proxmox - DID NOT SOLVE !
7) Shutdown all Proxmox and start again - DID NOT SOLVE !
8) Reflash the coordinator with the same FW (sonoff_zbdonglee_zigbee_ncp_8.0.2.0_115200_sw_flow) - DID NOT SOLVE !
9) Power off the lights section in the elecrtic panel for 30 seconds - IT WORKED !! (but I had 2 re-pair 2 routers and 2 end devices.).

### What did you expect to happen?

First of all I would like Ikea LED2201G8 work under Zigbee2MQTT. The integration doesn't work.
Then I would like something like a PANIC BUTTON to solve the problem of the error "zh:ember:ezsp: The adapter has run out of buffers, causing general malfunction. Remediate network congestion, if present."
If I read "the adapter has run out of buffers" I expect to solve the problem acting the coordinator. I acted (unplugging and reflashing) but I didn't solve.

### How to reproduce it (minimal and precise)

Read "What Happened?" section.

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

8.0.2 [GA]

### Adapter

EmberZNet

### Setup

Proxmox 8.4.1 4 x Intel(R) Core(TM) i5-6500T CPU @ 2.50GHz, containerized (https://community-scripts.github.io/ProxmoxVE/scripts?id=zigbee2mqtt)

LeoeLeoeL, 2025-05-24:
CPU load when the problem occurs:

[image]

LeoeLeoeL, 2025-08-22:
I looked for my previous thread ad dicovered all the network issues always happen 15/20 days after a Z2M update.

arossmann, 2025-09-04:
have the same issue since last update on zigbee2mqtt. Only difference for me: I have not added or changed any zigbee device

LeoeLeoeL, 2025-09-04:
Last august (with Z2M 2.6.0) I had the issue without any change. I was on vacacation.

Ed-Ross, 2025-11-03:
Same issue here.

u15jtau, 2026-10-27:
Just in case someone comes back, I had the issue again and commented out 
#  last_seen: ISO_8601_local
Restarted then the issue went away