Koenkk/zigbee2mqtt #27509: [New device support]: TS0601 _TZE284_n4ttsck2

GilDev, 2025-05-25:
### Link

https://fr.aliexpress.com/item/1005005869729913.html

### Database entry

{"id":23,"type":"EndDevice","ieeeAddr":"0xa4c13883be3c5a16","nwkAddr":18195,"manufId":4417,"manufName":"_TZE284_n4ttsck2","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"��/\u0011","65506":56,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE284_n4ttsck2","powerSource":3,"zclVersion":3,"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1748201929622}

### Zigbee2MQTT version

2.3.0

### Comments

Smoke sensor that was added a few months ago [in ZHA here](https://github.com/zigpy/zha-device-handlers/issues/3820).

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_n4ttsck2',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

Nothing is exposed aside from the link quality.

Koenkk, 2025-05-26:
`_TZE284_n4ttsck2` is supported in the dev branch.

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-05-26 by Koenkk

GilDev, 2025-05-26:
Oh sorry, thanks a lot! 🙂