Koenkk/zigbee2mqtt #27510: [New device support]: Philips Hue White and Color Ambiance GU10 Xamento 3-Pack

MattJeanes, 2025-05-26:
### Link

https://www.philips-hue.com/en-gb/p/hue-white-and-colour-ambiance-three-pack-xamento-recessed-spotlight/8720169321014

### Database entry

See https://github.com/Koenkk/zigbee2mqtt/issues/26076

### Zigbee2MQTT version

2.3.0

### Comments

This is the same as the supported 929003812901 model but just literally with _01 , _02 and _03 suffixed to the end of their model numbers for some reason. Functionality appears mostly normal even accepting OTA upgrades but shows as unsupported.

This is the same issue as https://github.com/Koenkk/zigbee2mqtt/issues/26076 but I am just opening a new issue as that was closed unresolved.

### External definition

```shell
import * as philips from 'zigbee-herdsman-converters/lib/philips';

export default {
    zigbeeModel: ['929003812901_03'],
    model: '929003812901_03',
    vendor: 'Signify Netherlands B.V.',
    description: 'Automatically generated definition',
    extend: [philips.m.light({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};
```

### What does/doesn't work with the external definition?

Light appears to function normally as far as I can see, just shows as unsupported.

Koenkk, 2025-05-26:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `MODEL.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

MattJeanes, 2025-05-26:
@Koenkk added device image:https://github.com/Koenkk/zigbee2mqtt.io/pull/3825

Ruvetuve, 2025-07-26:
the following device is a variant on this one, is this easy to support as well with the same config?

`929003812801`

Ref: https://github.com/Koenkk/zigbee2mqtt/issues/26076

Koenkk, 2025-07-27:
@Ruvetuve please create a new device support request ticket (https://github.com/Koenkk/zigbee2mqtt/issues)