Koenkk/zigbee2mqtt #27511: Problem with “NEO NAS-AB02B ZigBee 3.0 Smart Alarm Sirene”

Gandalf8266, 2025-05-26:
### What happened?

The device works after parking, but after some minutes, the communication ist broken, the device has status “deaktived” the logile shows this:

`Error 2025-05-26 06:14:36z2m: Publish 'set' 'alarm' to 'Sirene Garagenkeller' failed: 'Error: ZCL command 0xa4c13815f4cfe743/1 manuSpecificTuya.dataRequest({"seq":37,"dpValues":[{"dp":13,"datatype":1,"data":[1]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))' `

normale, the device is supported:
https://www.zigbee2mqtt.io/devices/NAS-AB02B2.html#neo-nas-ab02b2

All other zigbee devices are working fine

[image]
[image]

### What did you expect to happen?

It should work like all devices

### How to reproduce it (minimal and precise)

I connected the device, it works, after a few minutes, it stopped working.

### Zigbee2MQTT version

2.3.0-1

### Adapter firmware version

20210708

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle

### Setup

Addon on home assistant runnung in a Proxmox VM with a intel NUC N100

### Debug log

Error 2025-05-26 06:30:24z2m: Publish 'set' 'alarm' to 'Sirene Garagenkeller' failed: 'Error: ZCL command 0xa4c13815f4cfe743/1 manuSpecificTuya.dataRequest({"seq":41,"dpValues":[{"dp":13,"datatype":1,"data":[1]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
Info 2025-05-26 06:30:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Rocket Espresso', payload '{"child_lock":"UNLOCK","countdown":0,"current":0.77,"energy":10.79,"indicator_mode":"off/on","linkquality":39,"power":19,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"voltage":218}'

JonMagon, 2025-07-27:
I have the exact same device, and when powered by a USB adapter, there's no such problem. It's probably just a defective unit. The device itself is already a marvel of Chinese engineering, and everyone definitely has the same issue where if you send multiple commands at once, some of them just get ignored.

JonMagon, 2025-07-27:
Is it close enough to the USB dongle?

Gandalf8266, 2025-07-27:
1 m, all in same room

Miwer, 2025-07-27:
Have you considered upgrading to a newer firmware on your coordinator?

Gandalf8266, 2026-09-11:
> I have the exact same problem.
> 
> Pairs fine. works for a minute. Disconnects. Same error log.

I found a solution, I updated my zigbee coordinator to latest version, no it's stable (so far)