Koenkk/zigbee2mqtt #27514: [New device support]:

Idaho947, 2025-05-26:
### Link

https://www.domadoo.fr/fr/produits-zigbee/7815-mumubiz-capteur-de-pluie-zigbee-ewelink-jeedom-home-assistant-zigbee2mqtt.html

### Database entry

{"id":139,"type":"EndDevice","ieeeAddr":"0xa4c1388a42f61a5e","nwkAddr":43430,"manufId":4742,"manufName":"eWeLink","powerSource":"Battery","modelId":"SNZB-05","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SNZB-05","manufacturerName":"eWeLink","powerSource":3,"zclVersion":3,"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x84b4dbfffecc0043","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffecc0043","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffecc0043","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1748253771346,"checkinInterval":3600}

### Zigbee2MQTT version

2.3.0

### Comments

I the a rain sensor see as a leak sensor for now.

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['SNZB-05'],
    model: 'SNZB-05',
    vendor: 'eWeLink',
    description: 'Automatically generated definition',
    extend: [m.battery(), m.iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {},
};
```

### What does/doesn't work with the external definition?

Nothing. The sensor doesn't work

Idaho947, 2025-05-28:
```
info 2025-05-28 17:21:32z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c1388a42f61a5e/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c1388a42f61a5e/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":0,"identifiers":["zigbee2mqtt_0xa4c1388a42f61a5e"],"manufacturer":"eWeLink","model":"Zigbee water sensor","model_id":"SNZB-05","name":"0xa4c1388a42f61a5e","sw_version":"0122052017","via_device":"zigbee2mqtt_bridge_0x84b4dbfffecc0043"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","object_id":"0xa4c1388a42f61a5e_battery","origin":{"name":"Zigbee2MQTT","sw":"2.3.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c1388a42f61a5e","unique_id":"0xa4c1388a42f61a5e_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info 2025-05-28 17:21:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee water sensor","exposes":[{"access":5,"category":"diagnostic","description":"Remaining battery in %","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Indicates whether the device detected a water leak","label":"Water leak","name":"water_leak","property":"water_leak","type":"binary","value_off":false,"value_on":true},{"access":1,"category":"diagnostic","description":"Indicates whether the battery of the device is almost empty","label":"Battery low","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"SNZB-05","options":[],"supports_ota":false,"vendor":"eWeLink"},"friendly_name":"0xa4c1388a42f61a5e","ieee_address":"0xa4c1388a42f61a5e","status":"successful","supported":true},"type":"device_interview"}'
info 2025-05-28 17:21:32z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c1388a42f61a5e/water_leak/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c1388a42f61a5e/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":0,"identifiers":["zigbee2mqtt_0xa4c1388a42f61a5e"],"manufacturer":"eWeLink","model":"Zigbee water sensor","model_id":"SNZB-05","name":"0xa4c1388a42f61a5e","sw_version":"0122052017","via_device":"zigbee2mqtt_bridge_0x84b4dbfffecc0043"},"device_class":"moisture","object_id":"0xa4c1388a42f61a5e_water_leak","origin":{"name":"Zigbee2MQTT","sw":"2.3.0","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0xa4c1388a42f61a5e","unique_id":"0xa4c1388a42f61a5e_water_leak_zigbee2mqtt","value_template":"{{ value_json.water_leak }}"}'
info 2025-05-28 17:21:32z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c1388a42f61a5e/battery_low/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c1388a42f61a5e/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":0,"identifiers":["zigbee2mqtt_0xa4c1388a42f61a5e"],"manufacturer":"eWeLink","model":"Zigbee water sensor","model_id":"SNZB-05","name":"0xa4c1388a42f61a5e","sw_version":"0122052017","via_device":"zigbee2mqtt_bridge_0x84b4dbfffecc0043"},"device_class":"battery","entity_category":"diagnostic","object_id":"0xa4c1388a42f61a5e_battery_low","origin":{"name":"Zigbee2MQTT","sw":"2.3.0","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0xa4c1388a42f61a5e","unique_id":"0xa4c1388a42f61a5e_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low }}"}'
info 2025-05-28 17:21:33z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c1388a42f61a5e/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c1388a42f61a5e/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":0,"identifiers":["zigbee2mqtt_0xa4c1388a42f61a5e"],"manufacturer":"eWeLink","model":"Zigbee water sensor","model_id":"SNZB-05","name":"0xa4c1388a42f61a5e","sw_version":"0122052017","via_device":"zigbee2mqtt_bridge_0x84b4dbfffecc0043"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0xa4c1388a42f61a5e_linkquality","origin":{"name":"Zigbee2MQTT","sw":"2.3.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c1388a42f61a5e","unique_id":"0xa4c1388a42f61a5e_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
```

nmaster2042, 2025-05-29:
Same device can be pourchased on aliexpress: https://fr.aliexpress.com/item/1005009030356992.html?spm=a2g0o.order_list.order_list_main.29.74385e5boRBxHe&gatewayAdapt=glo2fra

bernardesarthur, 2025-05-29:
Even it's recognized as a water leak sensor, is it works fine?

Idaho947, 2025-05-29:
Non the leak sensor stay at false even if it is under water.

waterchill, 2025-06-25:
Also have this device, as a leak sensor it will work and it is reporting leak when you touch it so very sensible. But I need it as rain sensor like the other solar rain sensor.