Koenkk/zigbee2mqtt #27524: IKEA INSPELNING Smart plug: Wrong energy metering

LeoeLeoeL, 2025-05-26:
### What happened?

The device involved is IKEA INSPELNING Smart plug E2206 (Firmware version 2.4.45)
The energy entity (KWh) shows much (but very much) more than the real energy used.

This is the active power for 24 hours
When the load is OFF the power is 0.3-0.4 watt
There is one hour (from 21.40 to 22.40) when the load is ON and is 7.7 watt.

[image]

Now the energy:

[image]

In 24 hours it begins at 424 and ends at 431. They are 7 KWh !!!!!
It's impossible because for 23 hours the load used 0.3-0.4 watt per hour.

### What did you expect to happen?

I expect to have a real energy consumption.

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

8.0.2 (GA)

### Adapter

EmberZNet (Sonoff Dongle E)

### Setup

Container in Proxmox

LeoeLeoeL, 2025-05-30:
Is there anybody else affected by the same issue?

closed 2025-08-24 by github-actions[bot]

Koenkk, 2025-11-25:
@Aljosha89 could you also provide the data/database.db entry of this device?

Aljosha89, 2025-11-26:
> So this would give a current consumption of `775533 / 1000` (`currentSummDelivered` / `divisor`) = 775.5 kWh, I'm not sure if that's realistic?

That calculation results in 775.5 kWh, which is definitely incorrect. The device worked perfectly from January to September (totaling 140 kWh), but since October, it has been erroneously reporting >100 kWh on some days even with a 10W load.

Given that christoph-morrison reported the same issue on 30.09, I suspect this could be related to a recent firmware update or a bug introduced around that date.

REUW, 2026-01-04:
I'm facing the same problem on my INSPELNING Plugs, except the jumps in energy usage are very few but very large.

This is a screenshot of the energy usage from the plug that is monitoring my fridge :

[image]

guardiande, 2026-01-18:
Not sure if I'm missing something here, but I'm facing the same issue that the used energy does not get updated by itself, but if I go to the device page in Z2M -> exposes and then click on the refresh button next to the energy state, the value gets updated.