Koenkk/zigbee2mqtt #27525: haElectricalMeasurement "UNSUPPORTED_ATTRIBUTE" on WS-K07E

modem7, 2025-05-26:
### What happened?

Currently getting the following error coming up.

```
 2025-05-26 23:54:23z2m: Failed to configure 'Alex Office Light Switch', attempt 3 (Error: ZCL command 0x54ef441000ef099e/1 haElectricalMeasurement.read(["acCurrentDivisor","acCurrentMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:363:28) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:987:26) at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@4.0.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:459:29) at result.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.36.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:2066:37) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.36.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21) at Configure.configure (/app/lib/extension/configure.ts:124:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
```

[image]

### How to reproduce it (minimal and precise)

Unsure, this keeps coming up every few seconds. Seems to only have started since latest update, but could be wrong.

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

8.0.2 [GA]

### Adapter

SLZB-06M

### Setup

HAOS add-on.
Proxmox VM.

Koenkk, 2025-05-28:
@kirovilya could you check this?

zavrazhny, 2025-06-04:
I confirm the same error for  WS-K07E and WS-K08E.

winstonm, 2025-08-03:
Spot on. Same issue here. 

HA in NAS, zigbee2mqtt [2.6.0] , with WS-K07E and WS-K08E Aqara H2 switches. 
Did upgraded HA to latest 2025.7.5 . upgraded SLZB-06P7 to latest zigbee version and core.

zigbee-herdsman-converters version
24.11.0
zigbee-herdsman version
5.0.4

Koenkk, 2025-08-04:
Are the values like current and power updating properly?

winstonm, 2025-08-04:
Current is throwing error when we test by refresh icon. Unsupported Attribute 

 Power is fine. Energy is also fine Looks like. But wondering it is working instantaneously though.

Koenkk, 2025-08-05:
Could you provide the data/database.db entry of this device?

leroyloren, 2025-08-05:
`{"id":138,"type":"Router","ieeeAddr":"0x54ef44100110acc9","nwkAddr":17044,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.switch.agl010","epList":[1,2,3,4,5,21],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[2820,1794,18,5,4,6,3,0,64704],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"61441":2,"modelId":"lumi.switch.agl010"}},"manuSpecificLumi":{"attributes":{"4":4,"22":"541C2A8E8251C12877760FB3C1E9E5FCA2A7E0D3838E444FDE1A6F0D117BBC35","223":{"type":"Buffer","data":[40,35,64,0,0,0,3,35,252,152,0,0,4,33,105,23,5,33,64,1,41,35,0,0,0,0,42,35,0,0,0,0,43,33,149,2,44,41,192,255]},"238":4885,"240":0,"247":{"type":"Buffer","data":[3,40,24,5,33,39,0,9,33,0,24,10,33,0,0,13,35,21,19,0,0,100,16,0,101,16,0,149,57,193,240,244,60,152,57,0,0,0,0,151,57,0,0,0,0,155,32,0,156,65,8,0,0,0,0,0,0,0,0]},"512":0,"554":2,"617":1,"645":1,"648":0,"1303":1}},"genOnOff":{"attributes":{"245":37917696,"onOff":0}},"genMultistateInput":{"attributes":{"presentValue":1}},"haElectricalMeasurement":{"attributes":{"acPowerDivisor":10,"acPowerMultiplier":1}}},"binds":[{"cluster":64704,"type":"endpoint","deviceIeeeAddress":"0x00124b00258b7701","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00258b7701","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[18,5,4,6,64704],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"245":37917696,"onOff":0}},"manuSpecificLumi":{"attributes":{"512":0,"617":1,"645":1}},"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[{"cluster":64704,"type":"endpoint","deviceIeeeAddress":"0x00124b00258b7701","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00258b7701","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[64704],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[18,64704],"outClusterList":[],"clusters":{"manuSpecificLumi":{"attributes":{"646":2}},"genMultistateInput":{"attributes":{"presentValue":255}}},"binds":[{"cluster":64704,"type":"endpoint","deviceIeeeAddress":"0x00124b00258b7701","endpointID":1}],"configuredReportings":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":0,"inClusterList":[18,64704],"outClusterList":[],"clusters":{"manuSpecificLumi":{"attributes":{"646":2}}},"binds":[{"cluster":64704,"type":"endpoint","deviceIeeeAddress":"0x00124b00258b7701","endpointID":1}],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":0,"inClusterList":[12],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":21,"stackVersion":27,"hwVersion":1,"dateCode":"20241209","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1754424745158}
`

Koenkk, 2025-08-06:
Pushed a possible fix, let me know if it works.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

leroyloren, 2025-08-12:
Thanks.

It seems that using an external converter for WS-K07E helped and the logs are fine. For WS-K08E (double rocker) the problem persists.

Koenkk, 2025-08-12:
Thanks, integrated the fix! (also for WS-K08E)

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

HobboRobin, 2025-11-22:
so i have had almost the same Problem with the current 2.6.3

But now after readding the device it is not possible to get any binding or reporting configurated.
I tried already to reset the device to factory setting.

When i move the device to a different Z2m that i have running it works there. But adding it back is resulting in the same Problem

tgudlek, 2025-11-22:
> so i have had almost the same Problem with the current 2.6.3
>
> But now after readding the device it is not possible to get any binding or reporting configurated. I tried already to reset the device [...]

I’m still unable to configure binding / reporting of my Aqara devices, but I suspect it’s an Aqara problem. More context: https://github.com/Koenkk/zigbee2mqtt/discussions/28742