Koenkk/zigbee2mqtt #27526: [New device support]: Nova Digital SY-ZB 6W - _TZE284_tdhnhhiy

felipebarbosaa, 2025-05-27:
### Link

https://www.hubsmarthome.com.br/MLB-3204802834-interruptor-zigbee-inteligente-4x4-6-botoes-touch-alexa-_JM#polycard_client=search-nordic-mshops&position=8&search_layout=grid&type=item&tracking_id=0c9a8479-a9bb-46b0-b8cb-4ba914002ed2&wid=MLB3204802834&sid=search

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xa4c138f595458a21","nwkAddr":18911,"manufId":4417,"manufName":"_TZE284_tdhnhhiy","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0010��/i\u0011��/i\u0012��/i\u0013��/i@��/iA��/iA��/iB��/iC��/iD��/iD��/iE��/i","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE284_tdhnhhiy","powerSource":1,"zclVersion":3,"appVersion":78,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":78,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1946894323},"lastSeen":1748323375763}

### Zigbee2MQTT version

 2.3.0-1

### Comments

This device is detected, interview completes successfully, but the exposed controls do not function.
We attempted a custom external converter using datapoints [1–6] mapped to state_l1 through state_l6, using tuya.valueConverter.onOff, but failed with errors like:

Error: No datapoint defined for 'state'
Cannot read properties of undefined (reading 'ID')

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_tdhnhhiy',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

First, I apologize in advance for any language issues—this text was translated using ChatGPT.

I created an external converter based on the official documentation. The device pairs correctly and reports the state of all 6 channels (state_l1 to state_l6) as expected in the MQTT logs. The state updates correctly when using the physical buttons.

However, sending MQTT payloads to change the state does not work. For example:

Topic: zigbee2mqtt/interruptor/set
Payload: { "state_l1": "ON" }

This results in the following error:
Publish 'set' 'state' to 'interruptor' failed: 'Error: No datapoint defined for 'state'

Z2M logs also show:
EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'ID')

Below is the full external converter used:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE284_tdhnhhiy'}],
    model: 'SY-ZB 6W',
    vendor: 'Nova Digital',
    description: 'Interruptor Zigbee de 6 canais',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.switch().withEndpoint('l1'),
        e.switch().withEndpoint('l2'),
        e.switch().withEndpoint('l3'),
        e.switch().withEndpoint('l4'),
        e.switch().withEndpoint('l5'),
        e.switch().withEndpoint('l6'),
    ],
    endpoint: (device) => ({
        l1: 1,
        l2: 2,
        l3: 3,
        l4: 4,
        l5: 5,
        l6: 6,
    }),
    meta: {
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'state_l2', tuya.valueConverter.onOff],
            [3, 'state_l3', tuya.valueConverter.onOff],
            [4, 'state_l4', tuya.valueConverter.onOff],
            [5, 'state_l5', tuya.valueConverter.onOff],
            [6, 'state_l6', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;

Thank you in advance for any help. I’ve been stuck on this for a few days now and I don’t have a Zigbee hub—only the Sonoff Zigbee Dongle-P. I’m also new to all of this.

Koenkk, 2025-05-27:
Doesn't this device have 6 gangs instead of 8?

eduseiden, 2025-05-27:
Hi ! Thanks a lot but I'd like to suggest to add light mode (none, relay, pos) and restart status.
I did this in an external converter. In case it helps, please let me know.

Part of the code for this in the 8-gang:

// Light mode (DP 15)
if (dp === 15) {
const lookup = {0: 'none', 1: 'relay', 2: 'pos'};
const value = tuya.getDataValue(data);
result.light_mode = lookup[value] || 'none';
}
// Restart status (DP 14)
else if (dp === 14) {
const lookup = {0: 'power_off', 1: 'power_on', 2: 'last'};
const value = tuya.getDataValue(data);
result.restart_status = lookup[value] || 'power_off';
}

    return result;

For the 6-gang:

    // Backlight (DP 16)
    if (dp === 16) {
        const value = data[0];
        result.backlight = value === 1 ? 'ON' : 'OFF';
    }

    return result;

And, when creating a Z2M group, even though we select L3 (for instance), it's always identified as endpoint 1.

Thanks in advance !

closed 2025-05-28 by Koenkk