Koenkk/zigbee2mqtt #27536: Mi light sensor (GZCGQ01LM)

MSL-DA, 2025-05-28:
### What happened?

https://www.zigbee2mqtt.io/devices/GZCGQ01LM.html#xiaomi-gzcgq01lm

Battery report N/A
Voltage report N/A

### What did you expect to happen?

% and mV

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

8.0.2

### Adapter

SLZB-06M

### Setup

Docker container

thogens, 2025-05-30:
is that something similar to what I'm facing here? #27538

MSL-DA, 2025-06-01:
Not at all. It's about it reporting unknown on battery. Something that has occurred after version 2.1.1

closed 2025-08-09 by github-actions[bot]

MSL-DA, 2025-08-09:
It is still relevant. GZCGQ11LM shows battery, GZCGQ01LM does not