Koenkk/zigbee2mqtt #27541: [New device support]: Tuya Coswall SM-PW801E

rackstar41, 2025-05-28:
### Link

https://fr.aliexpress.com/i/1005002518419678.html

### Database entry

{"id":63,"type":"Router","ieeeAddr":"0x60a423fffe5e22a1","nwkAddr":59542,"manufId":4098,"manufName":"_TZ3000_rul9yxcc","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_rul9yxcc","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightSwitch":1,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0028f55b96","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0028f55b96","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0028f55b96","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1748461872919}

### Zigbee2MQTT version

2.3.0

### Comments

The plug is discover as a "Tuya TS011F_plug_1"

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['TS011F'],
    model: 'TS011F',
    vendor: '_TZ3000_rul9yxcc',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2}}), m.onOff({"powerOnBehavior":false,"endpointNames":["1","2"]})],
    meta: {"multiEndpoint":true},
};
```

### What does/doesn't work with the external definition?

I can only control (On Off) the EU Plug.
Power consumption and other info stay at "N/A" state
The USB control is unavailable

bernardesarthur, 2025-05-29:
After read this device link I don't think has a power monitoring, but try this converter and check if you can control both circuits (USB is not control each one independent). Let me know if works.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    {
        fingerprint: tuya.fingerprint('TS011F', ['_TZ3000_rul9yxcc']),
        model: 'TS011F',
        vendor: 'Tuya',
        description: '2 gang smart plug with USB',
        endpoint: (device) => {
            return {
                '1': 1,
                '2': 2,
            };
        },
        meta: { multiEndpoint: true },
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
        exposes: [
            e.switch().withEndpoint('1'),
            e.switch().withEndpoint('2'),
        ],
        configure: async (device, coordinatorEndpoint, logger) => {
            await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        },
    },
};

module.exports = definition;
```

bernardesarthur, 2025-05-29:
I don't think that device has a power monitoring and the two USB is both controlled at the same time. 

Try this converter and let me know if this works:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint("TS0601", ["_TZE284_tdhnhhiy"]),
    model: "SYZB-6W",
    vendor: "Nova Digital",
    description: "6 gang switch",
    ota: true,
    extend: [
        tuya.modernExtend.tuyaOnOff({
            backlightModeLowMediumHigh: true,
            childLock: true,
            endpoints: ["l1", "l2"],
        }),
    ],
    endpoint: (device) => {
        return { l1: 1, l2: 2 };
    },
    meta: {
        multiEndpoint: true,
        multiEndpointSkip: ["power_on_behavior"],
    },
    configure: tuya.configureMagicPacket,
};

module.exports = definition;`

bernardesarthur, 2025-05-29:
I don't think that device has a power monitoring and the two USB is both controlled at the same time.

Try this converter and let me know if this works:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint("TS011F", ["_TZ3000_rul9yxcc"]),
    model: "TS011F_plug_with_two_gang",
    vendor: "Tuya",
    description: "2 gang switch",
    ota: true,
    extend: [
        tuya.modernExtend.tuyaOnOff({
            backlightModeLowMediumHigh: true,
            childLock: true,
            endpoints: ["l1", "l2"],
        }),
    ],
    endpoint: (device) => {
        return { l1: 1, l2: 2 };
    },
    meta: {
        multiEndpoint: true,
        multiEndpointSkip: ["power_on_behavior"],
    },
    configure: tuya.configureMagicPacket,
};

module.exports = definition;
```

rackstar41, 2025-05-29:
> I don't think that device has a power monitoring and the two USB is both controlled at the same time.
>
> Try this converter and let me know if this works:
>
> ```
> const fz = [...]

Thanks,

I'm out of home for few days.
I will try it at the end of week and I will let you know.

rackstar41, 2025-06-04:
Hello,

I did the test, and it's work perfectly !
I juste remove "backlightModeLowMediumHigh: true," because it's seems having no effect.

Finally, the external converter is :
```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint("TS011F", ["_TZ3000_rul9yxcc"]),
    model: "SM-PW801EZ",
    vendor: "Tuya",
    description: "2 gang switch",
    ota: true,
    extend: [
        tuya.modernExtend.tuyaOnOff({
            childLock: true,
            endpoints: ["l1", "l2"],
        }),
    ],
    endpoint: (device) => {
        return { l1: 1, l2: 2 };
    },
    meta: {
        multiEndpoint: true,
        multiEndpointSkip: ["power_on_behavior"],
    },
    configure: tuya.configureMagicPacket,
};

module.exports = definition;
```

Big thanks

I will know try to submit new devices !

bernardesarthur, 2025-06-04:
I'm gonna add this.

rackstar41, 2025-08-31:
Hello,

I reopen

bernardesarthur, 2025-08-31:
Hey Rackstar, sure I'm gonna help you. At this time I can't but asap I'll call you back.

closed 2025-09-02 by Koenkk