Koenkk/zigbee2mqtt #27545: [Device support request] Yale YSD100 Zigbee Lock

Battosuka, 2025-05-28:
Hello Zigbee2MQTT team,

I'd like to request support for the Yale YSD100 Zigbee smart lock. The device pairs and completes the interview process correctly, but it's currently marked as unsupported. It exposes multiple features, including locking state, pin management, and user access controls. Thanks.

Full logs, interview data, and current state are included below.

Device model: YSD100  
Device manufacturer: Yale  
Device type: Lock  
Zigbee2MQTT version: 2.3.0  
Coordinator: Sonoff ZBDongle-P  
Coordinator firmware: ZStack3x0 20210708  
Device firmware: 02.02.09

---

### Description

Request to add support for the Yale YSD100 smart lock. The device pairs and interviews successfully, and even exposes multiple properties, but is currently marked as `supported: false` and not usable by default.

---

### Logs

From `log.log` during pairing:

[2025-05-28 19:20:01] warning: z2m: Device '0x00158d00013c721f' with Zigbee model 'YSD100' and manufacturer name 'Yale' is NOT supported

From `state.json`:
```json
{
  "0x00158d00013c721f": {
    "state": "UNLOCK",
    "lock_state": "not_fully_locked",
    "linkquality": 36
  }
}

Interview Output
Definition reported by Z2M after interview:

{
  "model": "YSD100",
  "vendor": "Yale",
  "description": "Automatically generated definition",
  "supports_ota": false,
  "exposes": [
    {
      "type": "lock",
      "features": [
        {
          "name": "state",
          "property": "state",
          "access": 7,
          "type": "binary",
          "value_on": "LOCK",
          "value_off": "UNLOCK"
        },
        {
          "name": "lock_state",
          "property": "lock_state",
          "access": 1,
          "type": "enum",
          "values": ["not_fully_locked", "locked", "unlocked"]
        }
      ]
    },
    {
      "type": "composite",
      "property": "pin_code",
      "features": [
        {
          "name": "user",
          "property": "user",
          "type": "numeric",
          "access": 2
        },
        {
          "name": "user_type",
          "property": "user_type",
          "type": "enum",
          "access": 2,
          "values": ["unrestricted", "year_day_schedule", "week_day_schedule", "master", "non_access"]
        },
        {
          "name": "user_enabled",
          "property": "user_enabled",
          "type": "binary",
          "access": 2,
          "value_on": true,
          "value_off": false
        },
        {
          "name": "pin_code",
          "property": "pin_code",
          "type": "numeric",
          "access": 2
        }
      ]
    },
    {
      "name": "action_source_name",
      "property": "action_source_name",
      "type": "enum",
      "access": 1,
      "values": ["keypad", "rfid", "manual", "rf"]
    },
    {
      "name": "action_user",
      "property": "action_user",
      "type": "numeric",
      "access": 1
    },
    {
      "name": "auto_relock_time",
      "property": "auto_relock_time",
      "type": "numeric",
      "access": 7,
      "unit": "s",
      "value_min": 0,
      "value_max": 3600
    },
    {
      "name": "sound_volume",
      "property": "sound_volume",
      "type": "enum",
      "access": 7,
      "values": ["silent_mode", "low_volume", "high_volume"]
    },
    {
      "name": "action",
      "property": "action",
      "type": "enum",
      "access": 1,
      "values": [
        "unknown", "lock", "unlock",
        "lock_failure_invalid_pin_or_id", "lock_failure_invalid_schedule",
        "unlock_failure_invalid_pin_or_id", "unlock_failure_invalid_schedule",
        "one_touch_lock", "key_lock", "key_unlock", "auto_lock",
        "schedule_lock", "schedule_unlock", "manual_lock",
        "manual_unlock", "non_access_user_operational_event"
      ]
    },
    {
      "name": "linkquality",
      "property": "linkquality",
      "type": "numeric",
      "access": 1,
      "unit": "lqi",
      "value_min": 0,
      "value_max": 255
    }
  ]
}

Notes
The device exposes a rich set of functionalities, and appears to use standard Zigbee lock and user management clusters.
Willing to test custom converters and provide further debug logs.
Device appears as functional under MQTT with basic lock/unlock state changes.

Please let me know how to assist further in adding support.

Battosuka, 2025-05-30:
I'm running Zigbee2MQTT as an official Home Assistant OS add-on, and my system runs in a virtual machine. I don’t have direct file system access to the add-on's internal storage.

I’ve searched under `/config/zigbee2mqtt/` using File Editor and terminal, and there is no `database.db` present or exposed by the add-on. It seems the add-on is handling all device data dynamically via `state.json` and MQTT.

Unfortunately, I'm unable to access the internal database.db due to the Home Assistant OS VM environment.

I’ve already shared the full interview and exposes JSON block from the logs. Let me know if I can extract anything else that would be helpful.

Dhannte, 2026-11-16:
Was this solved? trying to connect mine.