Koenkk/zigbee2mqtt #27549: [New device support]: ZP-LZ-EU2U

JorgeLion93, 2025-05-29:
### Link

https://moeshouse.com/products/zigbee3-0-dual-usb-wireless-socket?variant=40015338307665

### Database entry

{"id":17,"type":"Router","ieeeAddr":"0xa4c138a376173be1","nwkAddr":386,"manufId":4417,"manufName":"_TZ3210_2uk4z8ce","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�)\u0000\u0000\u0012","65506":54,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3210_2uk4z8ce","zclVersion":3,"appVersion":192,"modelId":"TS011F","powerSource":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":0,"activePower":0,"rmsCurrent":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":0}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":2,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":0,"switchType":0}},"genIdentify":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002b4849e6","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b002b4849e6","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b002b4849e6","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":257}],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1736549749479}

### Zigbee2MQTT version

2.3.0 (Haos)

### Comments

**WHAT HAPPENED ** -

I added Moes ZP-LZ-EU2U via z2m and it has been recognized as TSO11F
and it shows up in home assistant as a regular socket and the two usb ports don't even show up so they end up being useless

WHAT DID YOU EXPECT TO HAPPEN -

on z2m add it as Moes ZP-LZ-EU2U and expose the socket and 2 other usb ports
and have control over all the features of this product [1 main socket and 2 additional usb ports]

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['TS011F'],
    model: 'TS011F',
    vendor: '_TZ3210_2uk4z8ce',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2}}), m.onOff({"powerOnBehavior":false,"endpointNames":["1","2"]}), m.electricityMeter()],
    meta: {"multiEndpoint":true},
};
```

### What does/doesn't work with the external definition?

Only the main plug

bernardesarthur, 2025-05-29:
https://github.com/Koenkk/zigbee2mqtt/issues/27541#issuecomment-2919757535

Can you try this one?

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint("TS011F", ["_TZ3210_2uk4z8ce"]),
    model: "TS011F_plug_with_two_gang",
    vendor: "Tuya",
    description: "2 gang switch",
    ota: true,
    extend: [
        tuya.modernExtend.tuyaOnOff({
            backlightModeLowMediumHigh: true,
            childLock: true,
            endpoints: ["l1", "l2"],
        }),
    ],
    endpoint: (device) => {
        return { l1: 1, l2: 2 };
    },
    meta: {
        multiEndpoint: true,
        multiEndpointSkip: ["power_on_behavior"],
    },
    configure: tuya.configureMagicPacket,
};

module.exports = definition;
```

bernardesarthur, 2025-05-29:
Do you have the same device? Good, I'll add later into a repo.

bernardesarthur, 2025-05-29:
> [@bernardesarthur](https://github.com/bernardesarthur) IT WORKS!! Thank you very much!

Do you have the same device? Good, I'll add later into a repo. Could you send me a print from exposes for I can check something?

JorgeLion93, 2025-05-29:
> [@bernardesarthur](https://github.com/bernardesarthur) IT WORKS!! Thank you very much!

At HomeAssistan? How did you do it?

bernardesarthur, 2025-05-29:
> > > [@bernardesarthur](https://github.com/bernardesarthur) IT WORKS!! Thank you very much!
> > 
> > Do you have the same device? Good, I'll add later into a repo. Could you send me a print from exposes for I can check something?
> 
> @bernardesarthur here you go:
> 
> [image]

Thanks, I'll add and this is gonna help @JorgeLion93.

bernardesarthur, 2025-05-30:
> @bernardesarthur these are my findings:
> - Child Lock option works, but clicking the physical button 3 times in a row override the lock and turn the switch on/off anyway (I guess this is how the device is designed)
> - Backlight mode is not low/medium/high, but instead it changes de color of the light from blue to red with the low and high options. The medium options doesn´t seem to do anything.
> - I realised turning the socket on/of by clicking the physical button doesn´t update the on/off state, not sure if it´s a design issue or some problem with the converter
> - Power-on behavior works completely fine

Thanks for the info, @heisenberg2980. Now, remove the converter and reinterview that and send me the exposes.

heisenberg2980, 2025-05-30:
@bernardesarthur this is without the external converter:

[image]

bernardesarthur, 2025-05-30:
Thanks @heisenberg2980, I can work from here. Today is gonna be rush, but when I can do this I'll be back.

JorgeLion93, 2025-05-30:
[image]
[image]

JorgeLion93, 2025-05-30:
> [@JorgeLion93](https://github.com/JorgeLion93) you can check if the external converted is being loaded in the z2m logs:
> 
> [image]
> 
> And also double check that your device is the correct model in the "about" screen :
> 
> [image]
> 
> And remember to restart Z2M after making any changes to the file, so Z2M loads the new file.

The device is the right one, but the converter does not charge.

closed 2025-08-07 by github-actions[bot]