Koenkk/zigbee2mqtt #27552: [New device support]: Розумний водо/газовий клапан _TZ3000_cmcjbqup

Valdis2411, 2025-05-29:
### Link

https://www.aliexpress.com/item/1005006913632540.html?spm=a2g0o.order_list.order_list_main.4.2d421802F2LKRn

### Database entry

{"id":56,"type":"Маршрутизатор","ieeeAddr":"0xa4c1383d06591772","nwkAddr":44240,"manufId":4417,"manufName":"_TZ3000_cmcjbqup","powerSource":"Мережеве живлення (один фаза)","modelId":"TS0001","epList":[1,242],"кінцеві точки":{"1":{"profId":260","epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"кластери":{"57344":{"атрибути":{"53251":"AAAA"}},"genBasic":{"атрибути":{"65503":"ü�/\u0013ļ�/\u0012","65506":56,"65 508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_cmcjbqup","zclVersion":3,"appVersion":85,"modelId":"TS0001","powe rSource":1,"hwVersion":1}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBac klightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x9035eafffe74e88e","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterL ist":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"УСПІШНО","meta":{"configured":332242049},"lastSeen":1748534672200}

### Zigbee2MQTT version

2.3.0 

### Comments

при підключенні [image]

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['TS0001'],
    model: 'TS0001',
    vendor: '_TZ3000_cmcjbqup',
    description: 'Automatically generated definition',
    extend: [m.onOff({"powerOnBehavior":false})],
    meta: {},
};
```

### What does/doesn't work with the external definition?

"indicator_mode": "off/on"

bernardesarthur, 2025-05-29:
Are you sure that's your valve has a indicator light?

Valdis2411, 2025-05-30:
[image] це такий же клапан, тільки Виробник Zigbee   _TZ3000_5ucujjts

mrmaximas, 2025-05-30:
@Valdis2411 To help you faster to write better in English. I know your language 🙌🏻, but I write in English. In general, there is no problem - you/we need to add new unknown device to the zigbee-herdsman converters for correctly detection.

Valdis2411, 2025-05-30:
Good afternoon! What do I need to do to make the valve work properly. I"m just starting to use Homa assistant

closed 2025-08-07 by github-actions[bot]