Koenkk/zigbee2mqtt #27556: Dev branch SNZB-02WD/SNZB-02D - No battery

stickpin, 2025-05-30:
### What happened?

Hi,

After changing to the dev branch, the battery is no longer reported for SNZB-02WD/SNZB-02D devices.
The rest seems to be ok. If I roll back to the 2.3.0 battery is present for SNZB-02D, but SNZB-02WD is obviously not supported. 

**SNZB-02WD:**
[image]
[image]

**SNZB-02D**
[image]
[image]

Any idea?

### Zigbee2MQTT version

2.3.0-dev commit: f5920fd5

### Adapter firmware version

8.0.2 [GA]

### Adapter

SLZB-06Mg24

### Setup

Add-on in the Home Assistance

### Debug log

```
[2025-05-30 12:38:39] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffe59e5d 9/battery/config', payload ''
[2025-05-30 12:38:39] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffe599b5f/battery/config', payload ''
[2025-05-30 12:38:39] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffe79ece0/battery/config', payload ''
[2025-05-30 12:38:39] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffe79f22e/battery/config', payload ''
[2025-05-30 12:38:39] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffe7a0025/battery/config', payload ''
[2025-05-30 12:38:39] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffe7a0891/battery/config', payload ''
[2025-05-30 12:38:39] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0ceff6fffe79e783/battery/config', payload ''
[2025-05-30 12:38:39] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c138019e75ffff/battery/config', payload ''
```

Koenkk, 2025-05-30:
Could you provide the debug log when configuring this device (yellow refresh button) on both latest dev and 2.3.0?

chris-1243, 2025-05-30:
@Koenkk SNZB-02WD is supported in dev branch since PR
https://github.com/Koenkk/zigbee-herdsman-converters/pull/9336

It should be available in next official release coming soon.... it depends on you

stickpin, 2025-05-30:
@Koenkk, thanks for the quick reply.

Here are the logs while reconfiguring SNZB-02D:
[log_debug_dev.txt](https://github.com/user-attachments/files/20527363/log_debug_dev.txt)
[log_debug_2.3.0.txt](https://github.com/user-attachments/files/20527364/log_debug_2.3.0.txt)

1. Thanks a lot in advance! :)

Koenkk, 2025-05-31:
Fixed, thanks for reporting!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-05-31 by Koenkk

chris-1243, 2025-05-31:
Fixed in latest-dev

2.3.0-dev commit: 711b456b068ef86c1add5d9cc16770f20a495622

Thanks

stickpin, 2025-06-05:
@maia sorry, you cannot change the SNZB-02WD update interval in Zigbee2MQTT.
But from my personal experience, I would suggest getting or building yourself an ESP32-based temp sensor, if you really need a real-time temp measurement.

maia, 2025-06-08:
> [@maia](https://github.com/maia) sorry, you cannot change the SNZB-02WD update interval in Zigbee2MQTT. But from my personal experience, I would suggest getting or building yourself an ESP32-based temp sensor, if you really need a real-time temp measurement.

Thanks, that’s what I feared. So cluster commands are ignored by the device?

Someone [mentioned here](https://smarthomescene.com/reviews/sonoff-temperature-and-humidity-snzb-02wd-snzb-02ld-review/#comment-71824) that it should be possible to change the SNZB-02WD reporting interval in Zigbee2MQTT:

> "In Zigbee2MQTT, see Reporting tab of the device. I might not always work to set it, only when it is awake. So I usually blow onto the sensor whilst doing this. For some devices you’ll have to set this during the 60s it has for pairing. You need to Apply for each row! “Min rep change” is value / 100. Eg. Set to 100 for 0.1º or 10 for 1% humidity change. Intervals are in seconds."

The review on the same page also says:

> "You would need to send cluster commands in order to adjust the comfort indicators, change reporting intervals and calibrate the readings."

As for the ESP32 sensor idea: I’m really just looking for a battery-powered room temperature display that sends updates to Home Assistant more frequently than every 5 minutes. I’m not confident with DIY builds — from selecting parts to coding and 3D-printing a case.