Koenkk/zigbee2mqtt #27558: Two Philips Hue products incorrectly assigned

baby0da99, 2025-05-30:
### What happened?

I added three 929003777201 Hue Being ceiling lights ([hue link](https://www.philips-hue.com/en-us/p/hue-white-ambiance-being-ceiling-light-white/046677588304)), white which come up as "unsupported". However, they are exactly the same product as the supported 3261031P6 model, except one is the US version (929003777201) and the other one appears to be the EU version (3261031P6). All features work normally under Exposes - I can run the lights on and off, adjust colors etc.

Same issue with adding a 929003817101 Hue Solo lightstrip 33 feet / 10 meter [hue link](https://www.philips-hue.com/en-us/p/lightstrips-solo-lightstrip-33-feet/046677589936), also comes up as "unsupported" and is recognized as model LCL010 which doesn't appear anywhere on any Philips Hue products lists. 929003817002 (16 feet / 5 meter) is supported, same product just shorter. Here, as well, all features work normally under Exposes - I can run the lights on and off, adjust colors etc.

### Zigbee2MQTT version

2.3.0-1

### Adapter firmware version

OS: v2.8.5.dev12, Zigbee: 20250321

### Adapter

SLZB-06

### Setup

HAOS ran in Proxmox VM, Zigbee2MQTT and Mosquitto broker as add-ons, MQTT as integration

chris-1243, 2025-06-01:
Your lights will be available in the latest-dev or in the July release (for the June release, it is too tight....) when accepted and merged by the dev team

chris-1243, 2025-06-01:
@baby0da99 

Could you close the issue. The PR has been accepted.

Thanks