Koenkk/zigbee2mqtt #27560: zigbee coordinator 20250325 update on slzb-06 broke my entire network

myk3y55, 2025-05-30:
### What happened?

z2m would not start after updating slzb-06 to zigbee coordinator 20250325 with core 2.8.5 dev12 already installed.

after setting pan_id: GENERATE z2m starts again but all devices are gone (list is empty in ui)

i looked into coordinator_backup.json and all devices were gone. entry looked like this: "devices": []

after i copy/paste the configuration.yaml, coordinator_backup and database.db from a previous backup, enter the new pan_id in configuration.yaml and downgrading back to 20250321 i can now see all devices but all show offline and cannot be controlled (logs show that device cannot be pinged).

z2m or ha restores from previous backups doesn't bring devices back and deletes devices from coordinator_backup.json

### What did you expect to happen?

have all devices continue to work as usual after the update

### How to reproduce it (minimal and precise)

not sure, but it hapenned immediately after coordinator firmware update

### Zigbee2MQTT version

2.3.0-1

### Adapter firmware version

20250325

### Adapter

SMLIGHT SLZB-06

### Setup

HA os in proxmox VM

### Debug log

[image]
[image]
[image]

danir-de, 2025-05-30:
Related to #27553.

myk3y55, 2025-05-31:
i saw the issue you mentioned earlier when i was trying to figure out a way to fix this without re-pairing all devices. 
as an update, i had to write GENERATE on all 3 id's for z2m to work again. 
issue is... i now have to re-pair all 165 devices in my network... luckily i managed to copy/paste all devices back in coordinator_backup.json and database.db so i don't have to rename every device back for the automations to work again!

did you figure out a way to have all devices working without re-pairing? thanks

myk3y55, 2025-05-31:
i already updated to 2.8.7 and latest zigbee before starting to re-pair everything and enabled "disable mdns" and the coordinator connects to z2m, but i already started the re-pair process and can't go back to the previous configs.

i had to re-pair at least some devices to not annoy my so :) but at least i hope this solves the issue for others! thanks for the suggestion!

XXBrain, 2025-05-31:
> I also did 20250325 firmware upgrade on my SMLIGHT SLZB-06, but I was able to stay on the new firmware version by simply doing a couple of things:
> 
>     1. Upon receiving the 'networking commissioning time out - most likely network with the same panId....' I stopped z2m and removed the antenna from my SLZB-06. It was positioned just outside a structured media panel made out of metal, so I stuck it in there for good measure. I assume you could turn off all your zigbee router devices and that would accomplish the same thing. Anyway, I started z2m again, and it was able to load, but there were no devices present.
> 
>     2. I then restored the config/zigbee2mqtt folder from the previous night.  I started Z2M and had all my devices, and they were controllable.  Battery-operated devices initially showed as offline, but they became online when activated.
> 
> 
> Currently, everything appears to be working fine, although I have not tested all 155 of my devices. And I haven't activated all my battery-operated devices. So I could still have issues. And I am running on SLZB OS Firmware v2.8.7 and Zigbee Controller Revision 20250325.
> 
> Before starting step 1, you can downgrade back to the previous firmware and then follow the above steps and that will also work if you don't want to stay on 20250325.

Thank you so much. 

I am not sure if downgrading is necessary but power off the routers did the trick. I did this only for the room with the coordinator and removed the antennas. (MR1). After that it started but I saw now devices., I somehow had an empty database but this could be my own fault from copy pasting while searching for a solution. I restored the database.db from before and it came back to live with all devices.

ruderthanyou, 2025-05-31:
No, you definitely don't need to downgrade if you don't want to.

While I was troubleshooting the first time it broke, I downgraded to 20250321 and then figured out the steps above. I was able to get everything working again on 20250321. But I'm a gluten for punishment, so I upgraded again back to 20250325, and again got an error.  It was a bit different, 'Error: target adapter aps link key data table size insufficient (size=3)'.  I found a link that said to delete the /config/zigbee2mqtt/coordinator_backup.json file and restart z2m.  Did that and got back to the 'networking commissioning error' which I then unplugged the antenna, and followed the rest of my steps above. Everything is working fine again.

Anyway, it's the next day, and everything is still working okay.

markus5790, 2025-06-01:
I was able to restore my zigbee Network with These instructions but after a while the whole Network collapses again. Could someone please reopen this issue so that the rootcause is understood and fixed?

domoticauser, 2025-06-01:
Is it safe to update without errors?

domoticauser, 2025-06-01:
> My findings are:
> 
>     1. Backup your Zigbee2MQQT folder prior uodating. You will need the Database.db File to restore your configuration in case your Network is broken after the Update
> 
>     2. It is safe to Update the coordinator Firmware to v2.8.7
> 
>     3. stay Zigbee Version 20250321, don't Update to 20250325
>        In case your Network Breaks: Stopp Zigbee2MQQT add-on, Power Off all Routers and the coordinator, restore your database.db from Backup, Power the coordinator and start the Zigbee2Mqqt add-on. It should Run without issues. Power on your routers

Thanks for your respond!

Tthe only update i did in the past, was the firmware, not the zigbee. 
The coordinator version is at this moment, Radio 20240716 ,  kern v2.7.1
ZIgbee2mqtt, 2.3.0

20250325 is only for test purposes.

myk3y55, 2025-06-02:
I genuinely missed the testing part there.. I flashed 20240710 (this is the latest public, non-testing version available in my ui) and everything went well so far.

markus-barta, 2025-06-05:
Two days stable now. Did upgrade to 2.4 and kept transmit_power: 10.

Zigbee2MQTT version: 2.4.0
Coordinator type: ZStack3x0
Coordinator revision: 20250325
Frontend version: 0.9.13
zigbee-herdsman-converters version: 23.53.0
zigbee-herdsman version: 4.1.0

myk3y55, 2025-06-07:
I just tried the update in a second location (same setup exept this one is an intel nuc vs proxmox in location 1) and it behaves the same way my first location did.. so I downgraded everything back (z2m, zigbee and core) to the way it was including all files from the z2m folder and still no dice... i get this error: 

z2m: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby (Error: AREQ - ZDO - stateChangeInd after 60000ms

Now z2m won't even start no matter what z2m-core-zigbee firmware version combination i try...

it looks like after updating, i have no choice but to regenerate pan id and re-pair everything in my network again..

I think one of these updates does something permanent to the coordinator and you just can't go back.. Now my entire zigbee network in location 2 is broken, the same way the fist one was..

p.s.: yes, i tried everything suggested above, but nothing worked.. it looks like i have to redo this one from zero as well...

myk3y55, 2025-06-07:
Update: i finally managed to make it work using @ruderthanyou 's workaround, but with a slight difference: i had to cut power to the coordinator before unplugging the antena and again before plugging the antenna back in, otherwise the same steps.

I don't know why it wouldn't do it without cutting power to the coordinator but this worked for me

Koenkk, 2025-06-19:
The `20250325` firmware is not an official Zigbee2MQTT firmware (it's made by the smlight team). Please open an issue on there side, if you want to use an official firmware use [20250521](https://github.com/Koenkk/Z-Stack-firmware/discussions/545)

closed 2025-06-19 by Koenkk

1aranzant, 2025-11-14:
just got the same issue while update SLZB06's core firmware + zigbee firmware... 
LOST everything...