Koenkk/zigbee2mqtt #27580: Frequent Z2MQTT crashes since updating to 2.40

fifibest, 2025-06-01:
### What happened?

I am noticing frequent restarts of zigbee2mqtt since updating to 2.4.0:

[2025-06-02 00:39:26] info: 	z2m: Disconnecting from MQTT server
[2025-06-02 00:41:07] info: 	z2m: Disconnecting from MQTT server
[2025-06-02 00:42:48] info: 	z2m: Disconnecting from MQTT server
[2025-06-02 00:44:29] info: 	z2m: Disconnecting from MQTT server
[2025-06-02 00:46:10] info: 	z2m: Disconnecting from MQTT server
[2025-06-02 00:47:51] info: 	z2m: Disconnecting from MQTT server
[2025-06-02 00:49:32] info: 	z2m: Disconnecting from MQTT server

This is what I see in the logs:

[2025-06-02 00:49:33] info: 	zh:controller: Wrote coordinator backup to '/opt/zigbee2mqtt/data/coordinator_backup.json'
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   Total frames: RX=3787, TX=4886
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   DATA frames : RX=3678, TX=1148
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   DATA bytes  : RX=110150, TX=30288
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   Retry frames: RX=9, TX=51
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   ACK frames  : RX=49, TX=3686
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   NAK frames  : RX=50, TX=0
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   Retry dupes     : RX=9
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=2
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash: Port closed.
[2025-06-02 00:49:33] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2025-06-02 00:49:33] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2025-06-02 00:49:33] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2025-06-02 00:49:33] info: 	z2m: Stopped zigbee-herdsman
[2025-06-02 00:49:33] info: 	z2m: Stopped Zigbee2MQTT

It was working quite stable with 2.3.0.

### How to reproduce it (minimal and precise)

It just happens continuously.

### Zigbee2MQTT version

2.40.0

### Adapter firmware version

8.0.2

### Adapter

Sonoff ZBDongle

### Setup

Z2M plain running on RPI 4+ with Homebridge.

### Debug log

[debug.log.zip](https://github.com/user-attachments/files/20542949/debug.log.zip)

fifibest, 2025-06-02:
That is the only issue that I can see in the logs, this happens from time to time but not directly before the restart. How can I fix this somehow? Is this a problem with my installation?

```
[2025-06-02 00:48:09] error: 	z2m: Could not init sd_notify: The module '/opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/Release/unix_dgram.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 127. This version of Node.js requires
NODE_MODULE_VERSION 115. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
[2025-06-02 00:48:09] debug: 	z2m: Error: The module '/opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/Release/unix_dgram.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 127. This version of Node.js requires
NODE_MODULE_VERSION 115. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1651:18)
    at Module.load (node:internal/modules/cjs/loader:1275:32)
    at Function.Module._load (node:internal/modules/cjs/loader:1096:12)
    at Module.require (node:internal/modules/cjs/loader:1298:19)
    at require (node:internal/modules/helpers:182:18)
    at bindings (/opt/zigbee2mqtt/node_modules/.pnpm/bindings@1.5.0/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/lib/unix_dgram.js:4:34)
    at Module._compile (node:internal/modules/cjs/loader:1529:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1613:10)
    at Module.load (node:internal/modules/cjs/loader:1275:32)
```

Also, how can I revert back to 2.3.0? This issue makes my setup practically unusable, due to the frequent restarts lights / switches do not get the commands, automations are not working.

ericvanbeurden, 2025-06-03:
For me it keep crashing as well, in most cases it won't start.

Nerivec, 2025-06-03:
Try removing the `node_modules` folder in your Z2M install folder and re-install packages:
```bash
rm -rf node_modules
pnpm i --frozen-lockfile
```
If it still does not work, run `pnpm rebuild unix-dgram`

fifibest, 2025-06-03:
thanks, I did the above steps, still same issue happens, and it did not even fix the unix-gram issue:

```
Jun 03 19:28:13 homebridge systemd[1]: zigbee2mqtt.service: start operation timed out. Terminating.
Jun 03 19:28:14 homebridge systemd[1]: zigbee2mqtt.service: Failed with result 'timeout'.
Jun 03 19:28:14 homebridge systemd[1]: Failed to start zigbee2mqtt.service - zigbee2mqtt.
Jun 03 19:28:14 homebridge systemd[1]: zigbee2mqtt.service: Consumed 16.264s CPU time.
Jun 03 19:28:24 homebridge systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 1.
Jun 03 19:28:24 homebridge systemd[1]: Stopped zigbee2mqtt.service - zigbee2mqtt.
Jun 03 19:28:24 homebridge systemd[1]: zigbee2mqtt.service: Consumed 16.264s CPU time.
Jun 03 19:28:24 homebridge systemd[1]: Starting zigbee2mqtt.service - zigbee2mqtt...
Jun 03 19:28:24 homebridge node[95109]: Starting Zigbee2MQTT without watchdog.
Jun 03 19:28:31 homebridge node[95109]: [2025-06-03 19:28:31] error:         z2m: Could not init sd_notify: The module '/opt/zigbee2mqtt/node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram/build/Release/unix_dgram.node'
Jun 03 19:28:31 homebridge node[95109]: was compiled against a different Node.js version using
Jun 03 19:28:31 homebridge node[95109]: NODE_MODULE_VERSION 127. This version of Node.js requires
Jun 03 19:28:31 homebridge node[95109]: NODE_MODULE_VERSION 115. Please try re-compiling or re-installing
Jun 03 19:28:31 homebridge node[95109]: the module (for instance, using `npm rebuild` or `npm install`).
```
can someone look into the debug logs? how can I move back to 2.3.0? I had no issue there and this version is unusable for me.

fifibest, 2025-06-03:
New debug attached:

[debug.txt.zip](https://github.com/user-attachments/files/20578486/debug.txt.zip)

Nerivec, 2025-06-03:
About the module issue, this is a bit strange. Can you make sure the nodejs version used by Homebridge is the same you use when you install zigbee2mqtt?
Can you post the output of the `pnpm rebuild unix-dgram` command?

These restarts do not look like they are from errors, they look more like something is triggering a restart explicitly.
Any chance you have an automation that could be doing this?
Otherwise, do you see something in the Homebridge logs that could point to a reason? _I don't know Homebridge at all, can't help much here._

You can clone any version of Z2M by using the `-b` param with one of the [Z2M tags](https://github.com/Koenkk/zigbee2mqtt/tags). For example:
```bash
git clone -b 2.3.0 https://github.com/Koenkk/zigbee2mqtt
```
_Not sure if Homebridge has any kind of auto-install or if you can manage versions, or if it's a plain clone you do yourself..._

fifibest, 2025-06-03:
Module issue:
```
/opt/zigbee2mqtt $ sudo pnpm rebuild unix-dgram
node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram: Running install script, done in 6.6s
```

The strange thing that everything is exactly the same, nothing was added or modified on the homebridge side, the only change is the updating of zigbee2mqtt. Nothing is visible in the home bridge logs, and nothing is triggering any action in that time, so I seriously doubt that it would be a homebridge issue. Also, this issue started as soon as 2.4.0 started up. 

Example, it's standard colour info update, and it's 1 minute before the restart is triggered, nothing is visible.

```
[03/06/2025, 20:35:35] [homebridge-z2m] adaptive_lighting: Kert Biloba: skipped color_temp (new: 437; old: 437)
[03/06/2025, 20:35:35] [homebridge-z2m] Handled device update for Kert Diszvilagitas: {"brightness":25,"color":{"hue":26,"saturation":93,"x":0.5149,"y":0.4147},"color_mode":"color_temp","color_temp":472,"level_config":{"on_level":254},"linkquality":48,"state":"ON","update":{"installed_version":17196032,"latest_version":17196032,"state":"idle"}}
[03/06/2025, 20:36:23] [homebridge-z2m] Zigbee2MQTT is ONLINE
```

I did not manage to go back to the previous 2.3.0, I am not sure besides this command what else is needed.

Nerivec, 2025-06-03:
Somehow the dgram module gets compiled with node v20, but Homebridge is using node v22 (based on the logs). I just don't know Homebridge, so hard to say what could be going on, and why it's showing up like this (background update?).

fifibest, 2025-06-05:
I reinstalled my whole system, both the gram modul install and the constant restarts have been resolved!

Thanks for your help, looks like there was some glitch on my side.

closed 2025-06-05 by Koenkk