Koenkk/zigbee2mqtt #27591: TS0601 Dual Water Valve not working correct

scobby, 2025-06-02:
### What happened?

The Device https://www.zigbee2mqtt.io/devices/TS0601_water_switch.html is interviewed correctly but it doesnt work:

Version:  2.4.0-1

- Battery not showing
- Voltage not showing 
- Valve state only switches when manual pushing the button on the device
- Switching the switch in Zigbee2mqtt not working
- Countdown und valve duration are unknown

Zigbee Model TS0601
Zigbee Manufacturer _TZE284_fhvpaltk

Bug exists also in Edge Addon of Zigbee2mqtt (dev branch)

how can i support?

### What did you expect to happen?

Data Visible and switching possible

### Zigbee2MQTT version

2.4.0

### Adapter firmware version

20230507

### Adapter

Sonoff Dongle-p

### Setup

Home Assistant Zigbee2mqtt Stable Addon

scobby, 2025-06-02:
[image]

chris-1243, 2025-06-02:
You may have a look here

https://github.com/Koenkk/zigbee-herdsman-converters/pull/8898