Koenkk/zigbee2mqtt #27604: All LQI are 255 in Device list Z2M

jocamane, 2025-06-02:
### What happened?

If you check the map, several devices appear where the LQI is not 255, however on the devices page ALL devices have LQI 255.
How to solve this?

[image]

[image]

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

0x26780700

### Adapter

 ConBee2

### Setup

HA with HAOS in RPI 4b with 8Gb mem

Koenkk, 2025-06-03:
This is due to a bug in the Conbee firmware, upgrading it might fix it

jocamane, 2025-06-03:
Hi, but in the link I use to see the versions, the latest one for Conbee II is the one I have installed.

https://deconz.dresden-elektronik.de/deconz-firmware/

[deCONZ_ConBeeII_0x26780700.bin.GCF](https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26780700.bin.GCF)

Is there any other place where the firmwares can be found?

Which is recommended?

Thanks

jocamane, 2025-06-03:
ups

closed 2025-08-11 by github-actions[bot]