Koenkk/zigbee2mqtt #27605: Zigbee2MQTT 2.4.0, OTA. Incorrect detection of firmware version of all devices

garry0garry, 2025-06-03:
### What happened?

Zigbee2MQTT 2.4.0, OTA. Incorrect detection of firmware version of all devices.

[image]

### How to reproduce it (minimal and precise)

Update Zigbee2MQTT from 2.3. to 2.4.0

### Zigbee2MQTT version

2.4.0

### Adapter firmware version

7.4.5 [GA]

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus V2” (model "ZBDongle-E"). Ember protocol.

### Setup

Zigbee2MQTT in docker.
Linux MyHomeAssistant 6.8.0-60-generic #63-Ubuntu SMP PREEMPT_DYNAMIC Tue Apr 15 19:04:15 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu 24.04.2 LTS

alting, 2025-06-03:
confirm

toxacj, 2025-06-03:
the same story, I confirm

sjorge, 2025-06-03:
@Koenkk Seems Aqara (only have those my self) and Tuya don't have useful data in the installed_version and available_version displayed now:

Which was changed in https://github.com/nurikk/zigbee2mqtt-frontend/commit/d76f3cc96c3e2936b21e1113c18110679210cc9a

[image]

Vs some more standard following devices

[image]

Koenkk, 2025-06-04:
@Ricc68 are you sure it's decoding properly? (related PR https://github.com/nurikk/zigbee2mqtt-frontend/pull/2470)

Ricc68, 2025-06-04:
The decoding is based on the ZigBee protocol definition of file version which is used to decide if an OTA update is available or not:

[image]

Note all file version numbers are considered as hex and decoded to strings as hex (as is) by breaking down the file version 32 bit hex value padded on the left with zeroes.

I see a potential decoding issue when -1 is reported.
@garry0garry or @AlexKay88 can you please post the value of the TS011F_plug_3->status->installed_version and available_version from the z2m UI?
Also you can check the decoding turning these 32 bits integers to hex and breaking down the result as per the above protocol table.

Consider that the firmware version and date strings encoded by the manufacturer are not mandatory and are at the manufacturer discretion therefore they are not necessarily related to the file versions used by the OTA logic (in fact they are not used by the OTA logic).

You can always find the manufacturer version and date strings in the device->info page.

garry0garry, 2025-06-05:
The Tuya Plug TS011F_plug_3 might be a bad example: https://github.com/Koenkk/zigbee2mqtt/issues/25124

TS011F_plug_1, _TZ3000_gvn91tmx "0.0.00-0.0.c0"
192 == xC0
{
    "child_lock": "UNLOCK",
    "countdown": 0,
    "current": 0.69,
    "energy": 461.02,
    "indicator_mode": "off/on",
    "last_seen": "2025-06-05T10:42:36+03:00",
    "linkquality": 144,
    "power": 60,
    "power_outage_memory": "on",
    "state": "ON",
    "update": {
        "installed_version": 192,
        "latest_version": 192,
        "state": "idle"
    },
    "voltage": 229
}

```

AlexKay88, 2025-06-05:
@Ricc68 A general question: How to get the 4 bytes where `installed_version` is derived from? I know how to read swBuildId in Dev console from cluster genBasic, but the source of `installed_version` must be something else.

Ricc68, 2025-06-05:
> [@Ricc68](https://github.com/Ricc68) A general question: How to get the 4 bytes where `installed_version` is derived from? I know how to read swBuildId in Dev console from cluster genBasic, but the source of `installed_version` must be something else.

It's in OTA Upgrade cluster.

Ricc68, 2025-06-05:
> [@Ricc68](https://github.com/Ricc68) `-1` indeed comes from z2m and means no OTA available.

Thanks, I'll set the version to N/A in case it's -1.

Ricc68, 2025-06-06:
> > TS011F_plug_3
> > "softwareBuildID": "500+TZSKT31BS102",
> 
> [@Ricc68](https://github.com/Ricc68) What information can I collect so that the version is determined correctly? [#25124](https://github.com/Koenkk/zigbee2mqtt/issues/25124) : _TZ3000_w0qqde0g Firmware version 500+TZSKT21Z3102

Sorry but looks like this is the "correct" firmware version string hardcoded into that device therefore that garbage is what the manufacturer decided is the correct firmware version, at its complete discretion.
The same manufacturer that decided that the OTA Update cluster is non responding (or not existent) for that device.
I don't see any possibility of improvement for this device.

garry0garry, 2025-06-06:
20B+TZSKT11BS108
20B+TZSKT41BS101
500+TZSKT31BS102
500+TZSKT21Z3102
It's a shame that such versions will remain in this form :-(

Ricc68, 2025-06-07:
Sadly we can do nothing on the z2m side as we're just reading back what is reported by the device firmware.

@garry0garry if the above solves all your doubts and noone else reported further issues with the OTA page, I would suggest to close the ticket as not an issue.

Thanks for raising your doubts as it was a chance to double check the implementation.

Ricc68, 2025-06-07:
Because tuya devices are very much customised and the tuya firmware version you are seeing in the tuya app is encoded in a particular way (8bits total=2bits.2bits.4bits):

[image]

I very much doubt that the z2m developers want to chase all manufacturer specific deviations from the ZigBee standard.

Your 1.2.4 version is encoded to Dec 100 or 0x64 or 'd' ASCII but it doesn't seem to be encoded into the swBuildId unless there's some unknown trick to decode it.

Additionally Tuya device firmware updates, by design/default, are coordinated from the Tuya developers platform cloud and that's probably the reason the device is not responding to OTA requests. I think that these are all choices of the developers of the devices.

I'll leave as an exercise for you to imagine how many Tuya devices I own.

ams2990, 2025-06-08:
I'm also seeing surprising behaviors in my OTA page:

[image]

For the Hue lights, the device's page says `1.122.2` is installed (which matches both my memory and the Hue [release notes](https://www.philips-hue.com/en-us/support/release-notes/lamps)), but the OTA page says it's `0.1.00-2.5.02`. Similarly, my Inovelli switch page says `2.18` but the OTA page says `0.1.02-0.2.12`.

The OTA page _used to_ match the device page, but no longer does.

AlexKay88, 2025-06-09:
@garry0garry I think this can be closed now. We learned that OTA page does not show swBuildID and date code as it did before, but now shows installed version from another clusters source and is working as designed.

AlexKay88, 2025-06-09:
In the beginning I was confused to see the new format of installed version, but IMHO it makes sense to follow the official Zigbee specs, although some vendors like Tuya don't care about. 
@ams2990 You will still see the former swBuildId on the left in the same line per device on OTA page if you switch to the new frontend which has been introduced with z2m 2.4.

Ricc68, 2025-06-09:
> > You have the firmware version/build info in the device info page so nothing is missing.
> 
> As shown in the screenshot in my previous comment, the non-updated light does not show the available firmware version number 1.222.2 anywhere.

Works as expected.

ams2990, 2025-06-11:
I've created https://github.com/nurikk/zigbee2mqtt-frontend/pull/2502, which reverts the recent changes. I suggest any discussion on the merits of the old or new approach happen there.

Nerivec, 2025-06-11:
> As shown in the screenshot in my previous comment, the non-updated light does not show the available firmware version number 1.222.2 anywhere.

`swBuildId` and `dateCode` are only reported for the **currently running firmware** on a device. It is not available for pending OTAs (will update after the OTA update is completed and [the new firmware is then running](https://github.com/Koenkk/zigbee2mqtt/blob/bdb94da46e0461337f4a61b4f2a6bfa5172f608f/lib/extension/otaUpdate.ts#L353-L355)).
The metadata available for pending OTAs is limited to what the ZigBee OTA image format provides (anything else would have to be manually maintained):
https://github.com/Koenkk/zigbee-OTA/blob/8f31924eac025f5359862680aff716940b1eff26/src/common.ts#L102-L141

closed 2025-07-14 by Nerivec