Koenkk/zigbee2mqtt #27612: [Bug]: Philips Hue 8719514279131 (LWA004) color_temp exposed status mismatch in Edge

ctankep23, 2025-06-03:
### What happened?

Philips Hue White Ambiance A60 E27 (`LWA004`) bulb is sending `color_temp` data in its MQTT state messages, but Zigbee2MQTT Edge's "Exposes" capabilities for this device (and consequently Home Assistant's `supported_color_modes`) do not list `color_temp`. This results in Home Assistant not displaying a color temperature slider for the light and continuously logging `Invalid color mode 'color_temp' received for entity` warnings. The warnings appear frequently every time the light's state updates or is controlled.

### What did you expect to happen?

I expected Zigbee2MQTT Edge to accurately identify and expose `color_temp` for the Philips Hue `LWA004` bulb, as it is a 'White Ambiance' model that natively supports color temperature adjustment and is actively sending `color_temp` data via MQTT. This should lead to the `color_temp` control slider being available in Home Assistant for this light, and the absence of `Invalid color mode` warnings in the logs.

### How to reproduce it (minimal and precise)

1.  Ensure Zigbee2MQTT Edge is running and the Philips Hue White Ambiance A60 E27 (`LWA004`) bulb is paired.
2.  Access the Zigbee2MQTT Edge web UI, navigate to the `LWA004` device's details page. Observe that the "Exposes" section **does not list `color_temp`**.
3.  In Home Assistant, go to Developer Tools -> States and find the entity `light.philips_hue_livingroom_window_left_lamp_light`. Observe that its `supported_color_modes` attribute is `brightness` (i.e., `color_temp` is absent).
4.  Open the Home Assistant UI for the `light.philips_hue_livingroom_window_left_lamp_light` entity. Observe that **only a brightness slider is available, and there is no control for color temperature.**
5.  Control the light (e.g., turn it on/off, change brightness).
6.  Monitor the MQTT topic for the device (e.g., `zigbee2mqtt/Philips Hue Livingroom Window Left Lamp Light`) using an MQTT Explorer.
7.  Observe that the MQTT state payload consistently includes `"color_mode": "color_temp"` and a `color_temp` value.
8.  Check Home Assistant logs (Home Assistant -> Settings -> Logs) for the warning: `Invalid color mode 'color_temp' received for entity light.philips_hue_livingroom_window_left_lamp_light`.

### Zigbee2MQTT version

1.25.1-dev commit: b379168

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

### Setup

Home Assistant OS on Raspberry Pi 4 (rpi4-64, 8 GB memory)
    * Home Assistant Core: 2025.5.3
    * Supervisor: 2025.05.3
    * Operating System: 15.2
    * Frontend: 20250516.0

`Linux core-ssh 6.6.74-haos-raspi #1 SMP PREEMPT Mon Apr 14 16:32:39 UTC 2025 aarch64 Linux`

### Debug log

--- Zigbee2MQTT Debug Log (Relevant Time-Matched Snippets) ---

**Z2M publishing color_temp for the group (relevant to HA warnings below)**
info 2025-06-03 23:33:43MQTT publish: topic 'zigbee2mqtt/Livingroom Media Table Group (z2m)', payload '{"brightness":138,"color_mode":"color_temp","color_temp":454,"state":"ON"}'

**Z2M log showing a command to your Philips Hue light failed (underlying issue)**
error 2025-06-03 23:34:44Publish 'set' 'brightness' to 'Philips Hue Livingroom Window Left Lamp Light' failed: 'Error: Command 0x001788010b81877c/11 genLevelCtrl.moveToLevelWithOnOff({"level":33,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
info 2025-06-03 23:34:44MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to 'Philips Hue Livingroom Window Left Lamp Light' failed: 'Error: Command 0x001788010b81877c/11 genLevelCtrl.moveToLevelWithOnOff({\"level\":33,\"transtime\":0}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"Philips Hue Livingroom Window Left Lamp Light"},"type":"zigbee_publish_error"}'

--- Home Assistant Log snippets (Invalid Color Mode Warnings - time-matched) ---

**Warnings corresponding to 23:33:xx Z2M MQTT publishes for the light/group**
2025-06-03 23:33:43.512 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'color_temp' received for entity light.philips_hue_livingroom_window_left_lamp_light
2025-06-03 23:33:43.527 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'color_temp' received for entity light.livingroom_window_lamps_group_z2m

**Other closely related warnings for the same entities, showing the ongoing issue**
2025-06-03 23:30:17.357 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'color_temp' received for entity light.livingroom_window_lamps_group_z2m
2025-06-03 23:30:17.360 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'color_temp' received for entity light.philips_hue_livingroom_window_left_lamp_light
2025-06-03 23:32:41.518 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'color_temp' received for entity light.philips_hue_livingroom_window_left_lamp_light
2025-06-03 23:32:41.534 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'color_temp' received for entity light.livingroom_window_lamps_group_z2m
2025-06-03 23:32:51.869 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'color_temp' received for entity light.philips_hue_livingroom_window_left_lamp_light
2025-06-03 23:33:06.075 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'color_temp' received for entity light.philips_hue_livingroom_window_left_lamp_light
2025-06-03 23:33:07.187 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'color_temp' received for entity light.philips_hue_livingroom_window_left_lamp_light
2025-06-03 23:33:27.567 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'color_temp' received for entity light.philips_hue_livingroom_window_left_lamp_light
2025-06-03 23:33:29.272 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'color_temp' received for entity light.philips_hue_livingroom_window_left_lamp_light
2025-06-03 23:33:45.777 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'color_temp' received for entity light.livingroom_window_lamps_group_z2m

ctankep23, 2025-06-03:
> I would first update Z2M. You are currently using a quite old version (latest Z2M is 2.4.0) and your issue might have been solved already.

Woah boy, yeah. I turned off edge and turned on the regular Z2M addon (2.4.1) before you replied! - and nothing broke, so thats good! But as of yewt the problem remains. I also updated the firmwre on the bulb but nothing changed in what it exposes.

ctankep23, 2025-06-04:
Certainly!

I found the following for the device with IEEE address `0x001788010b81877c`:

```json
{"id":39,"type":"Router","ieeeAddr":"0x001788010b81877c","nwkAddr":29527,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LWO003","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64515,64516],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LWO003","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":8,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20240902","swBuildId":"1.122.2"}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":33}}},"binds":[],"configuredReportings":[],"meta":{"scenes":{"10_7":{"name":"Movie_Time_Group_Scene_On_z2m","state":{"state":"ON","brightness":33,"color_mode":"color_temp","color_temp":347}},"20_7":{"name":"Movie_Time_Group_Scene_Off_z2m","state":{"state":"OFF","brightness":33,"color_mode":"color_temp","color_temp":347}}}}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20240902","swBuildId":"1.122.2","zclVersion":8,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1749016679157}
```

**Key Observation:**

My bulb is a **Philips Hue White Ambiance A60 E27 (Model: `LWA004`)**. However, the `database.db` entry shows its `modelId` as **`LWO003`**, and it is likely the root cause of the problem.

The `database.db` entry also shows the `lightingColorCtrl` cluster is present but its `attributes` section is empty, despite scenes explicitly using `color_mode: "color_temp"`.

chris-1243, 2025-06-04:
According to

https://zigbee.blakadder.com/Philips_LWO003.html

your light with product code : `8719514279131` has modelID `LWO003` as stated in your database.db.

I would remove your light, restart Z2M and pair it again to your network. Something went wrong durimg this phase.

Anyway, modelID `LWO003` or `LWA004` are white only lights.

ctankep23, 2025-06-04:
Thanks, that did it!

closed 2025-06-04 by ctankep23