Koenkk/zigbee2mqtt #27613: Nedis ZBWD20RD dimmer (_TZ3210_mt5xjoy6) only partially working

pullasuti, 2025-06-04:
### What happened?

Support for the device was added in the latest version, however it does not work "out-of-the-box" with the same converter as the other dimmers. There are problems getting/setting mostly the L2 values. 

Mentioned in the earlier thread, but don't know other way to report than to repost 🤷‍♂ 

https://github.com/Koenkk/zigbee2mqtt/issues/27364

### What did you expect to happen?

Expected that exposed controls would work  

### How to reproduce it (minimal and precise)

Install latest stable Z2M (2.4.0)
add dimmer (_TZ3210_mt5xjoy6) 
set/get: 
-Power-on behavior (Endpoint: l2)  
-Light type
-Switch type (Endpoint: l1)
-Switch type (Endpoint: l2)

### Zigbee2MQTT version

2.4.0

### Adapter firmware version

Revision: 8.0.2 [GA]

### Adapter

SLZB-06Mg24

### Setup

Running Z2M as Home Assistant (VM) addon 

### Debug log

`[2025-06-04 09:48:50] error: 	z2m: Publish 'get' 'power_on_behavior' to '0xa4c1388a986dabe7' failed: 'Error: ZCL command 0xa4c1388a986dabe7/2 genOnOff.read(["moesStartUpOnOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
[2025-06-04 09:48:57] error: 	z2m: Publish 'set' 'power_on_behavior' to '0xa4c1388a986dabe7' failed: 'Error: ZCL command 0xa4c1388a986dabe7/2 genOnOff.write({"moesStartUpOnOff":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
[2025-06-04 09:48:57] error: 	z2m: Publish 'set' 'power_on_behavior' to '0xa4c1388a986dabe7' failed: 'Error: ZCL command 0xa4c1388a986dabe7/2 genOnOff.write({"moesStartUpOnOff":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
[2025-06-04 09:49:02] error: 	z2m: Publish 'get' 'power_on_behavior' to '0xa4c1388a986dabe7' failed: 'Error: ZCL command 0xa4c1388a986dabe7/2 genOnOff.read(["moesStartUpOnOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
[2025-06-04 09:49:10] error: 	z2m: Publish 'get' 'light_type' to '0xa4c1388a986dabe7' failed: 'Error: ZCL command 0xa4c1388a986dabe7/1 genLevelCtrl.read([64514], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
[2025-06-04 09:49:13] error: 	z2m: Publish 'set' 'light_type' to '0xa4c1388a986dabe7' failed: 'Error: ZCL command 0xa4c1388a986dabe7/1 genLevelCtrl.write({"64514":{"value":1,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
[2025-06-04 09:49:19] error: 	z2m: Publish 'get' 'switch_type' to '0xa4c1388a986dabe7' failed: 'Error: ZCL command 0xa4c1388a986dabe7/1 genLevelCtrl.read([64514], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
[2025-06-04 09:49:22] error: 	z2m: Publish 'set' 'switch_type' to '0xa4c1388a986dabe7' failed: 'Error: ZCL command 0xa4c1388a986dabe7/1 genLevelCtrl.write({"64514":{"value":2,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
[2025-06-04 09:49:28] error: 	z2m: Publish 'set' 'switch_type' to '0xa4c1388a986dabe7' failed: 'Error: ZCL command 0xa4c1388a986dabe7/2 genLevelCtrl.write({"64514":{"value":0,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
[2025-06-04 09:49:28] error: 	z2m: Publish 'set' 'switch_type' to '0xa4c1388a986dabe7' failed: 'Error: ZCL command 0xa4c1388a986dabe7/2 genLevelCtrl.write({"64514":{"value":0,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
[2025-06-04 09:49:31] error: 	z2m: Publish 'get' 'switch_type' to '0xa4c1388a986dabe7' failed: 'Error: ZCL command 0xa4c1388a986dabe7/2 genLevelCtrl.read([64514], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

Cimmerholm, 2025-07-23:
I have the exact same issue.

gitvisual, 2025-10-24:
@pullasuti 
@Timonsteri 
@Cimmerholm 

You ever got it working? Im having same problem. Tried with external converter as mentioned in link above but didnt help