Koenkk/zigbee2mqtt #27628: [New device support]: Aqara Roller Shade Controller T1S

sendorm, 2025-06-05:
### Link

https://www.aqara.com/en/product/roller-shade-controller-t1s/

### Database entry

{"id":16,"type":"Router","ieeeAddr":"0x54ef441000d08988","nwkAddr":32992,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.curtain.vagl02","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,2,3,4,5,10,19,13,258],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":41,"stackVersion":2,"hwVersion":1,"dateCode":"07-03-2023","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1749114997041}

### Zigbee2MQTT version

2.4.0

### Comments

This device is sold and labeled as Aqara Roller Shade Driver T1S (CD-M03D), but it reports the same Zigbee model_id (lumi.curtain.vagl02) as the T1C in Zigbee2MQTT and is always detected as ZNGZDJ16LM (T1C). The external definition is just a place holder. The T1S also exposes genMultistateOutput which also receives open close and stop. 

interview_parameters:
  ieee_address: '0x54ef441000d08988'
  network_address: 32992
  manufacturer:
    id: 4447
    name: 'LUMI'
  model_id: 'lumi.curtain.vagl02'
  power_source: 'Mains (single phase)'
  firmware:
    application_version: 41
    stack_version: 2
    zcl_version: 3
  hardware:
    version: 1
    date_code: '07-03-2023'
  endpoints:
    1:
      profile_id: 260
      endpoint_id: 1
      device_id: 514
      input_clusters: [0, 2, 3, 4, 5, 10, 19, 13, 258]  
    242:
      profile_id: 41440
      endpoint_id: 242
      device_id: 97
      output_clusters: [33]  # Green Power
  interview_status:
    completed: true
    state: 'SUCCESSFUL'

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            modelID: 'lumi.curtain.vagl02',
            manufacturerName: 'LUMI',
            manufacturerID: 4447,
            applicationVersion: 41,
            endpoints: [
                {ID: 1, inputClusters: [0,2,3,4,5,6,9,1794,2820,19,13,258], outputClusters: [10,25]}
            ]
        }
    ],
    model: 'CD-M03D',
    vendor: 'Aqara',
    description: 'Aqara Roller Shade Driver T1S',
    fromZigbee: [fz.cover_position_tilt, fz.ignore_basic_report],
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
    exposes: [e.cover_position(), e.binary('running', ea.STATE, true, false)],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
        await reporting.currentPositionLiftPercentage(endpoint);
    }
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I couldn't get the external converter to work. it just defaults to T1C. The running state is always null. Other than that the current T1C implementation works. Can close, stop and open the cover. 
  - Device sends `manuSpecificTuya` commands (unusual for Aqara devices)
  - Shares model ID `lumi.curtain.vagl02` with T1C
  - Hardware version differs: T1S uses HW 1 vs T1C's HW 0

sendorm, 2025-06-09:
I was able to get the running states: 
Stopped: zhc:tz: Read result of 'genMultistateOutput': {"presentValue":2}
Opening: zhc:tz: Read result of 'genMultistateOutput': {"presentValue":1}
Closing: zhc:tz: Read result of 'genMultistateOutput': {"presentValue":0}

sendorm, 2025-06-10:
It was a problem with the homeassistant/cover/0xXXXXXXXXXXXXXX/cover/config message. 

the value_template was this : `{% if \"running\" in value_json and value_json.running %} {% if value_json.position > 0 %} closing {% else %} opening {% endif %} {% else %} stopped {% endif %} `

I have updated the code to this and exposed "direction". 
```
const fzLocal = {
    lumi_curtain_running_state: {
        cluster: 'genMultistateOutput',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('presentValue')) {
                const presentValue = msg.data.presentValue;
                let running = null;
                let direction = null;
                
                // Set running true for 0 (closing) or 1 (opening)
                if (presentValue === 0 || presentValue === 1) {
                    running = true;
                } else if (presentValue === 2) {
                    running = false;
                }

                // Map presentValue to direction
                if (presentValue === 0) {
                    direction = 'down';
                } else if (presentValue === 1) {
                    direction = 'up';
                } else if (presentValue === 2) {
                    direction = 'stopped';
                }

                return { running: running, direction: direction };
            }
        },
    },
};
```

`"value_template": "{% if \"running\" in value_json and value_json.running %} {% if value_json.direction == \"down\" %} closing {% else %} opening {% endif %} {% else %} stopped {% endif %}"`

Then published the homeassistant/cover/0xXXXXXXXXXXXXXX/cover/config message from mqtt.fx with the corrected value_template. Now the direction works everytime as it is published from the device itself.

closed 2025-08-17 by github-actions[bot]

Chao-chao00, 2026-01-11:
Hi i have the same problem. 

It is either the open/close command in Zigbee2MQTT works fine but the shade wired remote has reversed up/down controls.
Or if i reversed the wired remote controls (by holding the stop button for 5s) then the remote controls are correct but the open/close command in Zigbee2MQTT is reversed.

So i think we need a toggle in Zigbee2MQTT to reverse open/close commands (should not affect the wired remote controls) just like the Aqara App has for that device.