Koenkk/zigbee2mqtt #27636: [New device support]: Somfy Glydea Ultra ZB

rickd1994, 2025-06-06:
### Link

https://www.somfysystems.com/en-us/blog/post/introducing-the-glydea-ultra-range

### Database entry

{"id":104,"type":"Router","ieeeAddr":"0x4cc206fffe308d44","nwkAddr":6169,"manufId":4640,"manufName":"SOMFY","powerSource":"Mains (single phase)","modelId":"Glydea Ultra Curtain","epList":[1,232,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,3,4,5,258],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"Glydea Ultra Curtain","manufacturerName":"SOMFY","powerSource":1,"zclVersion":8,"swBuildId":"5"}},"closuresWindowCovering":{"attributes":{"currentPositionLiftPercentage":0,"windowCoveringType":0}}},"binds":[{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x0ceff6fffe939abd","endpointID":1}],"configuredReportings":[{"cluster":258,"attrId":8,"minRepIntval":1,"maxRepIntval":65000,"repChange":1}],"meta":{}},"232":{"profId":260,"epId":232,"devId":5,"inClusterList":[0,25],"outClusterList":[0,25],"clusters":{"genBasic":{"attributes":{"modelId":"Glydea Ultra Curtain","manufacturerName":"SOMFY","powerSource":1,"zclVersion":8,"swBuildId":"5"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"swBuildId":"5","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1749211933473}

### Zigbee2MQTT version

2.1.3

### Comments

[image]

### External definition

```shell
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['Glydea Ultra Curtain'],
    model: 'Glydea Ultra Curtain',
    vendor: 'SOMFY',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"232":232}}), m.windowCovering({"controls":["lift","tilt"]})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

The External definition shows "tilt" but the motor does not have a tilt feature

chris-1243, 2025-06-06:
Would you import this as an `*.mjs` file. It is quite close to your file, I just removed th `tilt` function and add a description. Let us know if everything is fine and it will be added...

```javascript
import * as m from "zigbee-herdsman-converters/lib/modernExtend";

export default {
    zigbeeModel: ["Glydea Ultra Curtain"],
    model: "Glydea Ultra Curtain",
    vendor: "SOMFY",
    description: "Glydea curtain motorization",
    extend: [
        m.deviceEndpoints({"endpoints":{"1": 1,"232": 232}}),
        m.windowCovering({"controls":["lift"]}),
    ],
    meta: {"multiEndpoint": true},
};
```

rickd1994, 2025-06-06:
✅ **Update**: I replaced my file with your suggested `.mjs` version — everything seems to work perfectly! Thank you for the cleanup.

---

🔧 **Device info:**

I’ve added some photos of the device boxes I have — this setup is working for me:

- **Motor**: `Somfy Glydea Ultra 35e RTS N`
- **Zigbee Module**: paired with Somfy Curtain Motor Zigbee Module, **Ref. 9028412A**  
  (very similar to the one you shared — works on **Glydea Ultra 60e RTS N** as well)

📸 Box images:  
[image]  
[image]

---

🔄 **Pairing notes:**

I struggled with pairing initially. Here’s what worked for me:

1. Tried **universal pairing mode** (double jog after reset hold) — this did **not** seem to work on my unit.
2. Instead, I followed this guide:  
   👉 https://mrmelair.com/2024/07/10/hacking-the-somfy-sonesse2-zigbee-motors/

3. I used an iOS app called **BT Inspector** to pair via BLE manually:
   - Connected to the motor over BLE.
   - Sent the **3-byte PIN code** to:
     ```
     0000000B-cad9-46c6-a2ea-2ca16d57b4a5
     ```
     Format: 3-byte int, little-endian. In my case: just the **raw numeric code**, not hex.
   - Then sent `0x01` to:
     ```
     00020001-cad9-46c6-a2ea-2ca16d57b4a5
     ```

Right after that, the motor paired with **ZHA/Home Assistant**, and later worked fine in **Zigbee2MQTT** (after fixing the `tilt` issue).

---

Let me know if you'd like me to test anything else. Thanks again for the support — happy to contribute more if needed!

chris-1243, 2025-06-07:
@rickd1994 Could you try this converter and report if everything is working ? Use an `*.mjs` extension.

```javascript
import * as m from "zigbee-herdsman-converters/lib/modernExtend";

export default {
    zigbeeModel: ["Glydea Ultra Curtain"],
    model: "9028412A",
    vendor: "SOMFY",
    description: "Glydea Curtain motor Zigbee module",
    extend: [m.windowCovering({"controls":["lift"]})],
};
```

rickd1994, 2025-06-10:
@chris-1243 thanks for the suggestion. I gave it a try, but unfortunately it doesn’t seem to work — the device shows up as unsupported again. Something strange is happening though: even when I revert back to your first .mjs version, it still shows up as unsupported.

However, as soon as I put back my original .js version, the device is recognized as supported again. I might be doing something wrong, but at the moment, only the original .js version seems to work.
Any idea what might be going on?

rickd1994, 2025-06-10:
This works perfect also got the picture included. sorry i did not mention my Z2M version 

[image]

rickd1994, 2025-06-10:
Yes, this is from the latest version you sent me, saved as *.js. As for the functions, I'll have to double-check when I get home to see if the curtains are actually moving 😉 I'll report back around 17:30 local time (Netherlands), which is 15:30 UTC.

closed 2025-09-22 by github-actions[bot]