Koenkk/zigbee2mqtt #27649: Z2m request backup don’t work after update Z2M 2.4. Z2M restart.

Razer0ne, 2025-06-07:
### What happened?

When manual request backup don’t work after update from 2.3 to 2.4. 

Z2M reatart every time manual request backup in Home assistant Z2M

Backup works fine before update

### What did you expect to happen?

Z2m restart and no backup i want

### Zigbee2MQTT version

2.4.0

### Adapter firmware version

v2.8.8/20250212 

### Adapter

Smlight slzb-06MG24 Ember

### Setup

Home assistant Rpi 4B with z2m addon

### Debug log

2025-06-07 20:02:29] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-06-07 20:02:29] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-06-07 20:02:29] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-06-07 20:02:29] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-06-07 20:02:29] error: 	z2m: Adapter disconnected, stopping
[20:02:31] INFO: Preparing to start...
[20:02:31] INFO: Socat not enabled
[20:02:32] INFO: Starting Zigbee2MQTT...

Razer0ne, 2025-06-08:
> I cannot replicate with a Dongle-E and ember. Nothing's changed in the ember driver between these versions either.
> 
> Did you change firmware since the last working backup request you remember? Or move the adapter (port/location) maybe?

I dont know.  Smlight 06mg24 has worked perfectly, only after updating to 2.4 manual backup does not work, z2m restarts every time I press request backup. everything else is stable except for so long after updating. since backup worked on 2.3, it must be something with the update

I updates to last core fw for smlight from 2.8.5 to 2.8.8. Mayby last fw with z2m 2.4 do the backup issue?

Nerivec, 2025-06-08:
Did you try downgrading back the core fw?

Razer0ne, 2025-06-08:
> Did you try downgrading back the core fw?

I roll back to  smlight v2.8.5.dev12 i used and works with manual backup before update to z2m 2.4. Same happend,  z2m restart when manual backup off z2m.  So has to be 2.4 makes this issue

Nerivec, 2025-06-08:
Can you provide a couple minutes worth of `debug` logs before these?

Razer0ne, 2025-06-08:
> Can you provide a couple minutes worth of `debug` logs before these?

Log when press request backup in z2m:

2025-06-08 20:51:32] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-06-08 20:51:32] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-06-08 20:51:32] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-06-08 20:51:32] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-06-08 20:51:32] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-06-08 20:51:32] error: 	z2m: Adapter disconnected, stopping
[20:51:34] INFO: Preparing to start...

This error i not understand why,  from 2.3. 2.4 came with new interface , i addet 

frontend:
  enabled: true
  package: zigbee2mqtt-windfront

This line was new in config:   package: zigbee2mqtt-windfront

Maybe new i interface do this. 

Debug say this when running. 

[2025-06-08 21:11:46] frontend:api: Sending {"topic":"bridge/request/options","payload":{"options":{"advanced":{"log_level":"info"}},"transaction":"pa14s-1"}}
[2025-06-08 21:11:46] z2m: Successfully changed options
[2025-06-08 21:11:47] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"pa14s-1"}'

All zigbee sensor works like before.  And only manuell backup issue z2m restart

Razer0ne, 2025-06-08:
> The error is from the adapter crashing, but since I can't reproduce with a different adapter (same frontend), and the same version, this is not a "global" 2.4 problem but something specific in your setup.
> 
> If you roll back to 2.3 without changing anything, it works again?
> 
> Also check if the backup is working with 2.4 when you shutdown Z2M (it will make a backup automatically right before). You can check the timestamp in the `coordinator_backup.json` to see if it was updated (or the `debug` logs).

Yes, 2.4 and 06mg24 do this this error and restart.  Its works on 2.3 and release before. since i allways take manual backup before update etc.  roll back to 2.3 wil works since works before update. . Hope next update will fix this.  in the meantime take backup file for file in location folder of all z2m files.   smlight 06mg24 is the newest coordinator from Smlight with EFR32MG24 Zigbee chip.

Razer0ne, 2025-06-09:
> > roll back to 2.3 wil works since works before update
> 
> That's what needs testing, because if something else changed (in your network, in your surroundings, etc.), we could be looking for a problem that isn't about Z2M itself (which is likely since the code related to that adapter has not changed at all between these versions).

Maybe, i got no any error and z2m 2.4 is stable.  So has to be any issue when manual request backup to my 06mg24 coordinator. something from 2.3 backup works, to 2.4 backup get error, adapter no longer answer and restart when request bacup.  have seen any error for smlight adapter like mine error, for other reason, and adapter restart?  

for info, my adapter have low temp, SoC temperature:	29.13°C	32.22°C. never higher temp , i have seen pepole say high temp. I monitor the temperature and allways low

Razer0ne, 2025-06-09:
> Since the error is about missing ACKs, and the backup appears to work fine when Z2M is shutting down (same process as manual), I'd say it's a timing issue. When Z2M is shutting down, the adapter is no longer receiving requests from Z2M, whereas when you request a manual backup, the network is live and talking (more noise). Any change on your ethernet network (the one the 06mg24 is connected to) that could be affecting the overall response time?

No change on network or settings.  Same like before.  Only update from 2.3 to 2.4 and fw for smlight was the same error men roll back core fw i tested yesterday.
If there was a network error or something else, I would notice it by restarting itself or other errors. The adapter works 100%, as long as I don't take request backup, and the adapter stops responding.

Nerivec, 2025-06-10:
Any chance you can try this by using the USB mode of the adapter?

appleimperio, 2025-06-10:
I'm going to increase the transmit power and see if it helps if not I will try by USB. The thing is it's start working but after a few hours the addon quits. If you think that I should try by USB first let me know. Thank You

Nerivec, 2025-06-10:
Are you saying your issue is not specific to "requesting a manual backup" it just throws this error after a while without doing anything in particular right before?

appleimperio, 2025-06-10:
ohhh you are right mine just happened without requesting a manual backup like just happens right now after the change I made but I should start a new issue because is different

appleimperio, 2025-06-11:
I downgrade the firmware from 2.88 that was release a few days ago to 2.7.1 and so far is working great tomorrow if nothing happens I will try to install it again just to be sure.

10der, 2025-07-06:
```
Zigbee2MQTT version
[2.5.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.5.1) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Coordinator type
EmberZNet
Coordinator revision
7.4.3 [GA]
Coordinator IEEE Address
0xe0798dfffed1f56b
Frontend version
0.9.16
zigbee-herdsman-converters version
23.69.0
zigbee-herdsman version
4.3.1
```

```
[2025-07-06 04:15:46] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-07-06 04:15:46] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-07-06 04:15:46] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-07-06 04:15:46] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-07-06 04:15:46] debug: 	zh:controller: Adapter disconnected
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   Total frames: RX=63, TX=97
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   DATA frames : RX=46, TX=34
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   DATA bytes  : RX=647, TX=343
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   Retry frames: RX=15, TX=0
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=62
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   Retry dupes     : RX=15
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2025-07-06 04:15:46] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2025-07-06 04:15:46] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=33 Len=37]
[2025-07-06 04:15:46] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler: type=UNICAST apsFrame={"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":189} packetInfo:{"senderShortId":24087,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":112,"lastHopRssi":-72,"lastHopTimestamp":0} messageContents=09260200021602000400000002
[2025-07-06 04:15:46] debug: 	zh:controller: Received payload: clusterID=61184, address=24087, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=112, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":38,"commandIdentifier":2},"payload":{"seq":512,"dpValues":[{"dp":22,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,2]}}]},"command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}}
[2025-07-06 04:15:46] debug: 	zh:controller:endpoint: ZCL command 0xa4c138a95718daa6/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":38,"writeUndiv":false})
[2025-07-06 04:15:46] debug: 	zh:ember: ~~~> [ZCL to=0xa4c138a95718daa6:24087 apsFrame={"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":0,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":38,"commandIdentifier":11}]
[2025-07-06 04:15:46] debug: 	zh:ember:ezsp: [SEND COMMAND] NOT CONNECTED

```

appleimperio, 2025-07-07:
@10der In my case what I ended up doing is Nerivec suggestion. I try it in USB mode and that way I can maintain the connection.