Koenkk/zigbee2mqtt #27657: SONOFF TRVZB - climate.set_temperature also sets thermostat HVAC mode to heat

nocin, 2025-06-08:
### What happened?

Hi,

I noticed that the SONOFF TRV's are the only TRV's I have, that are turned on (HVAC mode heat) when setting the temperature by using `climate.set_temperature.`

Example: I have manually turned off my TRV's and I want them to stay off.
First thermostat is a TRV601, second the SONOFF TRVZB.

[image]

Now, some automations are changing the target temperatures for my thermostats like this:

```YAML
action: climate.set_temperature
metadata: {}
data:
  temperature: 21
target:
  entity_id:
    - climate.0xa4c13832ba136190
    - climate.0xa4c138d8b8094f89
```
The temperatures are set successfully, but it also changes the HVAC mode to heat for the SONOFF TRV.

### What did you expect to happen?

I'm expecting that the thermostat stays off (HVAC mode off), when changing the temperature. But instead the SONOFF TRV's thermostats are turned on. This does not happen with other brands.

[image]

### How to reproduce it (minimal and precise)

Turn off the thermostat of your SONOFF TRV. Change the temperature via action `climate.set_temperature`. The corresponding thermostat will be set to heat, instead of staying off. 

### Zigbee2MQTT version

2.4.0

### Adapter firmware version

20240710

### Adapter

ZStack3x0

### Setup

Add-on on Home Assistant OS as Proxmox VM

pedolsky, 2026-02-09:
I would also be interested in seeing this question clarified.

nocin, 2026-08-25:
Please reopen the issue