Koenkk/zigbee2mqtt #27671: Latest version，When the end device iasZone "Zone States Definition" returns a value of "ZONE-STATE_NOT-INROLLED", it will prompt pairing failure

13717033460, 2025-06-09:
### What happened?

When the end device iasZone "Zone States Definition" returns a value of "ZONE-STATE_NOT-INROLLED", it will prompt pairing failure。
log:

[image]
[image]

Error 2025-06-09 18:53:54zh:controller: Interview failed for '0xa4c13858e5cfc587 with error 'Error: Interview failed because of failed IAS enroll (zoneState didn't change ('0xa4c13858e5cfc587')'
Error 2025-06-09 18:53:54z2m: Failed to interview '0xa4c13858e5cfc587', device has not successfully been paired

### Zigbee2MQTT version

2.4.0

### Adapter firmware version

20211207

### Adapter

ZStack3x0 

### Setup

Add-on on Home Assistant OS On Raspberry Pi

13717033460, 2025-06-11:
Hello,Thank you for your reply！
There are many devices that respond with 'enrollment the state' and do not use iasZone on Zigbee2mqtt, but the tuya dp point function is normal ，The old version of Zigbee2mqtt works fine, but this' therefore pairing failures as it should 'only appeared after the updated version

Koenkk, 2025-06-11:
> There are many devices that respond with 'enrollment the state' and do not use iasZone on Zigbee2mqtt,

It seems to me this is an issue of the device, it has the iAS cluster but does not support enrolling.

> The old version of Zigbee2mqtt works fine, but this' therefore pairing failures as it should 'only appeared after the updated version

Which z2m version is that, can you provide the debug log of pairing with that version?

13717033460, 2025-06-13:
Yes, it's just that the "ZCL BASIC MODEL ID" has been changed to "ZG-204ZM" and the "ZCL BASIC MFG NAME" has been changed to "HOBEIAN". The previous old devices "TS0601, _TZE200_kb5noeto" also returned "0" in zoneState, but there was no such problem

13717033460, 2025-06-13:
As long as the "ZCL BASIC MODEL ID" and "ZCL BASIC MFG NAME" are changed back to the old ones, but the zone state remains the same as "0", it is normal.

[image]
[image]
[image]

13717033460, 2025-06-13:
Next version update， add HOBEIAN

Koenkk, 2025-06-13:
@13717033460 could you provide the data/database.db entry of the problematic device?

Popkornium18, 2025-06-13:
For me it is:

```json
{
  "id": 67,
  "type": "EndDevice",
  "ieeeAddr": "0xa4c138005bb87889",
  "nwkAddr": 47059,
  "manufId": 4742,
  "manufName": "HOBEIAN",
  "powerSource": "Battery",
  "modelId": "ZG-204ZM",
  "epList": [
    1
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 1026,
      "inClusterList": [
        0,
        3,
        1280,
        61184,
        1,
        1024
      ],
      "outClusterList": [],
      "clusters": {
        "msIlluminanceMeasurement": {
          "attributes": {
            "measuredValue": 28070
          }
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 148,
  "stackVersion": 2,
  "hwVersion": 1,
  "swBuildId": "0122052017",
  "zclVersion": 3,
  "interviewCompleted": false,
  "interviewState": "FAILED",
  "meta": {},
  "lastSeen": 1749845140135
}
```

Koenkk, 2025-06-14:
Thanks, will be fixed in https://github.com/Koenkk/zigbee-herdsman/pull/1422

n84ck, 2025-06-14:
Hi,
I have a EZVIZ CS-T9C Smart Siren with a very similar issue. Could you help me how I can check if this solution works for me as well?

[image]

This is the generated external definition:

```
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['CS-T9C-A0-BG'],
    model: 'CS-T9C-A0-BG',
    vendor: 'EZVIZ',
    description: 'Automatically generated definition',
    extend: [m.battery(), m.iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]}), m.iasWarning()],
    meta: {},
};

```

Koenkk, 2025-06-15:
@n84ck could you provide the data/database.db entry of this device?

osnwt, 2025-06-16:
https://www.zigbee2mqtt.io/devices/ZG-223Z.html - same issue

Koenkk, 2025-06-16:
Should be fixed now!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

If not, provide the debug logging when pairing that device.

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

closed 2025-06-16 by Koenkk