Koenkk/zigbee2mqtt #27677: [New device support]: Blind Roller TS0601 _TZE284_koxaopnk Moes ZC-LS02

carlosrey18, 2025-06-09:
### Link

https://es.aliexpress.com/item/1005009040515645.html?spm=a2g0o.order_list.order_list_main.23.7728194d1n43A9&gatewayAdapt=glo2esp

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c138efcba0f8a0","nwkAddr":64922,"manufId":4417,"manufName":"_TZE284_koxaopnk","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"d\u0000�/g�\n�/\u0011�\r�/\u0011�\u000f�/\u0011�\u0011�/\u0011�?�/\u0011�@�/\u0011�@�/\u0011�A�/\u0011\u0018L�/\u0011!Y�/\u0011","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE284_koxaopnk","powerSource":3,"zclVersion":3,"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1946894323},"lastSeen":1749476818334}

### Zigbee2MQTT version

2.4.0

### Comments

The truth is, I'm new to all of this. This is my first device. After many attempts, I managed to get it to connect with the following code, and then I added the details.

### External definition

```shell
import {access as ea, presets as e, options} from "zigbee-herdsman-converters/lib/exposes";
import * as tuya from "zigbee-herdsman-converters/lib/tuya";

export default {
    fingerprint: tuya.fingerprint("TS0601", ["_TZE284_koxaopnk"]),
    model: "ZC-LS02",
    vendor: "Tuya",
    description: "Moes Zigbee Roller Blind (ZC-LS02)",
    onEvent: tuya.onEvent(),
    configure: tuya.configureMagicPacket,
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.cover_position().setAccess("position", ea.STATE_SET),
        e.battery(),
    ],
    meta: {
        tuyaDatapoints: [
            [
                1,
                "state",
                tuya.valueConverterBasic.lookup({
                    OPEN: tuya.enum(0),
                    STOP: tuya.enum(1),
                    CLOSE: tuya.enum(2),
                }),
            ],
            [2, "position", tuya.valueConverter.coverPosition],
            [3, "position", tuya.valueConverter.raw],
            [
                5,
                "motor_direction",
                tuya.valueConverterBasic.lookup({
                    normal: tuya.enum(0),
                    reversed: tuya.enum(1),
                }),
            ],
            [101, "battery", tuya.valueConverter.raw],
        ],
    },
};
```

### What does/doesn't work with the external definition?

The problem is that the button that appears in HomeAssistant (HA) works the other way around. When the blind is open, the HA home button marks it as closed and vice versa. The same goes for the up and down button. It also doesn't show the battery level. It only did this once. Could you help me find out what I'm doing wrong? I'm very new to all of this.

Koenkk, 2025-06-10:
Can you try with:

```js
                [2, "position", tuya.valueConverter.coverPositionInverted],
                [3, "position", tuya.valueConverter.coverPositionInverted],
```

carlosrey18, 2025-06-10:
Wow, you are the Best is working now, only the battery allways have "?", could you help me or is normal?

Thank you for all your support

Koenkk, 2025-06-11:
Make this change; 

```
toZigbee: [tuya.tz.datapoints],
extend: [m.forcePowerSource({powerSource: "Battery"})], // <-- try adding this line
```

if that works: Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

carlosrey18, 2025-06-11:
Unfortunately, by adding that line, it made the converter invalid and I renamed the file and discarded it. It still doesn't show the battery. In all this time, I only managed to see a value at 100 once, but it didn't come back; it continued with the "?" sign. I think I'll leave it like that since at least it works, although I don't know how much battery it has, unless you have any other ideas. Thanks.

carlosrey18, 2025-06-17:
Is this one:

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c138efcba0f8a0","nwkAddr":64922,"manufId":4417,"manufName":"_TZE284_koxaopnk","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"�/�/gs0�/g�5�/g","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZE284_koxaopnk","zclVersion":3,"appVersion":77,"modelId":"TS0601","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1946894323},"lastSeen":1750168817574}

Koenkk, 2025-06-17:
I see `Battery` is the `powerSource` now, could you also provide a screenshot of the device About and State page in the z2m frontend?

carlosrey18, 2025-06-17:
[Image 3](https://www.dropbox.com/scl/fi/nmnnrelffbdjobrkxpt9j/Screenshot-2025-06-17-134807.png?rlkey=28g3l5lsvi20hucuhbaiv7lpa&dl=0)

Sorry, is this one.

Koenkk, 2025-06-19:
Probably it's using a different one, to find out which see: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

Koenkk, 2025-06-19:
If you leave z2m running with debug logging enabled for 24 hours, do you see any of these messages:

```
Datapoint XX not defined for '_TZE284_koxaopnk' with value XX
```

That could enable us to reverse engineer the datapoints without the need for a Tuya hub.

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Koenkk, 2025-06-21:
Likely not correct, I guess this datapoint means something different than battery.

carlosrey18, 2025-06-21:
I already check, and that is the only datapoint that show me... :(

carlosrey18, 2025-06-22:
I think that I do it, I don't know if I make it correctly sorry by that, if I found another thing about the battery I let you know.

Thank you.

ray87, 2025-06-25:
Correct me if I am wrong, but the PR hasn't been created yet ? I am really looking forward to use that device.

carlosrey18, 2025-06-25:
I already upload, but if not you can use the external coverter.

👍🏻

Koenkk, 2025-06-25:
@carlosrey18 I did not see the PR yet, can you provide me a link to it?

carlosrey18, 2025-06-25:
Could you explain to me how can upload, is my first time. and I can't found the changes that I made in the link that you send me, and I can't send it again.

Sorry

Koenkk, 2025-06-25:
I can also add it, is the converter in the [OP](https://github.com/Koenkk/zigbee2mqtt/issues/27677#issue-3130496042) the final version?

carlosrey18, 2025-06-25:
import {access as ea, presets as e, options} from "zigbee-herdsman-converters/lib/exposes";
import * as tuya from "zigbee-herdsman-converters/lib/tuya";
import * as m from "zigbee-herdsman-converters/lib/modernExtend";

export default {
    fingerprint: tuya.fingerprint("TS0601", ["_TZE284_koxaopnk"]),
    model: "ZC-LS02",
    vendor: "Tuya",
    description: "Moes Zigbee Roller Blind (ZC-LS02)",
    onEvent: tuya.onEvent(),
    configure: tuya.configureMagicPacket,
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.cover_position().setAccess("position", ea.STATE_SET),
        e.battery(),
    ],
    meta: {
        tuyaDatapoints: [
            [
                1,
                "state",
                tuya.valueConverterBasic.lookup({
                    OPEN: tuya.enum(0),
                    STOP: tuya.enum(1),
                    CLOSE: tuya.enum(2),
                }),
            ],
            [2, "position", tuya.valueConverter.coverPositionInverted],
            [3, "position", tuya.valueConverter.coverPositionInverted],
            [
                5,
                "motor_direction",
                tuya.valueConverterBasic.lookup({
                    normal: tuya.enum(0),
                    reversed: tuya.enum(1),
                }),
            ],
            [7, "battery", tuya.valueConverter.raw],
        ],
        powerSource: 'Battery',
    },
};

this is the Final version, Thank you

Koenkk, 2025-06-27:
Does the battery also work with this one?

carlosrey18, 2025-06-27:
Not sorry.. :(

carlosrey18, 2025-06-27:
I already do the upload, Thank you for all.

Koenkk, 2025-06-27:
@carlosrey18 I don't see it, could you link me to it?

carlosrey18, 2025-06-27:
https://github.com/carlosrey18/zigbee2mqtt.io/commit/6325560a9604156d5519dccfae1b37c13e8829e4

Sorry if I make it wrong, is my first time whit this.
Please let me know if I do it correctly, if not here is the image.

https://www.dropbox.com/scl/fi/x5qxn6wl30szjmhcl62q2/ZC-LS02.png?rlkey=zz66flndfsawzqeyyfukpwrii&dl=1

Koenkk, 2025-06-29:
@carlosrey18 thanks, added it in https://github.com/Koenkk/zigbee2mqtt.io/pull/3899