Koenkk/zigbee2mqtt #27685: Update from 1.41 to 2.x fails

ashkii7, 2025-06-10:
### What happened?

Running update.sh or same with git pull --no-rebase

pi@RPI-JEEDOM:/opt/zigbee2mqtt $ ./update.sh 
Checking Zigbee2MQTT status...
Creating backup of configuration...
Checking out changes to package-lock.json...
0 chemin mis à jour depuis l'index
Updating...
Mise à jour 69f72829..bdb94da4
error: Vos modifications locales aux fichiers suivants seraient écrasées par la fusion :
        data/configuration.yaml
Veuillez valider ou remiser vos modifications avant la fusion.
Abandon

pi@RPI-JEEDOM:/opt/zigbee2mqtt $ sudo git pull --no-rebase
Mise à jour 69f72829..bdb94da4
error: Vos modifications locales aux fichiers suivants seraient écrasées par la fusion :
        data/configuration.yaml
Veuillez valider ou remiser vos modifications avant la fusion.
Abandon

### What did you expect to happen?

i expected that : Smooth upgrade, or instructions on how to overcome this issue. By advance Thanks.

### How to reproduce it (minimal and precise)

0: sudo apt update
0: sudo apt upgrade
0: cd /opt/zigbee2mqtt
1: sudo systemctl stop zigbee2mqtt
2: git checkout data/configuration.example.yaml
3: git checkout data/configuration.yaml
4: mv data/configuration.yaml data/configuration.yaml.bak
5: mv data/configuration.example.yaml data/configuration.example.yaml.bak
6: sudo npm install -g pnpm
7: sudo git pull --no-rebase
8: ./update.sh

### Zigbee2MQTT version

1.41.0 (commit #69f72829)

### Adapter firmware version

    20240710

### Adapter

zStack3x0 (Zigbee Sonoff Dongle-Plus)

### Setup

In my Z2M configuration.yaml file, i put:
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false

pi@RPI-JEEDOM:~ $ cat /proc/cpuinfo | grep Model
Model           : Raspberry Pi 4 Model B Rev 1.2

pi@RPI-JEEDOM:~ $ uname -a
Linux RPI-JEEDOM 6.6.51-v8+ #1799 SMP PREEMPT Thu Sep 26 16:11:55 BST 2024 aarch64 GNU/Linux

pi@RPI-JEEDOM:~ $ cat /etc/issue.net
Debian GNU/Linux 11

mrmaximas, 2025-06-10:
try this https://github.com/Koenkk/zigbee2mqtt/issues?q=Checking%20out%20changes%20to%20package-lock.json...

ashkii7, 2025-06-10:
@mrmaximas 
thanks, i execute the commands find here : [https://github.com/Koenkk/zigbee2mqtt/issues/25517#issuecomment-2660952080](url)
and it solved the issue for me 👍😀👍
Thanks to @AGitDev

sudo systemctl stop zigbee2mqtt
apt update
apt upgrade
git checkout data/configuration.example.yaml
git checkout package-lock.json
mv data/configuration.yaml data/configuration.yaml.bak
git stash
git stash drop
npm install
npm audit fix
npm ci
npm install -g pnpm
git pull --no-rebase
./update.sh
mv data/configuration.yaml.bak data/configuration.yaml
sudo systemctl start zigbee2mqtt

closed 2025-06-10 by ashkii7