Koenkk/zigbee2mqtt #27706: [New device support]: Sercomm SZ-ESW02N-CZ3 Telstra smart plug 2

Please-Send-Help, 2025-06-12:
### Link

https://www.zigbee2mqtt.io/devices/SZ-ESW02.html

### Database entry

{"id":146,"type":"Router","ieeeAddr":"0x000d6f0014ba8003","nwkAddr":42467,"manufId":4401,"manufName":"Sercomm Corp.","powerSource":"Mains (single phase)","modelId":"SZ-ESW02N-CZ3","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,1794,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"zclVersion":3}},"genOnOff":{"attributes":{"onOff":1}},"seMetering":{"attributes":{"divisor":1000000,"multiplier":1,"instantaneousDemand":834,"currentSummDelivered":23437}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c31db2","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c31db2","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1},{"cluster":1794,"attrId":1024,"minRepIntval":10,"maxRepIntval":65000,"repChange":5000},{"cluster":1794,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":100000}],"meta":{}},"2":{"profId":260,"epId":2,"devId":259,"inClusterList":[0,3],"outClusterList":[3,6],"clusters":{"genBasic":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c31db2","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":20,"stackVersion":98,"hwVersion":16,"dateCode":"20190307","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1749753897960}

### Zigbee2MQTT version

2.4.0

### Comments

I don't know enough to adjust the external converter to show the correct values but this plug should function exactly like the [SZ-ESW02](https://www.zigbee2mqtt.io/devices/SZ-ESW02.html) which are already supported. 

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['SZ-ESW02N-CZ3'],
    model: 'SZ-ESW02N-CZ3',
    vendor: 'Sercomm Corp.',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2}}), m.onOff({"powerOnBehavior":false}), m.electricityMeter({"cluster":"metering"}), m.commandsOnOff({"endpointNames":["2"]})],
    meta: {"multiEndpoint":true},
};
```

### What does/doesn't work with the external definition?

On/Off state is reporting correctly as well as Linkquality, but Power and Energy consumption are shown as null.  

These plugs are the same as [SZ-ESW02](https://www.zigbee2mqtt.io/devices/SZ-ESW02.html) but I don't know enough to configure the external definition. 

If I look at the state, I can see the values reported under energy_1 and power_1, but they're not reported on the dashboard.

{
    "energy_1": 0.023437,
    "last_seen": "2025-06-12T15:21:10-04:00",
    "linkquality": 218,
    "power_1": 0.838,
    "state": "ON",
    "state_1": "ON",
    "energy": null,
    "power": null,
    "current": null,
    "voltage": null
}

I'm sure I'm missing something simple, but that's the story of my life.

Any help would be appreciated!

Koenkk, 2025-06-13:
Can you try with:

```js
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['SZ-ESW02N-CZ3'],
    model: 'SZ-ESW02N-CZ3',
    vendor: 'Sercomm Corp.',
    description: 'Automatically generated definition',
    extend: [m.onOff({"powerOnBehavior":false}), m.electricityMeter({"cluster":"metering"})],
};
```

Please-Send-Help, 2025-06-13:
That did it! 

Thanks! 

[image]

Please-Send-Help, 2025-06-13:
Do I need to create a PR?

Please-Send-Help, 2025-06-13:
@Koenkk  I would love to but I've never done one lol 

Hopefully I can figure it out!