Koenkk/zigbee2mqtt #27710: [New device support]: NovaDigital Tuya Water Valve TS0601 -  _TZE200_fphxkxue

soareseneves, 2025-06-13:
### Link

https://www.novadigitalsmart.com.br/produtos/valvula-com-medidor-zvl-pro

### Database entry

{"id":68,"type":"EndDevice","ieeeAddr":"0x4c97a1fffe298b55","nwkAddr":9690,"manufId":4098,"manufName":"_TZE200_fphxkxue","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���/gԭ�/fԭ�/\u0012խ�/\u0012","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_fphxkxue","stackVersion":0,"dateCode":"","appVersion":65}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1749821291435}

### Zigbee2MQTT version

2.4.0-1

### Comments

This device is very similar to https://www.zigbee2mqtt.io/devices/TS0049.html but despite trying to make it "identify" as one it did not work. I am not too familiar with coding external converters so I humbly ask for some help.

I paired it with Smart Life and got the properties.

`{
  "result": {
    "properties": [
      {
        "code": "switch",
        "custom_name": "",
        "dp_id": 1,
        "time": 1749825376800,
        "type": "bool",
        "value": false
      },
      {
        "code": "fault",
        "custom_name": "",
        "dp_id": 4,
        "time": 1749821305002,
        "type": "bitmap",
        "value": 0
      },
      {
        "code": "water_once",
        "custom_name": "",
        "dp_id": 5,
        "time": 1749821300790,
        "type": "value",
        "value": 0
      },
      {
        "code": "battery_percentage",
        "custom_name": "",
        "dp_id": 7,
        "time": 1749821305258,
        "type": "value",
        "value": 100
      },
      {
        "code": "weather_delay",
        "custom_name": "",
        "dp_id": 10,
        "time": 1749821304434,
        "type": "enum",
        "value": "cancel"
      },
      {
        "code": "countdown",
        "custom_name": "",
        "dp_id": 11,
        "time": 1749822375555,
        "type": "value",
        "value": 900
      },
      {
        "code": "work_state",
        "custom_name": "",
        "dp_id": 12,
        "time": 1749825377051,
        "type": "enum",
        "value": "idle"
      },
      {
        "code": "smart_weather",
        "custom_name": "",
        "dp_id": 13,
        "time": 1749821300790,
        "type": "enum",
        "value": "sunny"
      },
      {
        "code": "use_time_one",
        "custom_name": "",
        "dp_id": 15,
        "time": 1749825371554,
        "type": "value",
        "value": 3
      },
      {
        "code": "cycle_timing",
        "custom_name": "",
        "dp_id": 16,
        "time": 1749821300790,
        "type": "raw"
      },
      {
        "code": "timer",
        "custom_name": "",
        "dp_id": 17,
        "time": 1749821300790,
        "type": "raw"
      },
      {
        "code": "water",
        "custom_name": "",
        "dp_id": 101,
        "time": 1749821300790,
        "type": "value",
        "value": 0
      },
      {
        "code": "water_once1",
        "custom_name": "",
        "dp_id": 102,
        "time": 1749821300790,
        "type": "value",
        "value": 0
      }
    ]
  },
  "success": true,
  "t": 1749825505521,
  "tid": "10349743486411f0be34e23b0b987d40"
}`

and specifications:

`{
  "result": {
    "category": "sfkzq",
    "functions": [
      {
        "code": "switch",
        "desc": "{}",
        "name": "阀门开关",
        "type": "Boolean",
        "values": "{}"
      },
      {
        "code": "weather_delay",
        "desc": "{\"range\":[\"cancel\",\"24h\",\"48h\",\"72h\"]}",
        "name": "天气延时",
        "type": "Enum",
        "values": "{\"range\":[\"cancel\",\"24h\",\"48h\",\"72h\"]}"
      },
      {
        "code": "countdown",
        "desc": "{\"unit\":\"s\",\"min\":0,\"max\":86400,\"scale\":0,\"step\":1}",
        "name": "倒计时",
        "type": "Integer",
        "values": "{\"unit\":\"s\",\"min\":0,\"max\":86400,\"scale\":0,\"step\":1}"
      },
      {
        "code": "smart_weather",
        "desc": "{\"range\":[\"sunny\",\"clear\",\"cloud\",\"cloudy\",\"rainy\",\"snow\",\"fog\"]}",
        "name": "智能天气类型",
        "type": "Enum",
        "values": "{\"range\":[\"sunny\",\"clear\",\"cloud\",\"cloudy\",\"rainy\",\"snow\",\"fog\"]}"
      }
    ],
    "status": [
      {
        "code": "switch",
        "name": "阀门开关",
        "type": "Boolean",
        "values": "{}"
      },
      {
        "code": "water_once",
        "name": "单次水量",
        "type": "Integer",
        "values": "{\"unit\":\"L\",\"min\":0,\"max\":1000,\"scale\":0,\"step\":1}"
      },
      {
        "code": "battery_percentage",
        "name": "电池电量",
        "type": "Integer",
        "values": "{\"unit\":\"%\",\"min\":0,\"max\":100,\"scale\":0,\"step\":1}"
      },
      {
        "code": "weather_delay",
        "name": "天气延时",
        "type": "Enum",
        "values": "{\"range\":[\"cancel\",\"24h\",\"48h\",\"72h\"]}"
      },
      {
        "code": "countdown",
        "name": "倒计时",
        "type": "Integer",
        "values": "{\"unit\":\"s\",\"min\":0,\"max\":86400,\"scale\":0,\"step\":1}"
      },
      {
        "code": "work_state",
        "name": "工作状态",
        "type": "Enum",
        "values": "{\"range\":[\"auto\",\"manual\",\"idle\"]}"
      },
      {
        "code": "smart_weather",
        "name": "智能天气类型",
        "type": "Enum",
        "values": "{\"range\":[\"sunny\",\"clear\",\"cloud\",\"cloudy\",\"rainy\",\"snow\",\"fog\"]}"
      },
      {
        "code": "use_time_one",
        "name": "单次使用时间",
        "type": "Integer",
        "values": "{\"unit\":\"s\",\"min\":0,\"max\":86400,\"scale\":0,\"step\":1}"
      }
    ]
  },
  "success": true,
  "t": 1749826661011,
  "tid": "c0f34544486611f0be34e23b0b987d40"
}`

I hope this can help.

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_fphxkxue',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

Nothing works.

alanmgomes, 2025-12-31:
I just got a rejection because of this issue, just sharing my working code for this device:

```javascript
import fz from 'zigbee-herdsman-converters/converters/fromZigbee';
import tz from 'zigbee-herdsman-converters/converters/toZigbee';
import * as exposes from 'zigbee-herdsman-converters/lib/exposes';
import tuya from 'zigbee-herdsman-converters/lib/tuya';

const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_fphxkxue'
        }
    ],
    model: 'ZVL-PRO',
    vendor: 'Nova Digital',
    description: 'Water Valve (Advanced Map)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetLocalTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.switch().setAccess('state', ea.STATE_SET),
        e.battery(),
        e.numeric('countdown', ea.STATE_SET).withUnit('s').withValueMin(0).withValueMax(86400)
            .withDescription('Tempo de contagem regressiva em segundos'),
        e.enum('work_state', ea.STATE, ['auto', 'manual', 'idle'])
            .withDescription('Estado atual de opera    o'),
        e.numeric('water_once', ea.STATE).withUnit('L').withDescription('Consumo da   ltima rega'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff],
            [7, 'battery', tuya.valueConverter.raw],
            [11, 'countdown', tuya.valueConverter.raw],
            [12, 'work_state', tuya.valueConverter.raw], 
            [5, 'water_once', tuya.valueConverter.raw],
            [4, 'fault', tuya.valueConverter.raw],
        ],
    },
};

export default definition;

```

guigomesa, 2026-10-19:
Tenho uma _TZE200_fphxkxue (Nova Digital ZVL-PRO, Zigbee module 1.0.1, Z2M 2.14.2) e testei os datapoints que a definição atual não expõe:

- DP17: agenda interna, no mesmo formato do RTX ZVG1. Adicionar 00 03 HHMM duração dias 40 ativo 07e6080101, apagar 00 04 <espaço 0–3>. A resposta traz a lista completa: 00 + blocos de 13 bytes, cada um começando pelo espaço. Gravei 09:02/1 min e a válvula abriu sozinha e fechou após 60 s.
- DP15: duração da última rega, em segundos.
- DP101: litros da rega em andamento.
- DP12: 0 = rodando pela agenda, 1 = manual, 2 = parada.
- DP10: adiamento por chuva, enum 0–3.

Dois comportamentos que afetam qualquer escrita: com a válvula ociosa os comandos não chegam, porque ela só escuta logo depois de transmitir. E um comando recebido com o motor em movimento travou a válvula duas vezes, e ela só voltou depois de ficar sem pilha.

Testei isso com um conversor externo escrito com ajuda do Claude Code.

Com a definição atual, abrir e fechar pelo Z2M só funciona logo depois de apertar o botão da válvula. Abrir sob demanda com ela parada não é viável, porque ela dorme. Durante a rega ela aceita comandos, então fechar pelo Home Assistant funciona.

A limitação para alterar o agendamento é aguardar o próximo agendamento ou forçar isso abrindo manualmente (pressionando o botão diretamente na válvula). Não é o modo mais prático, mas permitiu usar o Home Assistant para criar agendamentos e saber o volume de água usado durante as regas.

Posso abrir um PR adicionando esses campos à definição da ZVL-PRO?

<details>
<summary>English (AI translation)</summary>

I have a _TZE200_fphxkxue (Nova Digital ZVL-PRO, Zigbee module 1.0.1, Z2M 2.14.2) and tested the datapoints the current definition doesn't expose:

- DP17: internal schedule, same format as the RTX ZVG1. Add 00 03 HHMM duration days 40 active 07e6080101, delete 00 04 <slot 0–3>. The response carries the full list: 00 + 13-byte blocks, each starting with the slot. I wrote 09:02/1 min and the valve opened by itself and closed after 60 s.
- DP15: duration of the last watering, in seconds.
- DP101: liters of the watering in progress.
- DP12: 0 = running from schedule, 1 = manual, 2 = idle.
- DP10: rain delay, enum 0–3.

Two behaviors that affect any write: while the valve is idle, commands don't arrive, because it only listens right after it transmits. And a command received while the motor was moving froze the valve twice; it only recovered after losing battery power.

I tested this with an external converter written with help from Claude Code.

With the current definition, opening and closing from Z2M only works right after pressing the valve's button. Opening on demand while it's idle isn't feasible because it sleeps. During watering it accepts commands, so closing from Home Assistant works.

The limitation for changing the schedule is waiting for the next scheduled run or forcing it by opening the valve manually (pressing the button on the valve). It's not the most practical way, but it made it possible to use Home Assistant to create schedules and track the volume of water used during watering.

Can I open a PR adding these fields to the ZVL-PRO definition?

</details>