Koenkk/zigbee2mqtt #27719: Starkvind: fan speed not reported in auto mode in main state topic

MB901, 2025-06-14:
### What happened?

Hello,

When the IKEA Starkvind is in "auto" mode, Zigbee2MQTT no longer includes the fan speed in the main reported state topic. However, the fan speed continues to vary in the background. Other integrations, like [Home Assistant Powercalc](https://github.com/bramstroker/homeassistant-powercalc), rely on the fan_speed to calculate power usage.

In manual mode, fan_speed is reported as expected. 

### What did you expect to happen?

fan_speed should always be included in the device’s state, even when mode is set to auto, so that the current fan speed can be monitored regardless of the mode.

### How to reproduce it (minimal and precise)

- Add an IKEA Starkvind to Zigbee2MQTT.

- Set mode to "auto" via MQTT or the frontend.

- Observe the state topic (e.g., zigbee2mqtt/starkvind_air_purifier_table) — the fan speed is missing.

- Set mode back to "manual" and adjust fan speed.

- Now the fan speed appears in the state topic again.

### Zigbee2MQTT version

2.4.0

### Adapter firmware version

20221226

### Adapter

ZStack3x0

### Setup

Add-on on HAOS.

Wikibear, 2025-06-15:
interesting, I have an old one without any problems. Are you using E2007?

[image]

MB901, 2025-06-15:
This is the view from mqtt explorer : 
Manual mode : 
`{"air_quality":"excellent","child_lock":"UNLOCK","device_age":null,"fan_mode":"2","fan_speed":2,"fan_state":"ON","filter_age":23510,"identify":null,"last_seen":"2025-06-15T19:18:25.195Z","led_enable":false,"linkquality":138,"pm25":6,"replace_filter":false,"update":{"installed_version":69633,"latest_version":69633,"state":"idle"}}`
Auto mode : 
`{"air_quality":"excellent","child_lock":"UNLOCK","device_age":null,"fan_mode":"auto","fan_speed":5,"fan_state":"ON","filter_age":23510,"identify":null,"last_seen":"2025-06-15T19:19:36.696Z","led_enable":false,"linkquality":142,"pm25":6,"replace_filter":false,"update":{"installed_version":69633,"latest_version":69633,"state":"idle"}}`

Wikibear, 2025-06-15:
Fan speed is on five in this data. So zigbee2mqtt send correct data. Now I can't support further I don't have homeassist.