Koenkk/zigbee2mqtt #27723: HAOS Add-on: Frontend Port Configuration Not Persisting for Multiple Instances

vgisbert, 2025-06-14:
### What happened?

When running multiple instances of the Zigbee2MQTT add-on in Home Assistant OS (HAOS), the frontend.port configuration, when changed via the add-on's GUI settings, does not persist after saving and restarting the add-on or the HAOS host. The port value reverts to its default or previous value (e.g., 8099), leading to port conflicts if multiple instances attempt to use the same port.
This issue has been reproduced on multiple clean Home Assistant OS virtual machines running on Proxmox, suggesting it is not an isolated corruption but potentially a bug in how the add-on's configuration is handled by the Home Assistant Supervisor or the add-on itself when multiple instances are in use.

### What did you expect to happen?

The frontend.port value, when changed in the add-on's configuration GUI and saved, should persist and be applied correctly after a restart of the add-on or Home Assistant OS. Each instance should be able to run its frontend on a unique, user-defined port.

### How to reproduce it (minimal and precise)

The frontend.port value does not persist when changed in the add-on's GUI configuration for the second (or subsequent) instance, reverting to 8099. This prevents running multiple Zigbee2MQTT frontend GUIs simultaneously due to port conflicts. The core MQTT functionality of Zigbee2MQTT (device communication, MQTT publishing) remains operational, but the web interface for the conflicting instance is inaccessible.
Troubleshooting Performed
•	Verified configuration.yaml: Confirmed that the frontend.port setting was correctly defined in the add-on's mounted configuration.yaml (e.g., /config/zigbee2mqtt_X/configuration.yaml), but the add-on appears to ignore this setting for the frontend port at startup, likely due to an overriding parameter from the Supervisor. 
o	Example from provided configuration.yaml (instance 1, intended for zigbee2mqtt_1): 
YAML
frontend:
  enabled: true
  port: 8099 # This was intended to be changed to 8098 for instance 1
•	Full HAOS Host Restart: Performed multiple full restarts of the Home Assistant OS host to ensure configuration changes are reloaded.
•	Direct options.json access (via Advanced SSH & Web Terminal - non-protected mode): 
o	Attempted to locate and directly modify options.json files within /mnt/data/supervisor/addons/data/<ADDON_ID>/ or /addon_configs/<ADDON_ID>/.
o	find / -name "options.json" only returned /data/options.json (which is for the SSH add-on itself).
o	ls -la /mnt/data/supervisor/addons/data/ or ls -la /addon_configs/ showed empty directories or no expected options.json files, indicating the persistent configuration for the frontend port is not exposed in these user-accessible locations or is managed differently.
o	This suggests the port setting might be injected as an environment variable or hardcoded/overridden by the Supervisor during container creation, bypassing standard configuration file mechanisms for this specific parameter.
•	Replication on Fresh VM: The issue was replicated on a brand-new Home Assistant OS virtual machine, suggesting it's not specific to the initial installation's data corruption.

### Zigbee2MQTT version

2.4.0-1

### Adapter firmware version

20250325

### Adapter

SLZB-06P10

### Setup

•	Home Assistant Core: homeassistant: 2025.6.1 
•	Home Assistant OS: Home Assistant OS 15.2 
•	Zigbee2MQTT Add-on Version: 2.4.0-1 
•	Installation Method: Home Assistant OS running as a VM on Proxmox.
•	Coordinators: Remote TCP/IP coordinators SLZB-06P10 Firmware: 20250325
•	MQTT Broker: Mosquitto add-on 6.5.1

configuration.yaml:

version: 4
mqtt:
  base_topic: zigbee2mqtt_1
  server: mqtt://core-mosquitto:1883
  user: addons
  password: xxx
serial:
  rtscts: false
  port: tcp://10.0.0.2:6639
  adapter: zstack
  baudrate: 115200
advanced:
  restart_policy: always
  transmit_power: 20
  log_level: info
  channel: 12
  network_key:
    - 77
    - 134
    - 111
    - 57
    - 139
    - 19
    - 42
    - 49
    - 121
    - 36
    - 226
    - 115
    - 3
    - 183
    - 164
    - 180
  pan_id: 1234
  ext_pan_id:
    - 9
    - 183
    - 85
    - 158
    - 228
    - 129
    - 255
    - 4
frontend:
  enabled: true
  port: 8099  # <-- IT IS IMPOSIBLE TO CHANGE
homeassistant:
  enabled: true
devices:

Koenkk, 2025-06-15:
This is expected, otherwise the ingress wont work. Why do you want to change the port?

vgisbert, 2025-06-18:
> This is expected, otherwise the ingress wont work. Why do you want to change the port?

Hello, first of all, thank you very much for replying.
Sorry, but I think it was a misunderstanding.
I'm installing two Smlight SLZB-06 remote coordinators, and I followed the steps indicated by the manufacturer at:
https://smlight.tech/manual/slzb-06/guide/multiple-adapters-setup/
The demo video:
https://www.youtube.com/watch?v=QOx733CU1p8&t=286s 
indicates changing the port.
Now I realize that in the addon's GUI application, activating the "Show disabled ports" button shows two ports:
- Socat tcp-listen port (8485/tcp)
- Frontend tcp-listen port (8099/tcp)
I assumed the video indicated that both ports should be changed, but when I saw that despite changing the frontend, both in the configuration.yaml file before starting the addon and in the GUI, when starting the addon, the frontend always returned to 8099.
I understood that this "unusual behavior" was a bug, which could cause problems if two addon windows were open, one with one coordinate and the other with the other.
Hence the reason for this issue.
But maybe I'm wrong and it doesn't affect anything.

closed 2025-08-26 by github-actions[bot]