Koenkk/zigbee2mqtt #27724: Multiple ThirdReality devices are showing MAINS powered instead of BATTERY. Causing availability flooding!

Emerica243, 2025-06-15:
### What happened?

First time Z2M user here. I started migrating a handful of my devices over from ZHA and alot of my ThirdReality battery powered devices are showing mains powered, when they are actually battery powered. Im not sure what other impact this has other then they keep showing offline because its using the availability settings thinking its a mains powered device, when its not. 

### What did you expect to happen?

Battery powered devices to come in as battery powered, not mains.

### How to reproduce it (minimal and precise)

Pair one of the following devices and it will show mains, even though theyre battery powered devices. 

Third Reality Smart Button - 3RSB22BZ
Third Reality Thermometer - 3RTHS24BZ

### Zigbee2MQTT version

2.4.0

### Adapter firmware version

20240710

### Adapter

SLZB06

### Setup

Add-on in HAOS

Koenkk, 2025-06-15:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

After updating reconfigure the devices by pressing the yellow refresh button (make sure it succeeds by waking-up the device before doing so) and restart z2m.

closed 2025-06-15 by Koenkk

3reality-support, 2025-07-02:
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const {deviceAddCustomCluster} = require('zigbee-herdsman-converters/lib/modernExtend');
const m = require('zigbee-herdsman-converters/lib/modernExtend');

module.exports = [{
    zigbeeModel: ["3RSB22BZ"],
    model: "3RSB22BZ",
    vendor: "Third Reality",
    description: "Smart button",
    fromZigbee: [fz.itcmdr_clicks],
    ota: true,
    exposes: [e.action(["single", "double", "hold", "release"])],
    extend: [
        m.battery({percentageReporting: false}),
        m.forcePowerSource({powerSource: "Battery"}),
    ],
}, ];

Sandriell, 2025-07-24:
Both devices reported are still showing as mains powered.

Version: 2.5.1-1

Emerica243, 2025-07-25:
Im a bit hesitant to perform an update from 2.4.0 to latest 2.5.1 based on what was last said a few posts above by [@a-k-home-assistant-integration 

Being new to z2m, and what hes mentioned, should i be concerned with my TR devices breaking after i update?

Sandriell, 2025-07-26:
> You will have to re-configure the device by pressing the refresh button the device about page to fix this (and then restart z2m).

Yeah, that fails, as previously reported.

`2025-07-26 01:58:46z2m: Failed to configure 'Button', attempt 4 (Error: ZCL command 0x282c02bfffee97cd/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:363:28) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:987:26) at processTicksAndRejections (node:internal/process/task_queues:105:5) at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:707:9) at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:160:17) at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:221:17) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21) at Configure.configure (/app/lib/extension/configure.ts:123:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:46:25) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))`

3reality-support, 2025-07-31:
Hello, we have received your feedback and the relevant developers are currently fixing this issue. Other 3R products in the future will fix this bug

3reality-support, 2025-08-04:
No, the button can work normally on z2m. Is there any problem encountered now

> Does that mean that the current Smart Buttons that I just bought will never work correctly with Z2M?

sienar1, 2025-08-31:
> No, the button can work normally on z2m. Is there any problem encountered now
> 
> > Does that mean that the current Smart Buttons that I just bought will never work correctly with Z2M?

Yes, the same problem is still happening. I don't know why this issue got closed. The suggested fixes do not work.

The buttons do not work in the current version of Z2M. When pairing the button to Z2M 2.6.0 it reports as Mains Powered and fails to configure.  It works for a while and then if you don't use it for some time Z2M reports it as dead because it expects regular regular activity in under 10 minutes for Mains Powered devices. 

`[2025-08-31 17:08:50] z2m: Failed to configure '0xffffb40e06007ca2', attempt 3 (Error: ZCL command 0xffffb40e06007ca2/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:372:28)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:1005:26)
    at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:727:9)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:160:17)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:221:17)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21)
    at Configure.configure (/app/lib/extension/configure.ts:123:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))`

I see a commit was made 2 weeks ago that I believe forces Z2M to treat the button as battery powered. Is that going to be included in the September release of Z2M?

Sandriell, 2025-10-08:
@3reality-support On the October release now and this issue has still not been corrected.

S-Z, 2025-11-03:
I had been experiencing similar issues with a 3RSB22BZ smart button but some combination of updating Z2M and an OTA update of the button's firmware appears to have finally fixed it. I'm unsure exactly which of the steps I took were responsible for solving the issue, so I'll list them all.

- Up to and including Z2M 2.6.2, attempting to reconfigure resulted in the "Failed to configure" error reported by others. Z2M reported that an OTA update was available, but I was unable to start the update.
- After updating to Z2M 2.6.3, reconfiguration still failed but the OTA update now worked. I updated my 3RSB22BZ's firmware to v47.
- I performed the reconfiguration operation again after updating the device firmware and it completed successfully. The button now correctly displays a battery as its power source.

[image]