Koenkk/zigbee2mqtt #27731: switches seen but not working

reinierdevl, 2025-06-15:
### What happened?

When I update from 1.43 to 2.3 everything seems to work fine, untill I try my switches. They are seen when I make the map but when I look at the device status is says it is disabled. Also I do not get any signals in in Node-Red from the switch...

I added all required lines in configuration.yaml, also tried legacy mode all without result...

Is there a solution for this?

[image]
[image]

### What did you expect to happen?

The switched were working

### How to reproduce it (minimal and precise)

Update from 1.41 to 2.x

### Zigbee2MQTT version

when going from 1.41 to 2.4

### Adapter firmware version

20230507

### Adapter

sonoff Dongle P

### Setup

Running HAOS on HP thinclient 630

Razer0ne, 2025-06-15:
Try to set availability to ennabled,  have you test it?

reinierdevl, 2025-06-16:
I can not change the availabillity in Z2M...

> Try to set availability to ennabled, have you test it?

reinierdevl, 2025-06-18:
anybody?

closed 2025-08-26 by github-actions[bot]