Koenkk/zigbee2mqtt #27748: Amina S problem reading charge_limit

jondavidsson, 2025-06-17:
### What happened?

When trying to read charge_limit from Amina S charger (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/amina.ts) I get an error when using latest firmware on the charger.

Response from Amina S developers with suggested solution:

> The software developer does not have the same adapter. He has sonoff adapter. He could reproduce the problem that you have experienced.
> 
> The cause for the problem is to use Manufacturer specific flag in Framecontrol-field of ZCL frame. The solution to the problem is changing Zigbee2MQTT code.
> There is a plugin for Amina-S that someone has developed. This plugin handles the control for Amina S over Zigbee.
> 
> Our software developer used "Raspberri Pi Linux " to recreate the problem and verify the fix to the problem. He has used the latest available version of Zigbee2MQTT. (2.4.0 commit: bdb94da4) .
> The path for the Amina-S plugin is " [/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@23.53.0](mailto:/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@23.53.0)/node_modules/zigbee-herdsman-converters/dist/devices/amina.js". Note that the path might be different in your case.
> 
> Existing code used manufacturerOptions. That is not needed.
> await [entity.read](http://entity.read/)("genLevelCtrl", ["currentLevel"], manufacturerOptions);
> 
> change that to
> await [entity.read](http://entity.read/)("genLevelCtrl", ["currentLevel"]);
> Please find the good-case and bad-case wireshark capture. Please find the the log for successful read.

### What did you expect to happen?

I expected to get a read of the current charge_limit.

### How to reproduce it (minimal and precise)

Refresh state in for example Zigbee2MQTT Home assistant front end.

### Zigbee2MQTT version

2.4.0

### Adapter firmware version

20240710

### Adapter

Electrolama zig-a-sig-ah!

### Setup

Add-on on Home assistant OS on a Raspberry Pi4

### Debug log

```
error 2025-06-17 08:28:24z2m: Exception while calling fromZigbee converter: Value is not a number, got undefined (undefined)}
error 2025-06-17 08:28:24z2m: Publish 'get' 'charge_limit' to 'Amina S' failed: 'Error: ZCL command 0xf4ce36546ae44f45/10 genLevelCtrl.read(["currentLevel"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":5179,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
```

thorgan3, 2025-06-23:
Solved: https://github.com/Koenkk/zigbee-herdsman-converters/pull/9534

Koenkk, 2025-06-24:
Thanks for the fix @thorgan3 !

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-06-24 by Koenkk