Koenkk/zigbee2mqtt #27750: IKEA INSPELNING E2206 does not report solar panels instant power.

ALaDoffe, 2025-06-17:
### What happened?

Hello 

I don't know if it is an issue with the plug itself or Z2M.
I bought this plug to follow my solar production.

When used with a bulb or toaster.... the plug reports correctly the instant power, current, voltage, energy.
But when plugged behind my solar panels all metrics are correct except the instant power that stays at 0W.

[image]

### What did you expect to happen?

Solar panel instant power must be reported.

### Zigbee2MQTT version

2.4.0

### Adapter firmware version

20240710

### Adapter

CC2652P

### Setup

Add-on on Home Assistant OS on mini C

### Debug log

[2025-06-17 16:14:06] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x94a081fffe5ede9f', payload '{"current":1.82,"energy":82.783,"identify":null,"linkquality":109,"power":0,"power_on_behavior":"on","state":"ON","update":{"installed_version":33816645,"latest_version":33816645,"state":"idle"},"voltage":230.4}'

[2025-06-17 16:14:14] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x94a081fffe5ede9f', payload '{"current":1.82,"energy":82.783,"identify":null,"linkquality":123,"power":0,"power_on_behavior":"on","state":"ON","update":{"installed_version":33816645,"latest_version":33816645,"state":"idle"},"voltage":229.1}'

[2025-06-17 16:14:16] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x94a081fffe5ede9f', payload '{"current":1.85,"energy":82.783,"identify":null,"linkquality":120,"power":0,"power_on_behavior":"on","state":"ON","update":{"installed_version":33816645,"latest_version":33816645,"state":"idle"},"voltage":229.1}'

Nerivec, 2025-06-20:
Plugs like this aren't meant for production metering (other way around).
You should look into something like a clamp: https://www.zigbee2mqtt.io/supported-devices/#s=clamp

tsbwn, 2025-06-29:
Interesting. I've bought recently 2 INSPELNING's and they both report energy, power, voltage as 0 despite given load.  Only current is reported.
Example of connected treadmill:
```
{
    "current": 1.11,
    "energy": 0,
    "linkquality": 108,
    "power": 0,
    "power_on_behavior": "previous",
    "state": "ON",
    "update": {
        "installed_version": 33816645,
        "latest_version": 33816645,
        "state": "idle"
    },
    "voltage": 0,
    "identify": null
}
```
The devices detected as `E2206` with firmware (OTA tab) `0.2.04-0.0.45` (no new versions at the moment).
Firmware ver from plug About page: Firmware build date `20240801`, Firmware version `2.4.45`
Z2M version: 
```
Zigbee2MQTT version: 2.4.0 commit: unknown
Coordinator type: ZStack3x0
Coordinator revision: 20210708
Coordinator IEEE Address: <removed>
Frontend version: 0.9.13
zigbee-herdsman-converters version: 23.53.0
zigbee-herdsman version: 4.1.0
```

alex0411, 2025-07-27:
Hi, does anybody has some updates on this?

closed 2025-10-04 by github-actions[bot]