Koenkk/zigbee2mqtt #27754: TS0601_water_switch works partially, does not switch

fenyvesi, 2025-06-18:
### What happened?

Z2M recognizes the device correctly, i can see the real information from the device, BUT not able the control it. If I switch L1 or L2, it sends messages but the device is not switched in reality.

If I switch the device manually, then I can see the result in HA Z2M.

I found a line in the log once:
error 2025-06-18 03:05:02z2m: Publish 'set' 'state' to 'Double_valve' failed: 'Error: No datapoint defined for 'state''

### How to reproduce it (minimal and precise)

Switch in HA, check the device.

### Zigbee2MQTT version

Bridge by Zigbee2MQTT 2.4.0 in HA OS

### Adapter firmware version

14.6.0.2 (c10686e-zbbrdgpro) , Tasmota TCP

### Adapter

Sonoff Zigbee Bridge Pro, Tasmota TCP

### Setup

Add-on on Home Assistant OS, x86-64

### Debug log

`[2025-06-18 11:08:17] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Double_valve', payload '{"battery":null,"countdown_l1":10,"countdown_l2":10,"linkquality":156,"state_l1":"ON","state_l2":"OFF","valve_duration_l1":12,"valve_duration_l2":11,"valve_status_l1":"idle","valve_status_l2":"idle","voltage":null}'
`

`June 18, 2025
turned on triggered by action Switch: Turn on
11:11:48 - In 1 second - György Fenyvesi
turned on triggered by action Switch: Turn on
11:08:17 - 4 minutes ago - György Fenyvesi`

ThorstenDtk, 2025-06-19:
I also managed to get it work. I used the external converter from https://github.com/Koenkk/zigbee2mqtt/issues/27591:

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

module.exports = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE284_fhvpaltk',
        },
    ],
    model: 'TS0601_valve_custom',
    vendor: 'Tuya',
    description: 'Custom working dual water valve with _l1/_l2 naming',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.binary('valve_l1', ea.STATE_SET, 'ON', 'OFF'),
        e.binary('valve_l2', ea.STATE_SET, 'ON', 'OFF'),
        e.numeric('countdown_l1', ea.STATE_SET).withValueMin(0).withValueMax(255).withUnit('min'),
        e.numeric('countdown_l2', ea.STATE_SET).withValueMin(0).withValueMax(255).withUnit('min'),
        e.enum('valve_status_l1', ea.STATE, ['idle', 'manual', 'remote']),
        e.enum('valve_status_l2', ea.STATE, ['idle', 'manual', 'remote']),
        e.numeric('battery', ea.STATE).withUnit('%'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'valve_l1', tuya.valueConverter.onOff],
            [2, 'valve_l2', tuya.valueConverter.onOff],
            [13, 'countdown_l1', tuya.valueConverter.countdown],
            [14, 'countdown_l2', tuya.valueConverter.countdown],
            [104, 'valve_status_l1', tuya.valueConverterBasic.lookup({
                idle: tuya.enum(2),
                manual: tuya.enum(0),
                remote: tuya.enum(1),
            })],
            [105, 'valve_status_l2', tuya.valueConverterBasic.lookup({
                idle: tuya.enum(2),
                manual: tuya.enum(0),
                remote: tuya.enum(1),
            })],
            [59, 'battery', tuya.valueConverter.raw],
        ],
    },
};
```

But note that the path mentioned there is incorrect:

[image]

external_converters must be in the data-folder where the configuration.yaml is located.

Koenkk, 2025-06-19:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

rbb-dev, 2025-06-30:
@Koenkk, how do you view custom FZs?
The fault DP103 only transmits if there is a fault.
If there is no fault, it is displayed as N/A or Null in the UI and only populated if a fault actually exists.
with this helper/hack it sets to 'ok'.

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const fzNullIsOk = {
    cluster: 'manuSpecificTuya',
    type: ['commandDataReport', 'commandDataResponse', 'attributeReport'],
    event: 'deviceAnnounce',
    convert: (_, __, publish, ___, meta) => {
        if (meta.state?.fault == null) publish({fault: 'ok'});
    },
};

module.exports = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE284_fhvpaltk']),
    model: 'TS0601_water_switch',
    vendor: 'Tuya',
    description: 'Dual-channel Zigbee water valve',
    icon: 'https://www.zigbee2mqtt.io/images/devices/TS0601_water_switch.png',

    fromZigbee: [tuya.fz.datapoints, fzNullIsOk],
    toZigbee:   [tuya.tz.datapoints],

    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,

    exposes: [
        e.switch().withDescription('Valve 1 on/off').withEndpoint('l1'),
        e.switch().withDescription('Valve 2 on/off').withEndpoint('l2'),
        e.enum('valve_status', ea.STATE, ['manual', 'auto', 'idle']).withDescription('Valve 1 current mode').withEndpoint('l1'),
        e.enum('valve_status', ea.STATE, ['manual', 'auto', 'idle']).withDescription('Valve 2 current mode').withEndpoint('l2'),
        e.numeric('countdown', ea.STATE_SET).withDescription('Valve 1 countdown (minutes)').withUnit('min').withValueMin(0).withValueMax(1440).withEndpoint('l1'),
        e.numeric('countdown', ea.STATE_SET).withDescription('Valve 2 countdown (minutes)').withUnit('min').withValueMin(0).withValueMax(1440).withEndpoint('l2'),
        e.numeric('valve_duration', ea.STATE).withDescription('Valve 1 last run duration (seconds)').withUnit('s').withEndpoint('l1'),
        e.numeric('valve_duration', ea.STATE).withDescription('Valve 2 last run duration (seconds)').withUnit('s').withEndpoint('l2'),
        e.enum('fault', ea.STATE, ['ok', 'flow_fault', 'motor_stuck']).withDescription('Fault status'),
        e.battery(),
    ],

    meta: {
        tuyaDatapoints: [
            [1,   'state_l1',           tuya.valueConverter.onOff],
            [2,   'state_l2',           tuya.valueConverter.onOff],
            [13,  'countdown_l1',       tuya.valueConverter.raw],
            [14,  'countdown_l2',       tuya.valueConverter.raw],
            [25,  'valve_duration_l1',  tuya.valueConverter.raw],
            [26,  'valve_duration_l2',  tuya.valueConverter.raw],
            [59,  'battery',           tuya.valueConverter.raw],
            [103, 'fault',             tuya.valueConverterBasic.lookup({flow_fault: 1, motor_stuck: 2}, 'ok')],
            [104, 'valve_status_l1',    tuya.valueConverterBasic.lookup({manual:0, auto:1, idle:2})],
            [105, 'valve_status_l2',    tuya.valueConverterBasic.lookup({manual:0, auto:1, idle:2})],
        ],
        multiEndpoint: true,
    },
    endpoint: () => ({l1: 1, l2: 1}),
};
```

Koenkk, 2025-06-30:
I'm not sure if that is correct, there could already be a fault (although it may not be very likely) before the device pairs. If we would then publish OK it would result in a wrong state. Therefore it is better to use N/A when the state is unknown (better than providing an incorrect state)