Koenkk/zigbee2mqtt #27756: [New device support]: SOIL MOISTURE TS0601_TZE284_33bwcga2

ifajardoj, 2025-06-18:
### Link

https://es.aliexpress.com/item/1005009041844879.html?spm=a2g0o.order_list.order_list_main.11.e751194dzwxsva&gatewayAdapt=glo2esp

### Database entry

{"id":30,"type":"EndDevice","ieeeAddr":"0xa4c138d06e6fd2fb","nwkAddr":27867,"manufId":4417,"manufName":"_TZE284_33bwcga2","powerSource":"Battery","modelId":"TS0601","epList":>

### Zigbee2MQTT version

2.4.0

### Comments

It is a Tuya soli moisture sensor similar to other supported.

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_33bwcga2',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

Nothing is exposed aside from the link quality.

Koenkk, 2025-06-21:
Could you try to:

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZE284_33bwcga2 ` to `_TZE200_myd45weu`
- Start Z2M and see if the device now works
- If it doesn't work, instead of `_TZE200_myd45weu` try with: `_TZE284_g2e6cpnw` or `_TZE284_aao3yzhs`

CocoLeo3680, 2025-06-23:
> The last one seems to work, although the battery percentage is fixed at 60% after 24 hours and the batteries were new...

I haven't been able to change it. Could you explain step by step how to do it? Thanks.

ifajardoj, 2025-06-23:
- Stop Z2M addon
- Edit your database.db file (verify the path in Z2M addon configuration data_path parameter, mine is /share/zigbee2mqtt)
- change all occurrences of _TZE284_33bwcga2  to _TZE284_aao3yzhs, twice in my case
- Start Z2M

CocoLeo3680, 2025-06-23:
> * _TZE284_33bwcga2

Ok, many  thanks. Now it has been detected
My datapath is in /homeassistant/zigbee2mqtt/database.db

My default code is:
`{"id":46,"type":"EndDevice","ieeeAddr":"0xa4c138ca9e4a8453","nwkAddr":61776,"manufId":4417,"manufName":"_TZE284_33bwcga2","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"���/f���/\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":80}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1750672247144}`

So there is only one. The new code:
`{"id":46,"type":"EndDevice","ieeeAddr":"0xa4c138ca9e4a8453","nwkAddr":61776,"manufId":4417,"manufName":"_TZE284_aao3yzhs","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"���/f���/\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":80}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1750672247144}`

Battery is too at 60%

Many thanks for your indications

CocoLeo3680, 2025-07-04:
Now is not working. Zigbe2mqtt version 2.5.1 
:(

Thanks

Heiherm, 2025-07-29:
Hi, I have the same problem:
Version 2.5.1
_TZE284_aao3yzhs
LinkQ and battery only, constant temperature 31°C Hum 0
_TZE200_myd45weu and _TZE284_g2e6cpnw have the same problem.

Thanks

closed 2025-10-06 by github-actions[bot]