Koenkk/zigbee2mqtt #27774: [Wrong device]: eWelink MS24032401 rain sensor detected as eWelink  NZB-05

nmaster2042, 2025-06-20:
### Link

https://fr.aliexpress.com/item/1005009030356992.html?spm=a2g0o.order_list.order_list_main.107.d9cc5e5b1Mmuse&gatewayAdapt=glo2fra

### Model

MS24032401

### Description

EWelink Zigbee rain sensor

### Vendor

EWelink

### Picture (link)

https://ae01.alicdn.com/kf/Sd6b9e73e6e1048699406a1bf158f3a9e0.png

### Database entry

{"id":34,"type":"EndDevice","ieeeAddr":"0xa4c1382176c00c08","nwkAddr":39465,"manufId":4742,"manufName":"eWeLink","powerSource":"Battery","modelId":"SNZB-05","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":158,"batteryVoltage":27}},"genBasic":{"attributes":{"modelId":"SNZB-05","manufacturerName":"eWeLink","powerSource":3,"zclVersion":3,"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0024c9f4f9","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":14400}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c9f4f9","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c9f4f9","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1750456875454,"checkinInterval":3600}

### Notes

Device is working, only Model and Picture are wrong

nmaster2042, 2025-07-02:
I don't know if device is exactly same because mine is actually working.
It's detected as water leak but if it rains the value is transmitted correctly and same when rain stops.

waterchill, 2025-07-02:
> I don't know if device is exactly same because mine is actually working. It's detected as water leak but if it rains the value is transmitted correctly and same when rain stops.

what values do you get? I get only leakage is detected or not

nmaster2042, 2025-07-02:
Yes leakage true if it rains, false otherwise.

I think the only issue is the device is detected as a water leakage detector instead of a rain detector.

closed 2025-09-09 by github-actions[bot]