Koenkk/zigbee2mqtt #27792: [New device support]: Solar smart soil sensor

b3nighted, 2025-06-23:
### Link

https://es.aliexpress.com/item/1005008785131043.html

### Database entry

{"id":16,"type":"EndDevice","ieeeAddr":"0xa4c138056f99018b","nwkAddr":22223,"manufId":4417,"manufName":"_TZE284_awepdiwi","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"m  /fn  /\u0012n  /ez  /iz  /i{  /i{  /i\u0000\u0000\u0000\u0000\u0011\u0000\u0000\u0000\u0000\u0011","65506":56,"65508":1,"appVersion":77,"stackVersion":0,"dateCode":""}},"manuSpecificTuya":{"attributes":{"0":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1299955489},"lastSeen":1750675247582}

### Zigbee2MQTT version

2.4.0

### Comments

I have tried using a script to wake it up, searched tuya wake-up signals, waited for several hours. All I get is link strength and nothing more. I ended up with a "debug" external converter made by a LLM to try and get ANYTHING out of it and nothing. Tried fresh, fully charged NIMH batteries, tried leaving it in the sun in or out of a glass of water.

### External definition

```shell
export default [
  {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE284_awepdiwi'}],
    model: 'TS0601_soil',
    vendor: 'Tuya',
    description: 'Soil sensor (raw debug mode)',
    fromZigbee: [function (type, data, device) {
      if (type === 'message') {
        console.log("🔍 Raw Tuya message:", JSON.stringify(data, null, 2));
      }
      return {};
    }],
    toZigbee: [],
    exposes: [],
    onEvent: async (type, data, device) => {
      if (type === 'deviceAnnounce' || type === 'deviceInterview') {
        try {
          const endpoint = device.getEndpoint(1);
          await endpoint.read('manuSpecificTuya', [0x0000]);
          console.log("✨ Tuya wake-up read sent");
        } catch (e) {
          console.warn("⚠️ Wake-up read failed:", e.message);
        }
      }
    },
    configure: async (device, coordinatorEndpoint) => {
      try {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('manuSpecificTuya', [0x0000]);
        console.log("✅ Configuration read success");
      } catch (e) {
        console.warn("⚠️ Configuration read failed:", e.message);
      }
    },
  },
];
```

### What does/doesn't work with the external definition?

Nothing comes out of the device. All I get is link strength,

Element903, 2025-06-24:
yes i have the same device it takes ages for information to be sent and you cant force a refresh, i think it has to do with saving battery that it only sends info when there is a change or a set time it updates.

jesualdomatos-mds, 2025-06-24:
You have more luck, mine doesn't control anything! As we did, my zigbee2mqqt says it is not supported

Element903, 2025-06-24:
i think the DP's are
3 Soil Moisture
5 Temperature 

it will display the first reading but then wont update to new measurements

duylong, 2025-06-26:
This external can work:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition =   {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE284_awepdiwi',
        },
    ],
    model: 'TS0601',
    vendor: 'Tuya',
    description: 'Soil sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.temperature(),
        e.soil_moisture(),
        e.battery(),
        tuya.exposes.batteryState(),
        tuya.exposes.temperatureUnit(),
        e.detection_interval(),
    ],
    meta: {
        tuyaDatapoints: [
            [3, 'soil_moisture', tuya.valueConverter.raw],
            [5, 'soil_temperature_celsius', tuya.valueConverter.divideBy10],
            [9, 'temperature_unit', tuya.valueConverter.temperatureUnitEnum],
            [14, 'battery_state', tuya.valueConverter.batteryState],
            [15, 'battery', tuya.valueConverter.raw],
            [101, 'soil_temperature_alarm', tuya.valueConverter.raw],
            [102, 'soil_moisture_alarm', tuya.valueConverter.raw],
            [103, 'max_soil_temperature_set_celsius', tuya.valueConverter.raw],
            [104, 'min_soil_temperature_set_celsius', tuya.valueConverter.raw],
            [105, 'max_soil_moisture_set', tuya.valueConverter.raw],
            [106, 'min_soil_moisture_set', tuya.valueConverter.raw],
            [107, 'soil_temperature_sensitivity_celsius', tuya.valueConverter.raw],
            [108, 'soil_moisture_sensitivity', tuya.valueConverter.raw],
            [109, 'detection_interval', tuya.valueConverter.raw],
            [110, 'soil_temperature_fahrenheit', tuya.valueConverter.divideBy10],
            [111, 'soil_temperature_sensitivity_fahrenheit', tuya.valueConverter.raw],
            [112, 'max_soil_temperature_set_fahrenheit', tuya.valueConverter.raw],
            [113, 'min_soil_temperature_set_fahrenheit', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

jumboli, 2025-06-26:
Thanks, a lot ! This looks quite good for me. The only thing is that  temperature does not give back anything, soil_temperature_celsius does though.

b3nighted, 2025-06-26:
@duylong thanks for sharing this. Z2M labels it as invalid and renames it. I am a total noob, this is the first time I'm trying this, I have no idea what to change to make it work.

I'm on latest HAOS and z2m.

Stullemon, 2025-06-29:
Hi all, I am fairly new around these parts but when I ran into this I wondered how to best implement the external converter. I ended up starting from the one @duylong provided and kind of referenced the TS0601_soil_2 from herdsman to merge from either what I found most suitable (also because .temperature() didn't work for me). Here goes:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition =   {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE284_awepdiwi',
        },
    ],
    model: 'TS0601',
    vendor: 'Tuya',
    description: 'Soil sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.numeric("soil_temperature_celsius", ea.STATE).withUnit("°C").withValueMin(-10).withValueMax(60).withDescription("Soil temperature"),
        e.numeric("soil_temperature_fahrenheit", ea.STATE).withUnit("°F").withValueMin(14).withValueMax(140).withDescription("Soil temperature"),
        e
            .numeric("soil_temperature_sensitivity_celsius", ea.STATE_SET)
            .withUnit("°C")
            .withValueMin(0.3)
            .withValueMax(1)
            .withValueStep(0.1)
            .withDescription("Temperature sensitivity"),
        e
            .numeric("soil_temperature_sensitivity_fahrenheit", ea.STATE_SET)
            .withUnit("°F")
            .withValueMin(0.3)
            .withValueMax(1)
            .withValueStep(0.1)
            .withDescription("Temperature sensitivity"),
        e.soil_moisture(),
        e.numeric("soil_moisture_sensitivity", ea.STATE_SET).withUnit("%").withValueMin(1).withValueMax(5).withDescription("Humidity sensitivity"),
        e.enum("soil_temperature_alarm", ea.STATE, ["lower_alarm", "upper_alarm", "cancel"]).withDescription("Temperature alarm state"),
        e.enum("soil_moisture_alarm", ea.STATE, ["lower_alarm", "upper_alarm", "cancel"]).withDescription("Humidity alarm state"),
        e
            .numeric("max_soil_temperature_set_celsius", ea.STATE_SET)
            .withUnit("°C")
            .withValueMin(0)
            .withValueMax(60)
            .withDescription("Upper temperature limit"),
        e
            .numeric("min_soil_temperature_set_celsius", ea.STATE_SET)
            .withUnit("°C")
            .withValueMin(0)
            .withValueMax(60)
            .withDescription("Lower temperature limit"),
        e.numeric("max_soil_moisture_set", ea.STATE_SET).withUnit("%").withValueMin(0).withValueMax(100).withDescription("Upper humidity limit"),
        e.numeric("min_soil_moisture_set", ea.STATE_SET).withUnit("%").withValueMin(0).withValueMax(100).withDescription("Lower humidity limit"),
        e.numeric("detection_interval", ea.STATE_SET).withUnit("min").withValueMin(5).withValueMax(60).withDescription("Report sensitivity"),
        e.battery(),
        tuya.exposes.batteryState(),
        tuya.exposes.temperatureUnit(),
    ],
    meta: {
        tuyaDatapoints: [
            [3, 'soil_moisture', tuya.valueConverter.raw],
            [5, 'soil_temperature_celsius', tuya.valueConverter.divideBy10],
            [9, 'temperature_unit', tuya.valueConverter.temperatureUnitEnum],
[... 20 lines cut ...]
            [103, 'max_soil_temperature_set_celsius', tuya.valueConverter.divideBy10],
            [104, 'min_soil_temperature_set_celsius', tuya.valueConverter.divideBy10],
            [105, 'max_soil_moisture_set', tuya.valueConverter.raw],
            [106, 'min_soil_moisture_set', tuya.valueConverter.raw],
            [107, 'soil_temperature_sensitivity_celsius', tuya.valueConverter.divideBy10],
            [108, 'soil_moisture_sensitivity', tuya.valueConverter.raw],
            [109, 'detection_interval', tuya.valueConverter.raw],
            [110, 'soil_temperature_fahrenheit', tuya.valueConverter.divideBy10],
            [111, 'soil_temperature_sensitivity_fahrenheit', tuya.valueConverter.divideBy10],
            [112, 'max_soil_temperature_set_fahrenheit', tuya.valueConverter.divideBy10],
            [113, 'min_soil_temperature_set_fahrenheit', tuya.valueConverter.divideBy10],
        ],
    },
};

module.exports = definition;
```

Now, I am not 100% fine with what I came up with, for instance because it's kind of cluttering the Overview page of Home Assistant by default exposing a number of data points as settings that I would otherwise expect in the specific settings portion of Zigbee2MQTT. However, what I kind of wonder is which approach would be more appropriate:

1. Simply extend fingerprints of TS0601_soil_2 (it seems like the same device, maybe just minor improvements?)
2. Try to stick as close as possible to TS0601_soil_2 while extending what wasn't there before (kind of what I went for without really understanding how @duylong identified the datapoints they added)
3. Do this in some other way properly - whatever that means...

Any suggestions or guidance?

elemakil, 2025-07-28:
According to the AliExpress item page this sensor can supposedly also measure soil quality presumably through a separate soil conductivity readout. I've ordered myself one of these sensors and got it to send data to my HomeAssistant through zigbee2mqtt but I don't see any conductivity sensor in HomeAssistant or the zigbee2mqtt device documentation.

Is the AliExpress listing incorrect or can we also get this part of the sensor working through zigbee2mqtt? I am happy to provide data from my setup, I just need to be told what data and how to get it :-)

mercuriete, 2025-08-19:
Mine is stuck at 100% humidity and 100°C.
Not sure why but at least mine is detected as 
https://www.zigbee2mqtt.io/devices/TS0601_soil_2.html#tuya-ts0601_soil_2

The device only reports batery status reliable everything else is 100.
```
{
    "battery": 86,
    "battery_state": "high",
    "humidity_alarm": "cancel",
    "humidity_sensitivity": 1,
    "last_seen": "2025-08-19T14:22:28.836Z",
    "linkquality": 163,
    "max_humidity_alarm": 70,
    "max_temperature_alarm": 40,
    "min_humidity_alarm": 20,
    "min_temperature_alarm": 10,
    "schedule_periodic": 5,
    "soil_moisture": 100,
    "temperature": 100,
    "temperature_alarm": "cancel",
    "temperature_f": 212,
    "temperature_sensitivity": 0.3
}
```

dougrathbone, 2026-02-01:
I have this device working, but the moisture sensitivity is waaaay off. Reading 11% after a full day of downpour