Koenkk/zigbee2mqtt #27806: [Feature request]: Ubisys H10 Thermostat - cool is missing as valid System-/RunningMode

egretzbergerj, 2025-06-24:
### Is your feature request related to a problem? Please describe

When Thermostats of ubisys H10 are set to SystemMode cool via binding with a roomcontroller an error ist logged in homeassistant cause cool is not implemented in the converter for H10.

```
            e
                .climate()
                .withEndpoint("th1")
                .withSystemMode(["off", "heat"], ea.ALL)
                .withRunningMode(["off", "heat"])
                .withSetpoint("occupied_heating_setpoint", 7, 30, 0.5)
                .withLocalTemperature()
                .withPiHeatingDemand(ea.STATE_GET),
```
```
Logger: homeassistant.components.mqtt.climate
Quelle: components/mqtt/climate.py:709
Integration: MQTT (Dokumentation, Probleme)
Erstmals aufgetreten: 22:04:55 (19 Vorkommnisse)
Zuletzt protokolliert: 22:05:03

Invalid modes mode: cool
```

### Describe the solution you'd like

- SystemMode "cool" is implemented for th1-th10
- Parameter PiCoolingDemand ist implemented as well for th1-th10 (nice to have)

### Describe alternatives you've considered

-

### Additional context

-

sjorge, 2025-06-25:
If you're able to test, that would be useful: https://github.com/Koenkk/zigbee-herdsman-converters/pull/9546

sjorge, 2025-06-25:
How are you running your z2m? I can probably just give you the generate reporting.js and ubisys.js files for you to replace.

egretzbergerj, 2025-06-25:
as addon in home assistant

sjorge, 2025-06-25:
Excellent, seems koenkk already merged it. It will be available in next months release.

Koenkk, 2025-06-27:
Thanks @sjorge !

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-06-27 by Koenkk